saki4510t/UVCCamera #6: Sample projects crash after selecting UVC camera

pshen1230, 2015-07-12:
Hello,

I'm an electrical engineer new to Android development.
We're trying to control a UVC camera on Android for a health technology application.

Anyways I compiled couple sample projects (#0, #6) from the library into .apk and installed on a Nexus.
When running the apps I was able to see and select the attached UVC camera, but the apps crashed right after I did so.
However the app CameraFi (which was created also by authors of UVCCamera) worked fine on my Nexus.

Possibly my NDK compiler is not working properly.
Can someone send me compiled .apk for any of the sample projects within the library?
Any other suggestions?

Thanks!
Paul

nmcdonaldd, 2015-07-13:
Having the same issue just running the debug apk file. Crashing after selecting a camera on Android 4.4.2 Samsung Galaxy Tab 8.1 Pro

nmcdonaldd, 2015-07-13:
Seems to be failing [here](https://github.com/saki4510t/UVCCamera/blob/master/sample/USBCameraTest5/src/com/serenegiant/usb/UVCCamera.java#L58). Says: `java.lang.NoClassDefFoundError: com/serenegiant/usb/UVCCamera`. But the other libraries load fine and I am not sure why if this one is going to fail. I'll keep researching.

nmcdonaldd, 2015-07-13:
Figured it out. [This line](https://github.com/saki4510t/UVCCamera/blob/master/libUVCCamera/jni/UVCCamera/serenegiant_usb_UVCCamera.cpp#L223) and [this line](https://github.com/saki4510t/UVCCamera/blob/master/libUVCCamera/jni/UVCCamera/serenegiant_usb_UVCCamera.cpp#L231) need to be changed to the correct directory path of your files. So for instance, I changed line 231 to `com/thd/usbcameratest/usb/UVCCamera` since that is my project structure, and it works fine then (and I did something similar for line 223).

saki4510t, 2015-07-14:
Hi, I'm sorry answer is delay.

If you want to change class name or package name of UVCCamera.java, you also need to change native side as nmcdonaldd said. Additionally need to add proguard settings in `proguard-project.txt` for release build.

Currently it is like this.    

    -keepclassmembers public class com.serenegiant.usb.UVCCamera {
        public *;
        protected *;
        private *;
    }

saki

chungithub, 2015-08-10:
hi,saik!
how to fix this issues?  follow nmcdonaldd,line 231,line 223 no path setting can find.

static jint nativeSetFrameCallback(JNIEnv *env, jobject thiz,
	ID_TYPE id_camera, jobject jIFrameCallback, jint pixel_format) {

	jint result = JNI_ERR;
	ENTER();                                                         //line 223
	UVCCamera *camera = reinterpret_cast<UVCCamera *>(id_camera);
	if (LIKELY(camera)) {
		jobject frame_callback_obj = env->NewGlobalRef(jIFrameCallback);
		result = camera->setFrameCallback(env, frame_callback_obj, pixel_format);
	}
	RETURN(result, jint);
}
                                                                              //line 231

saki4510t, 2015-08-10:
Hi,

Di you see any error message on log cat?  I assume you would forget to change line 846 of `serenegiant_usb_UVCCamera.cpp` to your FQN(fully qualified name).

As you can see on USBCameraTest5 project, `nativeSetFrameCallback` function(and it's dependency; UVCCamera#setFrameCallback and UVCPreview#setFrameCallback) does not depend on what class actually implements the IFrameCallback(but needs to keep existing the instance).  

Although I'm not sure why you want to change FQN...you usually don't change FQN of any classes on Android SDK, Java libraries etc. don't you? I think in most case it is just increase useless trouble...

with best regards,
saki

closed 2016-01-07 by saki4510t