saki4510t/UVCCamera #8: Multiple Cameras/Previews

btut, 2015-08-02:
Hi,
I started from USBCameraTest6 and tried to get it to show two cameras at once. I duplicated the camera button, the USBMonitor and the UVCCameraHandler. I am at the point that I have two CameraViews with a Camera Button on top of each of them. This works fine, as long as I only start one camera at a time. When I try to start the second camera while one is already running, the preview will not start.
What do I have to do to get two previews simultaneously with your library? Is this even possible?
Thank you!

btut, 2015-08-02:
this is the log output when I try to start a second camera:
08-02 19:07:51.739: I/UVCCameraHandler(11020): supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["640x360","320x240","640x480","960x720","1280x720"]},{"index":2,"type":6,"default":1,"size":["640x360","320x240","640x480","960x720","1280x720"]}]}
08-02 19:07:51.739: D/CameraThread(11020): handleStartPreview:
08-02 19:07:51.749: I/libUVCCamera(11020): [11569*UVCPreview.cpp:405:prepare_preview]:frameSize=(640,480)@MJPEG
08-02 19:07:51.759: E/libUVCCamera(11020): [11569*stream.c:1505:uvc_stream_start]:fail
08-02 19:07:51.759: W/libUVCCamera(11020): [11569*diag.c:88:uvc_perror]:failed start_streaming:Unknown error (-99)

saki4510t, 2015-08-03:
Hi,

The log `08-02 19:07:51.749: I/libUVCCamera(11020): [11569UVCPreview.cpp:405:prepare_preview]:frameSize=(640,480)@MJPEG` means negotiation with the specific resolution succeeded.  
The log `08-02 19:07:51.759: E/libUVCCamera(11020): [11569stream.c:1505:uvc_stream_start]:fail` means uvc_stream_start function failed.
I assume it fails when calling libusb_submit_transfer but unfortunately I don't have enough time to confirm the issue now, could you check where/why the function fail?  

FYI, you don't need to duplicate USBMonitor instance, just one USBMonitor instance is better otherwise it may become race condition.   

saki

btut, 2015-08-15:
Hi,
sorry it took me so long to check, I started working in August and did not have a lot of time for this experiment.
However, I found out that uvc_stream_start fails after calling libusb_set_interface_alt_setting
Going deeper, I saw that op_set_interface in android_usbfs.c called by libusb_set_interface_alt_setting in core.c returns LIBUSB_ERROR_OTHER.
Any Idea how I might be able to fix this?

btut, 2015-08-16:
I have checked the log once again, this time for the whole system not only the app, and found out that the second camera is removed due to "epoll hang-up event" before the call to ioctl.
Here is the log, would be great to get any input on how to solve this>

08-16 10:15:01.449: E/ViewRootImpl(12779): sendUserActionEvent() mView == null
08-16 10:15:01.449: I/libUVCCamera(12779): [12805*android_usbfs.c:186:dump_urb]:0:fd=72
08-16 10:15:01.449: I/libUVCCamera(12779): [12805*android_usbfs.c:191:dump_urb]:ファイフディスクリプタフラグ:20002
08-16 10:15:01.449: I/libUVCCamera(12779): [12805*android_usbfs.c:192:dump_urb]:O_ACCMODE:2
08-16 10:15:01.449: I/libUVCCamera(12779): [12805*android_usbfs.c:193:dump_urb]:ノンブロッキングかどうか:0
08-16 10:15:01.449: I/libUVCCamera(12779): [12805*android_usbfs.c:194:dump_urb]:0:type=1,endpopint=0x81,status=0,flag=0
08-16 10:15:01.449: I/libUVCCamera(12779): [12805*android_usbfs.c:195:dump_urb]:0:buffer=0xa190f118,buffer_length=32,actual_length=0,start_frame=0
08-16 10:15:01.449: I/libUVCCamera(12779): [12805*android_usbfs.c:196:dump_urb]:0:number_of_packets=0,error_count=0,signr=0
08-16 10:15:01.449: I/libUVCCamera(12779): [12805*android_usbfs.c:197:dump_urb]:0:usercontext=0xa1906320,iso_frame_desc=0xa191024c
08-16 10:15:01.449: I/EventHub(909): Removing device '/dev/input/event16' due to inotify event
08-16 10:15:01.459: I/UVCCameraHandler(12779): supportedSize:{"formats":[{"index":1,"type":6,"default":1,"size":["1280x720","640x480"]},{"index":2,"type":4,"default":1,"size":["1280x720","640x480","352x288","320x240"]}]}
08-16 10:15:01.459: D/CameraThread(12779): handleStartPreview:
08-16 10:15:01.479: I/libUVCCamera(12779): [12843*UVCPreview.cpp:405:prepare_preview]:frameSize=(1280,720)@MJPEG
08-16 10:15:01.509: I/libUVCCamera(12779): [12845*UVCPreview.cpp:171:callbackPixelFormatChanged]:PIXEL_FORMAT_RAW:
08-16 10:15:04.419: V/MainActivity(12779): onConnect:
08-16 10:15:04.419: D/CameraThread(12779): handleStopPreview:
08-16 10:15:04.419: D/UVCCameraHandler(12779): open:
08-16 10:15:04.419: D/CameraThread(12779): handleOpen:
08-16 10:15:04.419: D/CameraThread(12779): handleClose:
08-16 10:15:04.419: D/CameraThread(12779): handleStopRecording:mMuxer=null
08-16 10:15:04.419: V/RendererHolder(12779): getSurface:surface=Surface(name=android.graphics.SurfaceTexture@148d3945)/@0x2dbf6f5a
08-16 10:15:04.449: I/EventHub(909): Removing device Live! Cam Sync HD VF0770 due to epoll hang-up event.
08-16 10:15:04.449: I/EventHub(909): Removed device: path=/dev/input/event17 name=Live! Cam Sync HD VF0770 id=18 fd=322 classes=0x80000001
08-16 10:15:04.449: E/ViewRootImpl(12779): sendUserActionEvent() mView == null
08-16 10:15:04.519: I/InputReader(909): Device removed: id=18, name='Live! Cam Sync HD VF0770', sources=0x00000101
08-16 10:15:04.519: I/EventHub(909): Removing device '/dev/input/event17' due to inotify event
08-16 10:15:04.549: I/libUVCCamera(12779): [12814*android_usbfs.c:186:dump_urb]:0:fd=90
08-16 10:15:04.549: I/libUVCCamera(12779): [12814*android_usbfs.c:191:dump_urb]:ファイフディスクリプタフラグ:20002
08-16 10:15:04.549: I/libUVCCamera(12779): [12814*android_usbfs.c:192:dump_urb]:O_ACCMODE:2
08-16 10:15:04.549: I/libUVCCamera(12779): [12814*android_usbfs.c:193:dump_urb]:ノンブロッキングかどうか:0
08-16 10:15:04.549: I/libUVCCamera(12779): [12814*android_usbfs.c:194:dump_urb]:0:type=1,endpopint=0x83,status=0,flag=0
08-16 10:15:04.549: I/libUVCCamera(12779): [12814*android_usbfs.c:195:dump_urb]:0:buffer=0xaf87a078,buffer_length=32,actual_length=0,start_frame=0
08-16 10:15:04.549: I/libUVCCamera(12779): [12814*android_usbfs.c:196:dump_urb]:0:number_of_packets=0,error_count=0,signr=0
08-16 10:15:04.549: I/libUVCCamera(12779): [12814*android_usbfs.c:197:dump_urb]:0:usercontext=0xaf822400,iso_frame_desc=0xaf8321ec
08-16 10:15:04.559: I/UVCCameraHandler(12779): supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["640x360","320x240","640x480","960x720","1280x720"]},{"index":2,"type":6,"default":1,"size":["640x360","320x240","640x480","960x720","1280x720"]}]}
08-16 10:15:04.559: D/CameraThread(12779): handleStartPreview:
08-16 10:15:04.569: I/libUVCCamera(12779): [12856*UVCPreview.cpp:405:prepare_preview]:frameSize=(1280,720)@MJPEG
08-16 10:15:04.569: E/libUVCCamera(12779): [12856*stream.c:1445:uvc_stream_start]:libusb_set_interface_alt_setting failed
08-16 10:15:04.569: E/libUVCCamera(12779): [12856*stream.c:1524:uvc_stream_start]:fail
08-16 10:15:04.569: W/libUVCCamera(12779): [12856*diag.c:88:uvc_perror]:failed start_streaming:Unknown error (-99)

saki4510t, 2015-08-17:
Hello,  
I assume you will make some mistakes on handling UVCCamera instance.  
I just add new sample project `USBCameraTest7` to demonstrate how to use two camera at the same time(and show video images from each camera side by side) and it will be help you.  Theoretically multiple camera more than two will be possible if USB bandwidth permit it.
I don't have enough time to check and debug this sample project and it may have some issue but it works fine on my Nexus7(2013), Nexus5, Nexus9 etc. with my several cameras.   
saki

btut, 2015-08-17:
Hi,
Thanks a lot, that is some GREAT help.
I will check as soon as I get home from work.

btut, 2015-08-17:
Hello again,
unfortunately the sample 7 does not work on my Samsung Galaxy S5. I need it to work on this phone since it is the only phone I found on the market with USB OTG 3.0.
Do you think this could be due to the hub?

saki4510t, 2015-08-17:
Hi,  
I just tried `USBCameraTest7` and worked well on my GALAXY S5(SCL23, Android 5.0) with Logitech C910HD & C930e via USB3.0 OTG cable and self-powered USB2.0 hub. Unfortunately I don't have USB3.0 hub...
Did you see any error message?

saki

Darksideneo, 2015-08-24:
Hello.
I also tried to use USBCameraTest7 and it works fine only with the one camera. I used Minoru 3D, but it works the same as 2 webcams. I also tried to use 2 the same cameras and it still doesnt work. Did you tried 2 the same cameras? 
I did using Sony Z3 Tablet Compact.

saki4510t, 2015-08-25:
Hi,
Thank you for providing feedback. I had tested different type cameras and I just tested with two same type camera and worked them as only one camera. Sorry, I found it is because current libuvc/libusb implementation search UVC camera with it's vender id and product id now.

saki

Darksideneo, 2015-08-25:
Thank you for quick reply )
Will you fix it in your project branch?

saki4510t, 2015-08-25:
Yes, I will but it may take several days because I'm very busy now.
I will pass bus number and device address from Java to jni libraries and compare,
because some camera never return actual serial value(iSerialNumber will be zero) and/or return same serial value for every devices and comparing serial value is not a good idea for those cameras.  

saki

Darksideneo, 2015-08-25:
I found that maybe its not the only one problem why 2 cameras dont work in my setup.
Minoru 3d havent serial number as you said, but 2 cameras in it have different product id. When I turn on a one camera the second one always turns off.
In logcat I can see cameras like /dev/bus/usb/001/003 and /dev/bus/usb/001/004.

More details:

device=UsbDevice[mName=/dev/bus/usb/001/004,mVendorId=2760,mProductId=13328,mClass=239,mSubclass=2,mProtocol=1,mManufacturerName=Vimicro Corp.,mProductName=Vimicro USB2.0 PC Camera,mSerialNumber=null,mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=128,mMaxPower=64,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=Vimicro USB2.0 PC Camera,mClass=14,mSubclass=1,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=3,mMaxPacketSize=10,mInterval=5]]

and

device=UsbDevice[mName=/dev/bus/usb/001/003,mVendorId=2760,mProductId=13344,mClass=239,mSubclass=2,mProtocol=1,mManufacturerName=Vimicro Corp.,mProductName=Vimicro USB2.0 PC Camera,mSerialNumber=null,mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=128,mMaxPower=64,mInterfaces=[]]

And also I found exceptions like:

 E/AndroidRuntime﹕ FATAL EXCEPTION: USBMonitor
    Process: com.serenegiant.usbwebcamera, PID: 8983
    android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
            at android.view.ViewRootImpl.checkThread(ViewRootImpl.java:6264)
            at android.view.ViewRootImpl.requestLayout(ViewRootImpl.java:871)
            at android.view.View.requestLayout(View.java:17454)
            at android.view.View.requestLayout(View.java:17454)
            at android.view.View.requestLayout(View.java:17454)
            at android.view.View.requestLayout(View.java:17454)
            at android.view.View.requestLayout(View.java:17454)
            at android.view.View.requestLayout(View.java:17454)
            at android.view.View.requestLayout(View.java:17454)
            at android.view.View.requestLayout(View.java:17454)
            at android.widget.AbsSpinner.requestLayout(AbsSpinner.java:324)
            at android.widget.AbsSpinner.setAdapter(AbsSpinner.java:141)
            at android.widget.Spinner.setAdapter(Spinner.java:432)
            at com.serenegiant.cameracommon.CameraDialog.updateDevices(CameraDialog.java:244)
            at com.serenegiant.cameracommon.BaseFragment.updateCameraDialog(BaseFragment.java:922)
            at com.serenegiant.cameracommon.BaseFragment.access$0(BaseFragment.java:919)
            at com.serenegiant.cameracommon.BaseFragment$2.onAttach(BaseFragment.java:839)
            at com.serenegiant.usb.USBMonitor$2.run(USBMonitor.java:326)
            at android.os.Handler.handleCallback(Handler.java:739)
            at android.os.Handler.dispatchMessage(Handler.java:95)
            at android.os.Looper.loop(Looper.java:211)
            at android.os.HandlerThread.run(HandlerThread.java:61)

and this one:

E/AndroidRuntime﹕ FATAL EXCEPTION: main
    Process: com.serenegiant.usbcameratest7, PID: 24109
    java.lang.UnsatisfiedLinkError: dalvik.system.PathClassLoader[DexPathList[[zip file "/data/app/com.serenegiant.usbcameratest7-1/base.apk"],nativeLibraryDirectories=[/vendor/lib, /system/lib]]] couldn't find "libusb100.so"
            at java.lang.Runtime.loadLibrary(Runtime.java:366)
            at java.lang.System.loadLibrary(System.java:989)
            at com.serenegiant.usb.UVCCamera.<clinit>(UVCCamera.java:105)
            at com.serenegiant.usbcameratest7.MainActivity$2.onConnect(MainActivity.java:158)
            at com.serenegiant.usb.USBMonitor$3.run(USBMonitor.java:346)
            at android.os.Handler.handleCallback(Handler.java:739)
            at android.os.Handler.dispatchMessage(Handler.java:95)
            at android.os.Looper.loop(Looper.java:211)
            at android.app.ActivityThread.main(ActivityThread.java:5321)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1016)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:811)
08-24 19:21:14.257  25171-25171/? E/AndroidRuntime﹕ FATAL EXCEPTION: main
    Process: com.serenegiant.usbcameratest7, PID: 25171
    java.lang.UnsatisfiedLinkError: dalvik.system.PathClassLoader[DexPathList[[zip file "/data/app/com.serenegiant.usbcameratest7-1/base.apk"],nativeLibraryDirectories=[/vendor/lib, /system/lib]]] couldn't find "libusb100.so"
            at java.lang.Runtime.loadLibrary(Runtime.java:366)
            at java.lang.System.loadLibrary(System.java:989)
            at com.serenegiant.usb.UVCCamera.<clinit>(UVCCamera.java:105)
            at com.serenegiant.usbcameratest7.MainActivity$2.onConnect(MainActivity.java:158)
            at com.serenegiant.usb.USBMonitor$3.run(USBMonitor.java:346)
            at android.os.Handler.handleCallback(Handler.java:739)
            at android.os.Handler.dispatchMessage(Handler.java:95)
            at android.os.Looper.loop(Looper.java:211)
            at android.app.ActivityThread.main(ActivityThread.java:5321)
            at java.lang.reflect.Method.invoke(Native Method)
            at java.lang.reflect.Method.invoke(Method.java:372)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1016)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:811)

btut, 2015-09-06:
Hi,
I finally got my Hands on a USB OTG 2.0 cable and USB 2.0 HUB, but the demo is still failing on my Galaxy S5 (Android 5.0). I still get the "stream fail" error when I start the second camera. I will get another Webcam tomorrow and try with that one.

btut, 2015-09-09:
still no luck...

btut, 2015-09-09:
I checked the log and added a few LOGE calls and I can see that it is the function libusb_set_interface_alt_setting that fails. The function registered at usbi_backend->set_interface_altsetting seems to return -99 all the time. Any idea why this could be?
The logs are still the same, [11422*stream.c:1517:uvc_stream_start]:fail when starting the second camera...

saki4510t, 2015-09-09:
Hi,

Sorry for delay response and thank you for providing informations.  
The actual function of libusb_set_interface_alt_setting for Android is `op_set_interface` in `android_usbfs.c`.  The error will occur `ioctl` function in  `op_set_interface` function.  If possible, could you provide what `errno` value it returned?

saki

btut, 2015-09-09:
hi.
thanks for responding!
thats exactly what I found out right now, and ioctl returns error 28, No space left on device. I don't understand how this can be, since I am using USB 3.0. Just to be sure I set the resolution to 320*240 so the used bandwith is minimal, but I still get the same error

btut, 2015-09-11:
Hi,
is it even enough to set DEFAULT_PREVIEW_WIDTH = 320 and DEFAULT_PREVIEW_HEIGHT = 240; to set the Cameras resolution to 320*240? Or is this just how the image is displayed and the camera still runs at higher resolutions?
Also, I installed USB Host Viewer to get some information on my cameras and this is what I got, can anybody calculate the needed bandwith with this information? I don't understand any of it.

041e:4095 
Device : /dev/bus/usb/001/005
  Vendor   : 0x041e Creative Technology, Ltd
  Product  : 0x4095 
  Class    : 0xef Miscellaneous Device
  SubClass : 0x02 ?
  Protocol : 0x01 Interface Association

Interface (1/19)
  Class    : 0x0e Video
  SubClass : 0x01 Video Control
  Protocol : 0x00

Endpoint (1/1)
  Attr.    : 0x83 (num=3,IN)
  Type     : Interrupt
  Interval : 0x06
  Packsize : 0x0010 (16)

Interface (2/19)
  Class    : 0x0e Video
  SubClass : 0x02 Video Streaming
  Protocol : 0x00

Interface (3/19)
  Class    : 0x0e Video
  SubClass : 0x02 Video Streaming
  Protocol : 0x00

Endpoint (1/1)
  Attr.    : 0x81 (num=1,IN)
  Type     : Isochronous
  Interval : 0x01
  Packsize : 0x0080 (128)

Interface (4/19)
  Class    : 0x0e Video
  SubClass : 0x02 Video Streaming
  Protocol : 0x00

Endpoint (1/1)
  Attr.    : 0x81 (num=1,IN)
  Type     : Isochronous
  Interval : 0x01
  Packsize : 0x0200 (512)

Interface (5/19)
  Class    : 0x0e Video
  SubClass : 0x02 Video Streaming
  Protocol : 0x00

Endpoint (1/1)
  Attr.    : 0x81 (num=1,IN)
  Type     : Isochronous
  Interval : 0x01
  Packsize : 0x0400 (1024)

Interface (6/19)
  Class    : 0x0e Video
  SubClass : 0x02 Video Streaming
  Protocol : 0x00

Endpoint (1/1)
  Attr.    : 0x81 (num=1,IN)
  Type     : Isochronous
  Interval : 0x01
  Packsize : 0x0b00 (2816)

Interface (7/19)
  Class    : 0x0e Video
  SubClass : 0x02 Video Streaming
  Protocol : 0x00

Endpoint (1/1)
  Attr.    : 0x81 (num=1,IN)
  Type     : Isochronous
  Interval : 0x01
  Packsize : 0x0c00 (3072)

Interface (8/19)
  Class    : 0x0e Video
  SubClass : 0x02 Video Streaming
  Protocol : 0x00

Endpoint (1/1)
  Attr.    : 0x81 (num=1,IN)
  Type     : Isochronous
  Interval : 0x01
  Packsize : 0x1380 (4992)

Interface (9/19)
  Class    : 0x0e Video
  SubClass : 0x02 Video Streaming
  Protocol : 0x00

Endpoint (1/1)
  Attr.    : 0x81 (num=1,IN)
  Type     : Isochronous
  Interval : 0x01
  Packsize : 0x1400 (5120)

Interface (10/19)
  Class    : 0x01 Audio
  SubClass : 0x01 Control Device
  Protocol : 0x00

Interface (11/19)
  Class    : 0x01 Audio
  SubClass : 0x02 Streaming
  Protocol : 0x00

Interface (12/19)
  Class    : 0x01 Audio
  SubClass : 0x02 Streaming
  Protocol : 0x00

Endpoint (1/1)
  Attr.    : 0x82 (num=2,IN)
  Type     : Isochronous
  Interval : 0x04
  Packsize : 0x0024 (36)

Interface (13/19)
  Class    : 0x01 Audio
  SubClass : 0x02 Streaming
  Protocol : 0x00

Endpoint (1/1)
  Attr.    : 0x82 (num=2,IN)
  Type     : Isochronous
  Interval : 0x04
  Packsize : 0x0048 (72)

Interface (14/19)
  Class    : 0x01 Audio
  SubClass : 0x02 Streaming
  Protocol : 0x00

Endpoint (1/1)
  Attr.    : 0x82 (num=2,IN)
  Type     : Isochronous
  Interval : 0x04
  Packsize : 0x00c8 (200)

Interface (15/19)
  Class    : 0x01 Audio
  SubClass : 0x02 Streaming
  Protocol : 0x00

Endpoint (1/1)
  Attr.    : 0x82 (num=2,IN)
  Type     : Isochronous
  Interval : 0x04
  Packsize : 0x00d8 (216)

Interface (16/19)
  Class    : 0x01 Audio
  SubClass : 0x02 Streaming
  Protocol : 0x00

Endpoint (1/1)
  Attr.    : 0x82 (num=2,IN)
  Type     : Isochronous
  Interval : 0x04
  Packsize : 0x0024 (36)

Interface (17/19)
  Class    : 0x01 Audio
  SubClass : 0x02 Streaming
  Protocol : 0x00

Endpoint (1/1)
  Attr.    : 0x82 (num=2,IN)
  Type     : Isochronous
  Interval : 0x04
  Packsize : 0x0048 (72)

Interface (18/19)
  Class    : 0x01 Audio
  SubClass : 0x02 Streaming
  Protocol : 0x00

Endpoint (1/1)
  Attr.    : 0x82 (num=2,IN)
  Type     : Isochronous
  Interval : 0x04
  Packsize : 0x00c8 (200)

Interface (19/19)
  Class    : 0x01 Audio
  SubClass : 0x02 Streaming
  Protocol : 0x00

Endpoint (1/1)
  Attr.    : 0x82 (num=2,IN)
  Type     : Isochronous
  Interval : 0x04
  Packsize : 0x00d8 (216)

btut, 2015-09-13:
any idea on how to circumvent the "No space left on device" error?
You said USBCameraTest7 works on your Samsung Galaxy S5, what resolution are you using?

btut, 2015-09-13:
Here is the output of lsusb when the camera is connected to a USB port on my ubuntu computer, just in case. Does your library select the right interface for all resolutions? I still cannot find out why one 720p camera is working and two 320*240 are not...
```
Bus 001 Device 007: ID 041e:4095 Creative Technology, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x041e Creative Technology, Ltd
  idProduct          0x4095 
  bcdDevice           20.20
  iManufacturer           3 Creative Technology Ltd.
  iProduct                1 Live! Cam Sync HD VF0770
  iSerial                 2 2014091741833
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength         1264
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          4 Live! Cam Sync HD VF0770
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               5 Live! Cam Sync HD VF0770
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0 
      iInterface              5 Live! Cam Sync HD VF0770
      VideoControl Interface Descriptor:
        bLength                13
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.00
        wTotalLength          107
        dwClockFrequency       15.000000MHz
        bInCollection           1
        baInterfaceNr( 0)       1
      VideoControl Interface Descriptor:
        bLength                18
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0201 Camera Sensor
        bAssocTerminal          0
        iTerminal               0 
        wObjectiveFocalLengthMin      0
        wObjectiveFocalLengthMax      0
        wOcularFocalLength            0
        bControlSize                  3
        bmControls           0x0000000a
          Auto-Exposure Mode
[... 946 lines cut ...]
          Usage Type               Data
        wMaxPacketSize     0x00d8  1x 216 bytes
        bInterval               4
        bRefresh                0
        bSynchAddress           0
        AudioControl Endpoint Descriptor:
          bLength                 7
          bDescriptorType        37
          bDescriptorSubtype      1 (EP_GENERAL)
          bmAttributes         0x01
            Sampling Frequency
          bLockDelayUnits         0 Undefined
          wLockDelay              0 Undefined
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)
```

dwztommy123, 2015-09-14:
Hi.I run cameratest7,but only one camera have video display.

Darksideneo, 2015-09-20:
Does it have some progress to fix issue to make 2 camera works simultaneously?

btut, 2015-09-20:
Hi,
I am currently on vacation and have not done anything since my last post. I kinda hoped for somebody to find a fix while I enjoy my vacation :)

saki4510t, 2015-09-24:
Hi,
I improved the bandwidth issue when using two camera (same product) simultaneously.  
As my test result, any combination of two camera including two same product even if they have no serial number worked well. Some issues may still exist on test app(`UsbCameraTest7`) though. Please try updated version.  
There are some possibility if the camera transfer video images with bulk transfer mode, they may not work well simultaneously.

saki

Darksideneo, 2015-09-24:
Hi, thank you!

After 2nd attempt I saw that both cameras work (by indicator on cameras), but I cant see a preview on my screen in USBCameraTest7 application (even with one camera)

Darksideneo, 2015-09-24:
I changed 
`camera.setPreviewSize(UVCCamera.DEFAULT_PREVIEW_WIDTH, UVCCamera.DEFAULT_PREVIEW_HEIGHT, UVCCamera.DEFAULT_PREVIEW_MODE, 0.5f);`
The 1st time 0.5f to 1f then it becomes work like before (only 1 camera).
The 2nd time 0.5f to 0.7f then it becomes work fine using both cameras.
I am not sure why it works like it works, but I am happy that it finally works =)

saki4510t, 2015-09-24:
Hi,
The actual bandwidth value essentially should be calculated by resolution(width/height), frame rate(**dwFrameInterval**), **dwMaxPayloadTransferSize**, compression ratio(if you use compressing format like MJPEG) etc.  I assume there may be much better way to adjust bandwidth automatically but I don't find yet.

Also, I don't find yet why it works when using my two different model of camera even if  bandwidth parameter(last argument of `setPreviewSize` function) are both 1.0f, but does not work and needs to adjust bandwidth parameter when using my two same model camera...

saki

dwztommy123, 2015-09-25:
Hi:
Two camera can work normally now, but 3 camera can't work and  hints 
 [uvc_stream_start_bandwidth] fail;  Another question is run program in MTK CPU will not work ?

rwxrwrw, 2015-09-27:
Good day! Another strange issue with two identical devices. First stream starts ok with any bandwidth value and device combination. But second stream always:

09-27 16:51:52.205: I/CarVisionSystem(7115): [stream.c:1388:uvc_stream_start_bandwidth]:isochronous transfer mode
09-27 16:51:52.205: I/CarVisionSystem(7115): [stream.c:1504:uvc_stream_start_bandwidth]:Select the altsetting
09-27 16:51:52.205: I/CarVisionSystem(7115): [stream.c:1516:uvc_stream_start_bandwidth]:Set up the transfers
09-27 16:51:52.210: I/CarVisionSystem(7115): [stream.c:1551:uvc_stream_start_bandwidth]:create callback thread
09-27 16:51:52.210: I/CarVisionSystem(7115): [stream.c:1555:uvc_stream_start_bandwidth]:submit transfers
09-27 16:51:52.210: E/CarVisionSystem(7115): [8183*stream.c:1572:uvc_stream_start_bandwidth]:fail
09-27 16:51:52.210: W/CarVisionSystem(7115): [8183*diag.c:88:uvc_perror]:start_streaming:I/O error (-1)

saki4510t, 2015-09-27:
Hi,
As I mentioned, some camera like my Logitech C910HD/C930e/C270 etc. and their any combination do not need bandwidth adjustment(unfortunately I have only one for each). On the other hand, I have two UCAM-DLY300TA(ELECOM) and they need bandwidth adjustment. I don't confirm yet why they are different. Yesterday additional C270 come and I will test.  

Regarding MTK(and also Allwiner and Rockchip), as I know most of them always fail isochronous transfer inside their kernel driver(crash). 
As you will know, most UVC camera use isochronous transfer and when I tried bulk transfer mode with MTK powered device, it worked only few seconds and after several seconds from start streaming, the USB current shut down and the device rebooted... 
If you have rooted device and can access your camera via v4l2, you may use the camera but I don't confirm yet.

saki

btut, 2015-10-02:
Well, I still can't get my two creative cameras to work simultaneously, but I got one more camera now and I can use that one together with one of my creative cameras...
Thanks a lot for your help, saki!

closed 2015-10-02 by btut

quanhua92, 2015-10-20:
@btut : hi, so your solution is to change the camera. I have two usb camera of the same types. It works great on Nexus devices, Sony devices. However, it can't open simultaneously on others. Do you have any suggestion?

btut, 2015-10-20:
Hi,
you either change camera and hope the new one works, or use the bandwith adjustment saki has implemented now. If you play around with the parameters, you should be able to get it to work. Unfortunately this is hardware specific, so I cannot give a solution that works for you for sure.

saki4510t, 2015-10-20:
Hi,
@quanhua92 : If your issue is similar to #19 , that app crashed when opening second camera, try to add permanent permission to your app as described on #19.
As Darksideneo said and as I tested with two same camera of Logitech's and ELECOM's, I feel there are no general way of bandwidth adjustment and the values for each camera is not always same. Ex. Logitech ones work with 1.0 for both bandwidth factor but ELECOM's don't and need 0.9 is one and 0.7 for the others...
saki

quanhua92, 2015-10-21:
@saki4510t , @btut : actually, i can get both the camera work seamlessly on Nexus devices. But the other devices don't work. I can only open 1 camera at a time on the HTC or Asus phone. Besides, it may disconnected continuously. But everything is fine on Nexus & LG G3.

oferinfinityar, 2015-10-29:
Hi Saki,
I'm running usbCameraTest7 on nexus 5 with Minuro webcam via OTG cable.
the app runs, both camera seems to be active according to the lights on the camera but there is no preview on screen.

Will appreciate your support.
Thanks,
Ofer

logcat:

10-29 15:07:28.944: W/InputMethodManagerService(773): Window already focused, ignoring focus gain of: com.android.internal.view.IInputMethodClient$Stub$Proxy@4325aa80 attribute=null, token = android.os.BinderProxy@4330dc20
10-29 15:07:28.954: I/EventHub(773): Removing device '/dev/input/event6' due to inotify event
10-29 15:07:28.954: I/EventHub(773): Removed device: path=/dev/input/event6 name=Vimicro USB2.0 PC Camera id=16 fd=93 classes=0x80000001
10-29 15:07:28.964: I/InputReader(773): Device removed: id=16, name='Vimicro USB2.0 PC Camera', sources=0x00000101
10-29 15:07:28.974: I/libUVCCamera(4667): [4726*UVCPreview.cpp:477:prepare_preview]:frameSize=(640,480)@YUYV
10-29 15:07:31.824: D/dalvikvm(1834): GC_CONCURRENT freed 1112K, 7% free 17966K/19208K, paused 3ms+3ms, total 15ms
10-29 15:07:31.904: D/dalvikvm(1834): GC_CONCURRENT freed 374K, 7% free 18016K/19208K, paused 2ms+1ms, total 12ms
10-29 15:07:32.594: D/audio_hw_primary(189): select_devices: out_snd_device(2: speaker) in_snd_device(0: )
10-29 15:07:35.664: W/InputMethodManagerService(773): Window already focused, ignoring focus gain of: com.android.internal.view.IInputMethodClient$Stub$Proxy@42c93aa8 attribute=null, token = android.os.BinderProxy@4330dc20
10-29 15:07:35.674: I/EventHub(773): Removing device '/dev/input/event7' due to inotify event
10-29 15:07:35.674: I/EventHub(773): Removed device: path=/dev/input/event7 name=Vimicro USB2.0 PC Camera id=17 fd=94 classes=0x80000001
10-29 15:07:35.694: I/InputReader(773): Device removed: id=17, name='Vimicro USB2.0 PC Camera', sources=0x00000101
10-29 15:07:35.694: I/libUVCCamera(4667): [4738*UVCPreview.cpp:477:prepare_preview]:frameSize=(640,480)@YUYV
10-29 15:07:40.444: V/QcrilMsgTunnelSocket(2250): After reading offset = 0 remaining = 4 countRead = 4
10-29 15:07:40.464: D/QcrilMsgTunnelSocket(2250): messageLength extracted from first 4 inputStream reads = 29
10-29 15:07:40.464: V/QcrilMsgTunnelSocket(2250): offset = 0 remaining = 29 countRead = 0
10-29 15:07:40.464: V/QcrilMsgTunnelSocket(2250): offset = 0 remaining = 29 countRead = 29
10-29 15:07:40.464: D/QcrilMsgTunnelSocket(2250): readRilMessage: Buffer = [B@42baddd8 HexData = [0100000004040000110000005155414c434f4d4def0308000100000003]
10-29 15:07:40.464: V/QcrilMsgTunnelSocket(2250): Read packet: 29 bytes. Data Available = 32 Position = 0
10-29 15:07:40.464: D/QcrilMsgTunnelSocket(2250): processResponse. message type = 1. Data Available = 28
10-29 15:07:40.464: D/QcrilMsgTunnelSocket(2250): ByteArray from parcel = 5155414c434f4d4def0308000100000003
10-29 15:07:40.464: D/QcrilMsgTunnelSocket(2250): Received RIL_UNSOL_OEM_HOOK_RAW message
10-29 15:07:40.464: D/QcrilMsgTunnelSocket(2250): Oem ID in RIL_UNSOL_OEM_HOOK_RAW is QUALCOMM
10-29 15:07:40.464: D/QcrilMsgTunnelSocket(2250): OEM ID check Passed
10-29 15:07:40.464: D/QcrilMsgTunnelSocket(2250): Response ID in RIL_UNSOL_OEM_HOOK_RAW is 525295
10-29 15:07:40.474: D/QcrilMsgTunnelSocket(2250): Response ID 525295is not served in this process.
10-29 15:07:40.474: D/QcrilMsgTunnelSocket(2250): To broadcast an Intent via the notifier to external apps
10-29 15:07:40.474: D/QcrilMsgTunnelIfaceManager(2250): handleMessage what=0
10-29 15:07:40.474: D/QcrilMsgTunnelIfaceManager(2250): Broadcasting intent ACTION_UNSOL_RESPONSE_OEM_HOOK_RAW
10-29 15:07:40.474: V/QcrilMsgTunnelSocket(2250): Before reading offset = 0 remaining = 4 countRead = 0
10-29 15:08:05.554: I/PowerManagerService(773): Going to sleep due to screen timeout...
10-29 15:08:06.054: D/SurfaceFlinger(186): Screen released, type=0 flinger=0xb75c9450
10-29 15:08:06.054: D/qdhwcomposer(186): hwc_blank: Blanking display: 0
10-29 15:08:06.354: D/qdhwcomposer(186): hwc_blank: Done blanking display: 0
10-29 15:08:06.354: D/SurfaceControl(773): Excessive delay in blankDisplay() while turning screen off: 304ms
10-29 15:08:06.454: D/NfcService(1043): NFC-C OFF
10-29 15:08:06.464: I/Keyboard.Facilitator(989): onFinishInput()
10-29 15:08:11.434: V/KeyguardHostView(886): Initial transport state: 1, pbstate=0
10-29 15:08:11.494: V/KeyguardHostView(886): music state changed: 0
10-29 15:08:11.514: I/InputReader(773): Reconfiguring input devices.  changes=0x00000004
10-29 15:08:11.514: I/InputReader(773): Device reconfigured: id=4, name='touch_dev', size 1080x1920, orientation 0, mode 1, display id 0
10-29 15:08:11.524: I/InputReader(773): Reconfiguring input devices.  changes=0x00000004
10-29 15:08:11.524: I/ActivityManager(773): Config changes=480 {1.0 ?mcc?mnc en_US ldltr sw360dp w360dp h567dp 480dpi nrml port finger -keyb/v/h -nav/h s.14}
10-29 15:08:11.534: D/EventHub(773): No input device configuration file found for device 'Vimicro USB2.0 PC Camera'.
10-29 15:08:11.544: W/EventHub(773): Unable to disable kernel key repeat for /dev/input/event6: Function not implemented
10-29 15:08:11.544: I/EventHub(773): New device: id=18, fd=93, path='/dev/input/event6', name='Vimicro USB2.0 PC Camera', classes=0x80000001, configuration='', keyLayout='/system/usr/keylayout/Generic.kl', keyCharacterMap='/system/usr/keychars/Generic.kcm', builtinKeyboard=false, usingSuspendBlockIoctl=true, usingClockIoctl=true
[... cut ...]
10-29 15:08:11.694: V/KeyguardHostView(886): music state changed: 0
10-29 15:08:11.714: W/View(886): requestLayout() improperly called by android.widget.TextClock{442c1008 V.ED.... ......ID 223,0-659,320 #7f0a0047 app:id/clock_view} during layout: running second layout pass
10-29 15:08:11.714: W/View(886): requestLayout() improperly called by android.widget.TextClock{443194e0 V.ED.... ......ID 295,0-587,57 #7f0a004c app:id/date_view} during layout: running second layout pass
10-29 15:08:16.534: W/WindowManager(773): App freeze timeout expired.

jp1017, 2017-06-07:
unfortunately, I got IO error, this issue: #180 anybody can give me some advice?

heaveninfosoft, 2020-05-19:
> Hi:
> Two camera can work normally now, but 3 camera can't work and hints
> [uvc_stream_start_bandwidth] fail; Another question is run program in MTK CPU will not work ?

can you help me???