saki4510t/UVCCamera #14: Can't change brightness, contrast with new method released on 2015/07/19

BL-Lau, 2015-09-01:
Hi Saki-san. 

I tried to change the brightness using mUVCCamera.setBrightness(10) or some other value such as 80, but the brightness was not change. When I checked with mUVCCamera.getBrightness(), the returned value was 0. 

Kindly let me know if I have done anything wrong. 

Thanks !

Best regards,
BL

saki4510t, 2015-09-01:
Hi,

Do you know your camera supports the features? And did you see any error message on log cat?
You can check whether the camera supports the features using `UVCCamera#checkSupportFlag` method with `CTRL_XXX` and `PU_XXX` constants.  
As I know some camera need several hundreds milliseconds after open and connect to the camera to get those camera control values.

saki

BL-Lau, 2015-09-02:
Thanks for your quick reply.

The problem is solved! I put UVCCamera#getBrightness in a button control, to allow time to get the control values, it worked. 

I am not familiar with CTRL_XXX and PU_XXX constants, I don't know how to get those and could not use UVCCamera#checkSupportFlag method. Instead, I used  UVCCamera#updateCameraParams, it gave one time error as follows: "E/libUVCCamera(32750): [671*android_usbfs.c:2686: handle_control_completion]:LIBUSB_TRANSFER_STALL". However, the error did not appear again, I was able to change brightness to various values and saw preview updated accordingly. 

Thanks again for your help and your kind deeds to share the libraries.

closed 2015-09-02 by BL-Lau