saki4510t/UVCCamera #16: error when build on windows

bitdust, 2015-09-19:
the [build.gradle](https://github.com/saki4510t/UVCCamera/blob/master/libuvccamera/build.gradle) can't work well on windows.

task ndkBuild in your [build.gradle](https://github.com/saki4510t/UVCCamera/blob/master/libuvccamera/build.gradle) could be better:

    import org.apache.tools.ant.taskdefs.condition.Os
    task ndkBuild(type: Exec, description: 'Compile JNI source via NDK') {
        def ndkDir = project.plugins.findPlugin('com.android.library').sdkHandler.getNdkFolder()
        println(ndkDir)
        if (Os.isFamily(Os.FAMILY_WINDOWS)) {
            commandLine "$ndkDir/ndk-build.cmd",
                    'NDK_PROJECT_PATH=build',
                    'APP_BUILD_SCRIPT=src/main/jni/Android.mk',
                    'NDK_APPLICATION_MK=src/main/jni/Application.mk',
                    '-j8'
        }
        else {
            commandLine "$ndkDir/ndk-build",
                    'NDK_PROJECT_PATH=build',
                    'APP_BUILD_SCRIPT=src/main/jni/Android.mk',
                    'NDK_APPLICATION_MK=src/main/jni/Application.mk',
                    '-j8'
        }
    }

than:

    task ndkBuild(type: Exec, description: 'Compile JNI source via NDK') {
        def ndkDir = project.plugins.findPlugin('com.android.library').sdkHandler.getNdkFolder()
        println(ndkDir)
        commandLine "$ndkDir/ndk-build",
                'NDK_PROJECT_PATH=build',
                'APP_BUILD_SCRIPT=src/main/jni/Android.mk',
                'NDK_APPLICATION_MK=src/main/jni/Application.mk',
                '-j8'
    }

Refer to:https://stackoverflow.com/questions/27477158/android-studio-ndk-build-trouble-errorexecution-failed-for-task-appbuildnati

Darksideneo, 2015-09-19:
It works fine on Windows if you will correct paths in the right way. I can
show how tomorrow.

saki4510t, 2015-09-20:
Hi, thank you for providing feedback.  
I updated build.grade in libuvccamera.  
saki

closed 2015-10-20 by saki4510t