saki4510t/UVCCamera #17: Recording issue in samsung s5 device with 5.1

ravitejaGit, 2015-10-06:
I am trying your uvc camera library it crashing while using in Samsung devices tested in s5 and note 5

btut, 2015-10-06:
Hi,
I am using a Samsung Galaxy S5 without issues now.
Can you try with a different camera?

ravitejaGit, 2015-10-08:
Hi,

I have tired usbCameraTest5 example in Samsung S5, seem while recording the application was crashing and in log it showing "Failed to stop Muxer"

I have seen this crash only on this samsung S5 model, i have tried in other moto series it working fine in that.

Check this link for log/crash details
http://pastebin.com/4tH55hF8

Here S5 Device info
[image]

saki4510t, 2015-10-08:
Hi,
thank you providing information. I have also Galaxy S5 and know why it occur.  
It is because S5 with Android5.x has several issues on USB audio and audio capturing when using AudioRecord and while connecting USB audio device. Could you try to use `MediaRecorder.AudioSource.CAMCORDER` as a audio source instead of `MediaRecorder.AudioSource.MIC`?
saki

closed 2015-10-12 by ravitejaGit