saki4510t/UVCCamera #18: usbCameratest7 cleanup

btut, 2015-10-11:
Hi saki,
I have noticed that there is some inconsistency in your project structure. usbCameratest7 is the only sample project where the first 't' is in lower case, all other projects have a uppercase 'T'. Nevertheless, the file usbCameraTest7.iml has a uppercase 'T' just as the others. This messes up the AndroidStudio setup. Everytime I include this project, AndroidStudio creates a usbCameraTest7 Folder (note the upper case 'T') and searches for all files in that folder, of course without success.
Could you please clean this up in your master branch? Renaming the folder 'usbCameratest7' to 'usbCameraTest7' resolves all issues.
Thanks!

saki4510t, 2015-10-11:
Hi,
Thank you for providing notification. I renamed.  
I usually use OSX and it just ignores case of file name and directory name...
saki

closed 2015-10-11 by saki4510t

btut, 2015-10-11:
Thanks!
I am using OS X as well but it was causing problems.

Closed #18.