saki4510t/UVCCamera #19: Multiple cameras: first Camera crashes when requesting permission for the second

btut, 2015-10-14:
Hi saki,
I have another problem with the library when used with multiple cameras and previews. If I connect one camera and start the preview, then connect a second camera and try to start the preview the first preview stops when the dialog with the USB request shows up.
The only way to get two cameras working is to start the preview for the first, then stop it, then start the preview for the second and start the preview for the first again (this time the permission was already requested and the second camera preview continues).
I guess this is a problem with your USBMonitor implementation.

saki4510t, 2015-10-14:
Hi,
If you want to use two camera at the same time on this sample, it is better to give permanent permission to app for each camera at least before starting second camera. Because, as you can see sample project, if the activity lost focus (ex. by displaying permission request dialog of USB), app try to stop/release all camera and unregister/release USBMonitor on onPause/onDestroy method.  
It is possible to keep camera and USBMonitor working while background but just removing those code is not a good idea and it needs additional code.  
I assume at least following modification will need.
1. pass Application context instead of Activity when creating USBMonitor on `onCreate`
2. check whether mUSBMonitor is null or not when creation.
3. register USBMonitor after creating instead of `onResume`.
4. unregister USBMonitor on `onDestroy` instead of `onPause`.
5. add intent filter for USB attaching event so that app can keep permission permanently.
5. check whether Activity is active or not on all code related to USBMonitor and View and their life cycle(Note:all View and Activity context will become invalid if the Activity lost focus).
...
This is just sample and the main target is showing each concept by easy way/simple code. Those modifications will make app complex and difficult to understand what it do.

saki

btut, 2015-10-14:
Hi,
I don't mind if the preview is destroyed if the application looses focus.
How can I keep the permission permanently as you write?
And sorry, you are right. This is not an issue of the library but of the sample project, but I just can't figure out a way how to get it to work. I worked on this for quite a while.
Thanks for your help!

saki4510t, 2015-10-14:
Hi,

To Holding a permanent permission for USB, you need to add intent filter to `activity` section on`AndroidManifest.xml`. Additionally, it is better to add `launchMode` in `activity` section as `singleTask ` or`singleInstance`(I usually use `singleTask `).
(Regarding details about `launchMode` and why it is better to set as this, please check Android Developer site.)
```
<activity
	android:launchMode="singleTask"
	...
	<intent-filter>
		<action android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED" />
		<meta-data
			android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED"
			android:resource="@xml/device_filter" />
	</intent-filter>
	...
```
You can see this on `AndroidManifest.xml` of UsbCameraTest4 sample project.
If there is above intent filter in `AndroidManifest.xml` and once user choose the app as default app(always use the app for the device), Android system automatically give a permission to the Activity every time connecting the USB device (and you can not select other similar app). You can clear this setting on `Apps` setting.
You can also add `USB_DEVICE_DETACHED` intent filter to your `AndroidManifest.xml`, but some Android devices with some Android version can not handle this intent filter correctly.  
Please mind that adding above intent filter let app accept all connecting event of UVC camera. This means when connect UVC camera to your device and the app does not have permission for the device, Android system always ask you permission. If you only need specific UVC camera, please add the vendor id and product id to `device_filter.xml`

saki

btut, 2015-10-15:
Thanks for your help. Now, when I plug in the hub the app opens and the preview for both cameras starts fine, just as I wanted!

closed 2015-10-15 by btut