saki4510t/UVCCamera #20: Requesting USB permission failed on Samsung's Android5.1.1

saki4510t, 2015-10-22:
As I know, android.hardware.usb.UsbManager#requestPermission always throws java.lang.SecurityException related to android.permission.sec.MDM_APP_MGMT on Samsung's Android5.1.1 and never get USB access permission on those devices now.
```
java.lang.SecurityException: Admin  does not have android.permission.sec.MDM_APP_MGMT
	at android.os.Parcel.readException(Parcel.java:1546)
	at android.os.Parcel.readException(Parcel.java:1499)
	at android.hardware.usb.IUsbManager$Stub$Proxy.requestDevicePermission(IUsbManager.java:653)
	at android.hardware.usb.UsbManager.requestPermission(UsbManager.java:432)
```
saki

pkellis, 2015-12-17:
Confirmed. Why on earth would Samsung do that? No USB devices are supported any more.

saki4510t, 2015-12-17:
This issue have been occurring from early October. You can see related info on SAMSUNG DEVELOPERS site;
http://developer.samsung.com/forum/board/thread/view.do?boardName=General&messageId=285782
So SAMSUNG know there is a issue on there devices with Android 5.1.1. But unfortunately there are still no workaround, no fix now.
I don't confirm yet and want to know whether this issue still exist on Android6 of SAMSUNG.

abhishekkumar1902, 2017-08-26:
Something strange is happening with my case. I am using Samsung note edge running Android v5.1.1. I, where I compiled all the 9 uvccamera projects and tested on the phone and it was successfully able to detect the camera and fetch the video from the camera as well. Now, just what I did is just copied the same code and layout widget design from **usbCameraTest0** project line by line without any modification and integrated it into a new project targeted at **same compiler, API and SDK level**. There happens no problems in compilation and starting of this new app but the problem is that now when I press the "connect to" camera button, the dialog box comes in but it does not detect the camera at all. Please help me with this issue. At the same time, if I run the usbCameraTest0 project alone, it runs fine. Not sure what could be wrong. Please help.