saki4510t/UVCCamera #23: usbCameraTest7 - Minuro Dual WebaCam Previews

oferinfinityar, 2015-10-29:
Hi Saki,
I'm running usbCameraTest7 on nexus 5 with Minuro webcam via OTG cable.
the app runs, both camera seems to be active according to the lights on the camera but there is no preview on screen.

I saw a closed issue on this, but I still have and issue, so opening a new one.

Will appreciate your support.
Thanks,
Ofer

logcat (short one):
10-29 15:07:28.944: W/InputMethodManagerService(773): Window already focused, ignoring focus gain of: com.android.internal.view.IInputMethodClient$Stub$Proxy@4325aa80 attribute=null, token = android.os.BinderProxy@4330dc20
10-29 15:07:28.954: I/EventHub(773): Removing device '/dev/input/event6' due to inotify event
10-29 15:07:28.954: I/EventHub(773): Removed device: path=/dev/input/event6 name=Vimicro USB2.0 PC Camera id=16 fd=93 classes=0x80000001
10-29 15:07:28.964: I/InputReader(773): Device removed: id=16, name='Vimicro USB2.0 PC Camera', sources=0x00000101
10-29 15:07:28.974: I/libUVCCamera(4667): [4726*UVCPreview.cpp:477:prepare_preview]:frameSize=(640,480)@YUYV
10-29 15:07:31.824: D/dalvikvm(1834): GC_CONCURRENT freed 1112K, 7% free 17966K/19208K, paused 3ms+3ms, total 15ms
10-29 15:07:31.904: D/dalvikvm(1834): GC_CONCURRENT freed 374K, 7% free 18016K/19208K, paused 2ms+1ms, total 12ms
10-29 15:07:32.594: D/audio_hw_primary(189): select_devices: out_snd_device(2: speaker) in_snd_device(0: )
10-29 15:07:35.664: W/InputMethodManagerService(773): Window already focused, ignoring focus gain of: com.android.internal.view.IInputMethodClient$Stub$Proxy@42c93aa8 attribute=null, token = android.os.BinderProxy@4330dc20
10-29 15:07:35.674: I/EventHub(773): Removing device '/dev/input/event7' due to inotify event
10-29 15:07:35.674: I/EventHub(773): Removed device: path=/dev/input/event7 name=Vimicro USB2.0 PC Camera id=17 fd=94 classes=0x80000001
10-29 15:07:35.694: I/InputReader(773): Device removed: id=17, name='Vimicro USB2.0 PC Camera', sources=0x00000101
10-29 15:07:35.694: I/libUVCCamera(4667): [4738*UVCPreview.cpp:477:prepare_preview]:frameSize=(640,480)@YUYV

saki4510t, 2015-10-30:
Hi,

1. If you connect only one camera, it works well? 
2. If your camera supports mjpeg mode, please try it.
3. if you can, please try to set different bandwidth parameter for one camera and get what value app failed to get video images. And try to set the value for two camera.
4. Could you provide device descriptor of your camera?  
    I want to know whether your camera supports isochronous transfer and the value of dwMaxPayloadTransferSize etc. 
    you can get it several ways, 
    1. set `LOCAL_DEBUG` flag to 1 in UVCCamera.cpp and rebuild & run. then you can see it in logCat.
    2. or call getDescriptions function after open the camera to get device descriptor as C style string(null terminated string) and write it to logCat/file
    3. or if you can get my apps(UsbWebCamera/UsbWebCameraPro) from GooglePlay, please run it with your camera. The apps save device descriptor to local storage.
        The descriptor is saved under following pass.  
        `{local_storage}/Android/data/com.serenegiant.usbwebcamera/fiels/Descriptor.`
    4. or if you use Linux, connect the camera to the PC and run `lsusb -s bus:dev -v`(you can see bus and dev when run lsusb without option).
    etc.  

saki

oferinfinityar, 2015-10-31:
Hi Saki,
Thank you very much for your quick response.

I'm using the Minuro stereo web cam: http://www.minoru3d.com/
Its a very useful camera for computer vision applications.

1. one camera works on other tests I tried, but not two at the same time.
2. I don't think it supports mjpeg, in any case I need the data uncompressed.
3. I need 640x480, this is what I tried, should be ok for USB2 with two cameras of such resolution.
4. from what I read it does support isochronous transfer, see: http://comments.gmane.org/gmane.linux.drivers.uvc.devel/3909.
I will try and send you the value.

saki4510t, 2015-10-31:
Hi,
I try to find your camera information. I could't find actual descriptor but as I surface several site, including that you pointed out, I expects the camera will support isochronous transfer and mjpeg mode because it support 1280x480 & 800x600 with 30fps. 
> Output size - 1280x480(15/30fps), 800x600(15/30fps), 704x288(30fps), 640x480(15/30fps), 640x240(30fps), 352x288(30fps), 320x240(30fps) pixels
• Frame rate - 15 or 30 frames per second, user selectable

But your log said 
> 10-29 15:07:35.694: I/libUVCCamera(4667): [4738UVCPreview.cpp:477:prepare_preview]:frameSize=(640,480)@YUYV

The site that you pointed out shows that they could get 640x480 video images from two camera at the same time, but is was available only 15fps. Could you also try to use it with 15fps?
saki

oferinfinityar, 2015-11-01:
How do I set the frame rate?
I couldn't find anything on UVCCamera api.

oferinfinityar, 2015-11-01:
Hi Saki,
I changed this line to 15fps:
 #define DEFAULT_PREVIEW_FPS 15 //30

no change.

oferinfinityar, 2015-11-01:
descriptor from uvcwebcam:
{"description":{"venderId":2760,"productId":13328,"serialNumber":"[none]","manifuctureName":"Vimicro Corp.","productName":"Vimicro USB2.0 PC Camera","uvc":{"videoControl":256,"interfaces":[{"type":"videoStreaming","index":1,"endpointAddress":130,"formats":[{"index":1,"subType":4,"format":"UncompressedFormat","detail":{"bitsPerPixel":16,"GUID":"5955593200001000800000aa00389b71","defaultFrameIndex":1,"aspectRatioX":0,"aspectRatioY":0,"interlaceFlags":0,"copyProtect":0,"frameDescriptors":[{"index":1,"capabilities":0,"width":640,"height":480,"minBitRate":768000,"maxBitRate":196608000,"maxFrameBufferSize":614400,"defaultFrameInterval":333333,"defaultFps":30,"intervals":[{"index":0,"value":333333,"fps":30},{"index":1,"value":666667,"fps":14}]},{"index":2,"capabilities":0,"width":352,"height":288,"minBitRate":768000,"maxBitRate":196608000,"maxFrameBufferSize":202752,"defaultFrameInterval":333333,"defaultFps":30,"intervals":[{"index":0,"value":333333,"fps":30},{"index":1,"value":666667,"fps":14}]},{"index":3,"capabilities":0,"width":320,"height":240,"minBitRate":768000,"maxBitRate":196608000,"maxFrameBufferSize":153600,"defaultFrameInterval":333333,"defaultFps":30,"intervals":[{"index":0,"value":333333,"fps":30},{"index":1,"value":666667,"fps":14}]},{"index":4,"capabilities":0,"width":176,"height":144,"minBitRate":768000,"maxBitRate":196608000,"maxFrameBufferSize":50688,"defaultFrameInterval":333333,"defaultFps":30,"intervals":[{"index":0,"value":333333,"fps":30},{"index":1,"value":666667,"fps":14}]},{"index":5,"capabilities":0,"width":160,"height":120,"minBitRate":768000,"maxBitRate":196608000,"maxFrameBufferSize":38400,"defaultFrameInterval":333333,"defaultFps":30,"intervals":[{"index":0,"value":333333,"fps":30},{"index":1,"value":666667,"fps":14}]}]}}]}]}}}
********raw_descriptor********
12010002ef020140c80a10340001010200010902e1010201008040080b00020e03000209040000010e0100020d240100014f0080c3c90101011224020101020000000000000000030a00000b240502010000027b17000924030301010002001c2406045dc717a91941da11ae0e000d56ac7b4c0801010359808000070581030a00050525030a0009040100000e0200000e240101f30082000302010101001b240401055955593200001000800000aa00389b7110010000000022240501008002e00100b80b000000b80b006009001516050002151605002b2c0a0022240502006001200100b80b000000b80b001803001516050002151605002b2c0a0022240503004001f00000b80b000000b80b005802001516050002151605002b2c0a002224050400b000900000b80b000000b80b00c600001516050002151605002b2c0a002224050500a000780000b80b000000b80b009600001516050002151605002b2c0a001a240300058002e001600120014001f000b0009000a00078000006240d00000009040101010e0200000705820580000109040102010e0200000705820500020109040103010e0200000705820500040109040104010e02000007058205000b0109040105010e02000007058205000c0109040106010e0200000705820580130109040107010e02000007058205001401

saki4510t, 2015-11-02:
Hi,
As I parsed the descriptor, the camera only support YUY2 format with isochronous transfer(no mjpeg). And it's `dwMinBitRate` and `dwMaxBitRate` of Video Frame Descriptors have different value, so I assume it may be better to use them when negotiating video streaming control in `uvc_stream_start_bandwidth` function(in stream.c) instead of current implementation. It will take a time.
saki

```
*****raw_descriptor*****
12			bLength						0x12 = 18bytes
01			bDescriptorType				Device Descriptor
0002		bcdUSB						0x0200 = USB2.0
ef			bDeviceClass				Miscellaneous Device Class
02			bDeviceSubClass				Common Class
01			bDeviceProtocol				Interface Association Descriptor
40			bMaxPacketSize0				Control endpoint packet size is 64 bytes
c80a		idVendor
1034		idProduct
0001		bcdDevice
01			iManufacturer
02			iProduct
00			iSerialNumber
01			bNumConfigurations

09			bLength						0x09 = 9bytes
02			bDescriptorType				Configuration Descriptor
e101		wTotalLength				0x01e1 = 481bytes
02			bNumInterfaces				2 interfaces
01			bConfigurationValue			ID of this configuration
00			iConfiguration				unused
80			bmAttributes				bus powered, remote wakeup disabled
40			bMaxPower					128mA

Interface Association Descriptor(IAD, Video)
08			bLength						0x08 = 8bytes
0b			bDescriptorType				Interface Association Descriptor
00			bFirstInterface				0
02			bInterfaceCount				2
0e			bFunctionClass				CC_VIDEO
03			bFunctionSubClass
00			bFunctionProtocol			unused, should be PC_PROTOCOL_UNDEFINED(=0x00)
02			iFunction

Standard Control Interface Descriptor(Video)
09			bLength						0x09 = 9bytes
04			bDescriptorType				Interface Descriptor
00			bInterfaceNumber			Interface ID = 0
00			bAlternateSetting			AlternateSetting ID = 0
01			bNumEndpoints
0e			bInterfaceClass				CC_VIDEO
01			bInterfaceSubClass			SC_VIDEOCONTROL
00			bInterfaceProtocol			PC_PROTOCOL_UNDEFINED(0) for UVC1.0/1.0, PC_PROTOCOL_15(=0x01) for UVC1.5
02			iInterface

↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓
Class-specific Control Interface Header Descriptor(VideoControl)
0d			bLength						0x0d = 13bytes
24			bDescriptorType				CS_INTERFACE
01			bDescriptorSubType			VC_HEADER
0001		bcdUVC						0x100 = UVC1.00
4f00		wTotalLength				0x004f = 79 
80c3c901	dwClockFrequency			deprecated
01			bInCollection
01			baInterfaceNr				related video streaming number = 1

Input Terminal Descriptor(ITD, VideoControl)
12			bLength						0x12 = 18bytes
24			bDescriptorType				CS_INTERFACE
02			bDescriptorSubtype			VC_INPUT_TERMINAL
01			bTerminalID
0102		wTerminalType				0x0201 = ITT_CAMERA
00			bAssocTerminal
00			iTerminal
[... 313 lines cut ...]
05			bDescriptorType				Endpoint Descriptor
82			bEndpointAddress			input endpoint, address = 2
05			bmAttributes				Isochronous transfer type. Asynchronous synchronization type
8013		wMaxPacketSize				0x1380
01			bInterval					1

Standard VideoStream Interface Descriptor
09			bLength						0x09 = 9bytes
04			bDescriptorType				Interface Descriptor
01			bInterfaceNumber			interface ID = 1
07			bAlternateSetting			AlternateSetting ID = 7
01			bNumEndpoints				1
0e			bInterfaceClass				CC_VIDEO
02			bInterfaceSubClass			SC_VIDEOSTREAMING
00			bInterfaceProtocol			PC_PROTOCOL_UNDEFINED(0) for UVC1.0/1.0, PC_PROTOCOL_15(=0x01) for UVC1.5
00			iInterface

Standard VideoStream Isochronous Video Data Endpoint Descriptor
07			bLength						0x07 = 7bytes
05			bDescriptorType				Endpoint Descriptor
82			bEndpointAddress			input endpoint, address = 2
05			bmAttributes				Isochronous transfer type. Asynchronous synchronization type
0014		wMaxPacketSize				0x1400
01			bInterval					1
```

oferinfinityar, 2015-11-02:
Saki,
Can I hire your service to make it quick?
ofer@infinityar.com

saki4510t, 2015-11-03:
Hi,
The description of `dwMinBitRate` and `dwMaxBitRate` on UVC specification is just
> dwMinBitRate: Specifies the minimum bit rate at the longest frame interval in units of bps at which the data can be transmitted.
dwMaxBitRate: Specifies the maximum bit rate at the shortest frame interval in units of bps at which the data can be transmitted.

And using these is just my assuming that they may help to determine the bandwidth setting. Unfortunately I can't find additional informations about these on internet and as I know UVC kernel driver of Linux does not use these values and I'm not sure they should help to improve the issue (these values may just show the min/max bitrate that camera can prepare to send).

As I know the usual bandwidth estimation is as follows(when using uncompressed format);  
As you can see previous descriptor, uncompressed Video Frame Descriptors, ex. following one, this is YUV(4:2:2) frame descriptor and the `bBitsPerPixel ` values is 16[bits].
```
22          bLength                     0x22 = 34bytes
24          bDescriptorType             CS_INTERFACE 
05          bDescriptorSubtype          VS_FRAME_UNCOMPRESSED
01          bFrameIndex
00          bmCapabilities
8002        wWidth                      0x0280 = 640    640x480
e001        wHeight                     0x01e0 = 480
00b80b00    dwMinBitRate                0x000bb800 = 768000 bps[bits/sec]
0000b80b    dwMaxBitRate                0x0bb80000 = 196608000 bps[bits/sec]
00600900    dwMaxVideoFrameBufferSize   0x00096000 = 614400[bytes]
15160500    dwDefaultFrameInterval      0x00051615 = 333333 = 1/30 
02          bFrameIntervalType          number of supported frame intervals = 2
15160500    dwFrameInterval(1)          0x00051615 = 333333[100ns] = 1/30 [fps]
2b2c0a00    dwFrameInterval(2)          0x000a2c2b = 666667[100ns] = 1/15(1/14.9999925)[fps]
```
So one video frame size should be
```
video frame size = wWidth x wHeihjt x bBitsPerPixel / 8 = 640 x 480 x 16 /8 = 614400 [bytes]
```
And shortest frame interval is
```
333333[100ns]→30 [frame/sec] (30dps).
```
So we need to transfer;
```
614400 bytes x 30 [frame/sec] = 18432000 [bytes/sec]
```
If we can use USB 2.0(high speed, 480Mbps), we need to send each frame every 125[us],
So the frame size will be
```
18432000 / 1000 / 8 = 2304 [bytes/frame (=bytes/125us)]
```

And each UVC frame need header, it is 12 bytes, and the actual size will be
```
2304 + 12 = 2316 [bytes/frame (=bytes/125us)]
```
This value show that packet size smaller than 2316bytes can not transfer for 640x480@30fps. 
Then select endpoint using this value `2316 [bytes/frame (=bytes/125us)] ` and `wMaxPacketSize` in endpoint descriptor.

saki

oferinfinityar, 2015-11-03:
Is there a way to change to lower fps?

saki4510t, 2015-11-03:
when using your camera with 640x480, we should negotiate either 15fps(`dwFrameInterval(2)`) or 30fps(`dwFrameInterval(1)`) otherwise camera does not accept the value. And actual frame rate will be determined by camera according to the negotiation request(requested bFormatIndex, bFrameIndex, dwFrameInterval, dwMaxVideoFrameSize, dwMaxPayloadTransferSize etc).

Actually, Minuro webcam will come within several weeks and if it come I may improve the way to negotiation.
saki

oferinfinityar, 2015-11-03:
Saki,
If you'll send me your address I'll send you a unit right away.

saki4510t, 2015-11-03:
Hi,
My ordered one was already sipped and on the way from US.　Sipping information from seller show Estimated delivery Nov 11, 2015 - Nov 19, 2015. I'm not sure whether it is faster/slower that your sipping.
And I can not guarantee that I should be able to solve the issue even if i have same camera.

BTW, I remembered additional idea; using `uvc_query_stream_ctrl` with `UVC_GET_MIN` request.
If the camera returns correct minimum value of dwMaxPayloadTransferSize, it may improve the issue.
saki

saki4510t, 2015-11-10:
Hi,
Today Minuro came. As I tested, using bandwidth factor 0.67 can start stream two camera at the same time but camera always return payload with error flag and it's size is always invalid(614400 bytes required but only 1024 bytes return) when low bandwidth parameter.
I tried to install attached software from CD but there is no luck, installer always failed.

oferinfinityar, 2015-11-10:
Does single camera works for you?
It does for me.

saki4510t, 2015-11-10:
Both left and right one works but two camera at the same time do not work.

oferinfinityar, 2015-11-10:
on windows its working fine.
I also tried on android reading directly from the driver and it works.

Ofer Kruzel
VP R&D
-------------------------------
M: +972-54-4516627
www.InfinityAR.com <http://www.infinityar.com/>
www.facebook.com/InfinityAR
www.twitter.com/Infinity_AR

oferinfinityar, 2015-11-19:
Hi Saki,
Can you update on your plan of supporting this?
Thanks,
Ofer

Ofer Kruzel
VP R&D
-------------------------------
M: +972-54-4516627
www.InfinityAR.com <http://www.infinityar.com/>
www.facebook.com/InfinityAR
www.twitter.com/Infinity_AR

saki4510t, 2015-11-20:
Hi, unfortunately I couldn't get good result yet.
saki

oferinfinityar, 2015-11-20:
Are you planning to continue trying?

saki4510t, 2015-11-24:
Hi,
I just got images from both camera of Minuro Dual WebaCam with adjusting negotiation parameters by manually. Unfortunately I don't find the way yet to get the correct parameters automatically.

saki

oferinfinityar, 2015-11-24:
Manual is a good start. Can I get the fix?

saki4510t, 2015-11-24:
Not yet.
If you want to modify by yourself, always force select alt setting 4 in `uvc_stream_start_bandwidth` function in `stream.c` and use 15(14.99) fps.
saki

oferinfinityar, 2015-11-24:
Hi Saki,
can you send the line to change?

Ofer Kruzel
VP R&D
-------------------------------
M: +972-54-4516627
www.InfinityAR.com <http://www.infinityar.com/>
www.facebook.com/InfinityAR
www.twitter.com/Infinity_AR

saki4510t, 2015-11-25:
Hi,
I just add new branch `minoru001`. please check-out and try.
saki

oferinfinityar, 2015-11-25:
Hi Saki,
Its working, super cool!!!
Thanks you very much.
Ofer

Ofer Kruzel
VP R&D
-------------------------------
M: +972-54-4516627
www.InfinityAR.com <http://www.infinityar.com/>
www.facebook.com/InfinityAR
www.twitter.com/Infinity_AR

closed 2016-05-09 by saki4510t