saki4510t/UVCCamera #24: How do I quit dialog to usb permission?

fh127, 2015-11-03:
Hi, Saki.

I want to connect to USB Camera from app android without request permission to communicate with the device. is it possible?

Thank for attention.

btut, 2015-11-03:
Hi,
I had the same question, and saki advised to use an intent-filter. just 
use this code in your manifest file in the main activity:

<intent-filter><action 
android:name="android.intent.action.MAIN"/><category 
android:name="android.intent.category.LAUNCHER"/></intent-filter><intent-filter><action 
android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED"/></intent-filter><intent-filter><action 
android:name="android.hardware.usb.action.USB_DEVICE_DETACHED"/></intent-filter><meta-data 
android:name="android.hardware.usb.action.USB_DEVICE_ATTACHED" 
android:resource="@xml/device_filter"/>

You will be asked once if you want to allow the app to access the USB 
camera, and afterwards everytime you connect the camera the app is started.

Hope this helps!

saki4510t, 2015-11-03:
Hi,

You can enumerate connected USB devices using UsbManager and get vendor id, product id etc.
But if you want to open the device and get data from that, you always needs to request permission unless if you have root access and seting-up correctly.

Intent filter in AndroidManifest.xml is only way to give permanent permission to the Activity without root access. And if you don't need permanent permission, you can omit it and programatically request permission.

If you need permanent permission and don't want to start app every time the device(s) connected to your Android, it is also possible to handle it because all activity receive intent to indicate why/how it start.

1. get intent by calling `Activity#getIntent` in your activity,
2. check whether the intent is not null,
3. check whether the action of intent is USB_DEVICE_ATTACHED using `Intent#getAction`
4. get intent flags using `Intent#getFlags`
5. check whether the bit24&bit25 of the flags is on or not
6. if the bit24&bit25 is off(flags will be 0x10000000), it shows Android system automatically give permission to the activity because the activity has permanent permission.
7. do anything as you want. if you cancel startup, just call `Activity#finish`

Please mind if you don't show activity, you need to do above before `setContentView`.

saki

closed 2016-01-07 by saki4510t

SPURVESH, 2017-07-20:
@saki4510t can you please elaborate or provide an example to bypass usb permission for a rooted device ?

yeshiyuan00, 2018-04-18:
how to bypass the permission dialog on a rooted device ?

SPURVESH, 2018-04-18:
@yeshiyuan00 yes, and we did that change in the source code itself and build the image from it.

imprezakd, 2020-10-19:
> @ yeshiyuan00是的，我們在源代碼本身中進行了更改，並從中構建了圖像。

How can I modify it and whether it can provide teaching。thanks!