saki4510t/UVCCamera #28: Sample projects  dont work after selecting UVC camera

feng0403, 2015-12-14:
Sample projects  dont work after selecting UVC camera. The screen has no image. Then about ten seconds ,it crashed.
the logcat is blow:

12-14 11:32:54.451 13135-13135/com.Beyondh V/MainActivity: onResume:
12-14 11:32:54.681 13135-13135/com.Beyondh D/OpenGLRenderer: Enabling debug mode 0
12-14 11:32:54.691 13135-13135/com.Beyondh V/MainActivity: surfaceCreated:
12-14 11:32:54.691 13135-13135/com.Beyondh V/MainActivity: surfaceChanged:
12-14 11:36:24.201 13135-13135/com.Beyondh V/MainActivity: onConnect:
12-14 11:36:24.221 13135-16199/com.Beyondh I/MainActivity: supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["640x480","352x288","320x240","176x144","160x120"]},{"index":2,"type":6,"default":1,"size":["640x480","352x288","320x240","176x144","160x120"]}]}
12-14 11:36:24.231 13135-16202/com.Beyondh I/libUVCCamera: [16202*UVCPreview.cpp:489:prepare_preview]:frameSize=(640,480)@MJPEG
12-14 11:36:24.241 13135-16204/com.Beyondh W/art: Thread[16,tid=16204,Native,Thread*=0x449ea220,peer=0x64850468,"Thread-289"] attached without supplying a name
12-14 11:36:24.241 13135-16204/com.Beyondh I/libUVCCamera: [16204*UVCPreview.cpp:259:callbackPixelFormatChanged]:PIXEL_FORMAT_YUV:
12-14 11:37:05.311 13135-16201/com.Beyondh A/libc: Fatal signal 11 (SIGSEGV) at 0x4d858000 (code=2), thread 16201 (pool-1-thread-1)

saki4510t, 2015-12-14:
Hi,
Thank you for providing feedback. Could you provide 1) native crash dump(if will see them if you un-filter logCat), 2) your device information, 3) camera model, 4)which sample project do you use(I assume you changed something, if so, also let me know whether original master branch has same issue or not) 

saki

leehack, 2015-12-14:
@saki4510t Previously I noticed similar issue. At that time, it was related the fps. the uvc camera I tested haven't 30fps profile even it has 60fps. because of this, sometimes it crashed, sometimes it just frozen.
Maybe it would be nice to implement API for setting fps and add fps into the supportedSize list as well.

saki4510t, 2015-12-14:
Hi,

Ok, I will add feature so that app set fps(range) from Java side. But I don't think fps setting cause crash. If fps and bandwidth settings do not match with camera configuration, it just lead no Images instead of crash.

Unfortunately current implementation of libjpeg & libjpeg-turbo with most devices has no enough performance for decoding 1080p@60fps (this is also difficult on low - middle range PC).
Ex. Nexus9 & Nexus6p take at least 30-35msec/frame@1080p for mjpeg decoding(& rendering) and it is  approx.30fps. Lower resolution like 720p will much shorter and get faster fps though. I know theoretically there is a way to improve performance of mjpeg decoding more but there is no such library as open source now.
saki

leehack, 2015-12-14:
Hi @saki4510t,

Maybe I am wrong or missing something but we could improve this parts to have better fps? Seems like there's two format change(mjpeg->yuv->rgbx). Can't we just directly use `uvc_mjpeg2rgbx()` instead of `uvc_mjpeg2yuyv()`?
if we convert it directly to rgbx instead of yuyv, we will at least save one buffer copy before you put into Suface, right?

```
void UVCPreview::do_preview(uvc_stream_ctrl_t *ctrl)
...
// MJPEG mode
			for ( ; LIKELY(isRunning()) ; ) {
				frame_mjpeg = waitPreviewFrame();
				if (LIKELY(frame_mjpeg)) {
					frame = get_frame(frame_mjpeg->width * frame_mjpeg->height * 2);
					result = uvc_mjpeg2yuyv(frame_mjpeg, frame);   // MJPEG => yuyv
					recycle_frame(frame_mjpeg);
					if (LIKELY(!result)) {
						frame = draw_preview_one(frame, &mPreviewWindow, uvc_any2rgbx, 4);
						addCaptureFrame(frame);
					} else {
						recycle_frame(frame);
					}
				}
			}
```

saki4510t, 2015-12-16:
Hi,@feng0403, could you provide information(device descriptor) about your camera?

@leehack,
I added feature so that user can request fps range when negotiating with UVC camera.  
Of course you can use `uvc_mjpeg2rgbx`. Currently implementation is just sample and expects  the frame format is YUYV (YUV2) at most part to reduce code.  
As I tested, the fastest way now to render MJPEG frames is; decompress MJPEG frames using libjpeg-turbo as YCbCr planer and send them to GPU as OpenGL|ES2 texture and render them by fragment shader with color format conversion.

leehack, 2015-12-16:
@saki4510t Thanks for the FPS change. I'll try it soon.

closed 2016-01-07 by saki4510t

marvr, 2016-03-28:
I used `uvc_mjpeg2rgbx`,but it crashed sometimes.

06110902002, 2023-10-31:
@marvr  i  face the some problem had't you fix this issue?