saki4510t/UVCCamera #29: Dual camera recorder

pricsko, 2015-12-16:
Hi, i want to implement a dual camera recorder. I duplicated every class (in encoder, widget packages - with every innerclasses), but unfortunately, i always have only 1 recorded video, the other file (randomly from camera1 or camera2) is always an empty mp4 (that players cannot play).

saki4510t, 2015-12-16:
Hi,
I assume you tried to record audio for both video at the same time. Unfortunately app can not access audio device(mic) at the same time on Android. If you don't need to record audio, try to disable it.
As a sample, I add movie recording(w/o audio) for each camera on `UsbCameraTest7`.

If you want to record audio into to two or more movie at the same time, you need to make a special wrapper class to access audio(mic) and distribute audio data to each encoder.
saki

pricsko, 2015-12-17:
Thank you very much, i will check it, and find out how to record audio for both :)

pricsko, 2016-01-13:
Hi,

I think You can close this issue too.
The solution is pretty simple : 
new MediaAudioEncoder(mMuxer_left, mMuxer_right, mMediaEncoderListener, mMediaEncoderListener2); (rest is obvious)

mMediaEncoderListener2 is just for getting the outputpath of the second muxer.

closed 2016-01-14 by saki4510t