saki4510t/UVCCamera #30: Background Recording without a preview

zakir224, 2015-12-19:
Hi

I wanted to ask if it is possible to record the frames in the background of an activity without having to show the preview.

saki4510t, 2015-12-19:
HI,
Theoritically it is possible, please check usbCameraTest4. It works as Service and can record in background.
FYI, app needs Activity to get permanent permission(this is Android specification). If you want to completely hide Activity, finish Activity before `setContentView` after start Service.
saki

zakir224, 2015-12-21:
Thanks for your answer. I've already figured out an way..

There is another issue. when i record the video with usb cam the video is recorded well but the audio is only noise..when i record without usb cam then the audio is also ok. Do you have any idea why its happening. t happened with your sample project 2 and 5 both.

saki4510t, 2015-12-27:
Hi,
Sorry my answer delayed. I assume system software of your device has something trouble on audio(like recent GALAXY series). Try to use other constant for `AudioRecord` instead of  `MediaRecorder.AudioSource.MIC`. Ex. `MediaRecorder.AudioSource.CAMCORDER` may help you.
saki

saki4510t, 2016-05-09:
closed due to inactivity.

closed 2016-05-09 by saki4510t