saki4510t/UVCCamera #31: Pause recording

pricsko, 2016-01-03:
Hi,

Is there any possibilities, to pause and restart record? (basically to append one video record to another video file)

Peter

saki4510t, 2016-01-04:
Hi,

Technically it is possible. but actual implementation depends on what result you want and how long do you want to pause etc.
One example is available on my repository, [TimeLapseRecordingSample](https://github.com/saki4510t/TimeLapseRecordingSample "TimeLapseRecordingSample"). The video source of  "TimeLapseRecordingSample" is internal camera but you can use by similar way.
If you want very long time pause(once per hours or longer), I assume saving a still image periodically (as jpeg or png) and making a movie after getting all images.
saki

pricsko, 2016-01-06:
HI,

Thank you very much, i'll check the TimeLapseRecording. Technically, i dont want to pause the record, i just want to start 2 recording at the same time, and decide on some conditions to which recording i want to append the current frame.

pricsko, 2016-01-07:
Hi,

i checked the TimeLapseRecording, and i am wondering, is it possible to record the camera frame (onPreviewFrame - with PreviewCallback) as mp4? i tried to change handle open, handle preview etc... in camerahandler and changed iframecallback to previewcallback but the recorded video's color is not correct. I think it is because onPreviewFrame gives NV21 format (tried to change previewformat, but it was even worse). how to decode properly the onpreviewframe with uvccamera's camerahandler and mediacodec?

pricsko, 2016-01-07:
basically what i tried : 
change byte[] data (from onpreviewframe) to mat in jni, like this : 
IplImage *pYuvImage = cvCreateImage(cvSize(frame->width, frame->height), IPL_DEPTH_8U, 2);
pYuvImage->imageData = (char *) frame->data;
cv::Mat _yuv = cv::cvarrToMat(pYuvImage);
cvReleaseImage(&pYuvImage);
(which works well with uvccamera's frame with iframecallback set to PIXEL_FORMAT_YUV420SP - with nv21 it was the same color error)
i also tried with yuyv2iyuv420SP but it was the same :(

saki4510t, 2016-01-07:
Hi,
I assume you may initialize MediaCodec encoder with wrong parameter(s).

If you want to encode video frame data using MediaCodec, you must use same pixel format for both MediaCodec and video source data(may come from IFrameCallback). Actual pixel format(s) that MediaCodec supports depend(s) on each device that you use. So you need to confirm on runtime and select the usable color format for both. All mediaCodec encoder on all sample projects are initialized for using `Surface` as video source now (because using Surface as video source is the fastest & best way to use MediaCodec as video encoder). 
saki

pricsko, 2016-01-07:
Hi,

i solved my problem. It was a known issue. Reference and solution here : http://stackoverflow.com/questions/34145713/how-to-convert-yv12-to-color-formatyuv420semiplanar

closed 2016-01-07 by saki4510t