saki4510t/UVCCamera #32: USBCameraTest3 just can display the first frame

YBJAY00000, 2016-01-06:
Hi, i tried the test0/1/2 is ok, but the test3 just can display the first frame,when i touch the record button, the app isn't responding(ANR).
### logcat:
01-01 03:06:39.075 30965-30965/com.serenegiant.usbcameratest3 D/dalvikvm: Late-enabling CheckJNI
01-01 03:06:39.105 30965-30965/com.serenegiant.usbcameratest3 V/MainActivity: onCreate:
01-01 03:06:39.215 30965-30965/com.serenegiant.usbcameratest3 V/CameraThread: getHandler:
01-01 03:06:39.215 30965-30965/com.serenegiant.usbcameratest3 V/MainActivity: onResume:
01-01 03:06:39.215 30965-30965/com.serenegiant.usbcameratest3 V/UVCCameraTextureView: onResume:
01-01 03:06:39.465 30965-30965/com.serenegiant.usbcameratest3 D/libEGL: loaded /system/lib/egl/libEGL_mali.so
01-01 03:06:39.465 30965-30965/com.serenegiant.usbcameratest3 D/libEGL: loaded /system/lib/egl/libGLESv1_CM_mali.so
01-01 03:06:39.470 30965-30965/com.serenegiant.usbcameratest3 D/libEGL: loaded /system/lib/egl/libGLESv2_mali.so
01-01 03:06:39.475 30965-30965/com.serenegiant.usbcameratest3 W/linker: libion.so has text relocations. This is wasting memory and is a security risk. Please fix.
01-01 03:06:39.525 30965-30965/com.serenegiant.usbcameratest3 D/OpenGLRenderer: Enabling debug mode 0
01-01 03:06:39.545 30965-30965/com.serenegiant.usbcameratest3 V/UVCCameraTextureView: onSurfaceTextureAvailable:android.graphics.SurfaceTexture@41b81f20
01-01 03:06:39.545 30965-30965/com.serenegiant.usbcameratest3 V/UVCCameraTextureView: RenderThread#getHandler:
01-01 03:06:39.545 30965-30998/com.serenegiant.usbcameratest3 D/UVCCameraTextureView: RenderThread started
01-01 03:06:39.545 30965-30998/com.serenegiant.usbcameratest3 V/UVCCameraTextureView: RenderThread#init:
01-01 03:06:39.545 30965-30998/com.serenegiant.usbcameratest3 V/EGLBase: EGLBase:
01-01 03:06:39.545 30965-30998/com.serenegiant.usbcameratest3 V/EGLBase: init:
01-01 03:06:39.545 30965-30998/com.serenegiant.usbcameratest3 D/EGLBase: EGLContext created, client version 2
01-01 03:06:39.545 30965-30998/com.serenegiant.usbcameratest3 V/EGLBase: makeDefault:
01-01 03:06:39.545 30965-30998/com.serenegiant.usbcameratest3 V/EGLBase: createFromSurface:
01-01 03:06:39.545 30965-30998/com.serenegiant.usbcameratest3 V/EGLBase: EglSurface:
01-01 03:06:39.545 30965-30998/com.serenegiant.usbcameratest3 V/EGLBase: createWindowSurface:nativeWindow=android.graphics.SurfaceTexture@41b81f20
01-01 03:06:39.565 30965-30998/com.serenegiant.usbcameratest3 V/GLDrawer2D: loadShader:
01-01 03:06:47.175 30965-30965/com.serenegiant.usbcameratest3 V/MainActivity: onConnect:
01-01 03:06:47.175 30965-30965/com.serenegiant.usbcameratest3 V/UVCCameraTextureView: getPreviewTexture:
01-01 03:06:47.180 30965-30990/com.serenegiant.usbcameratest3 V/CameraThread: handleOpen:
01-01 03:06:47.180 30965-30990/com.serenegiant.usbcameratest3 V/CameraThread: handleClose:
01-01 03:06:47.180 30965-30990/com.serenegiant.usbcameratest3 V/CameraThread: handleStopRecording:mMuxer=null
01-01 03:06:47.180 30965-30998/com.serenegiant.usbcameratest3 I/UVCCameraTextureView: RenderThread#updatePreviewSurface:
01-01 03:06:47.180 30965-30998/com.serenegiant.usbcameratest3 V/GLDrawer2D: initTex:
01-01 03:06:47.180 30965-30998/com.serenegiant.usbcameratest3 V/UVCCameraTextureView: getPreviewSurface:tex_id=1
01-01 03:06:47.180 30965-30990/com.serenegiant.usbcameratest3 D/dalvikvm: Trying to load lib /data/app-lib/com.serenegiant.usbcameratest3-1/libusb100.so 0x41b4cd18
01-01 03:06:47.180 30965-30990/com.serenegiant.usbcameratest3 D/dalvikvm: Added shared lib /data/app-lib/com.serenegiant.usbcameratest3-1/libusb100.so 0x41b4cd18
01-01 03:06:47.180 30965-30990/com.serenegiant.usbcameratest3 D/dalvikvm: No JNI_OnLoad found in /data/app-lib/com.serenegiant.usbcameratest3-1/libusb100.so 0x41b4cd18, skipping init
01-01 03:06:47.180 30965-30990/com.serenegiant.usbcameratest3 D/dalvikvm: Trying to load lib /data/app-lib/com.serenegiant.usbcameratest3-1/libuvc.so 0x41b4cd18
01-01 03:06:47.180 30965-30990/com.serenegiant.usbcameratest3 D/dalvikvm: Added shared lib /data/app-lib/com.serenegiant.usbcameratest3-1/libuvc.so 0x41b4cd18
01-01 03:06:47.185 30965-30990/com.serenegiant.usbcameratest3 D/dalvikvm: No JNI_OnLoad found in /data/app-lib/com.serenegiant.usbcameratest3-1/libuvc.so 0x41b4cd18, skipping init
01-01 03:06:47.185 30965-30990/com.serenegiant.usbcameratest3 D/dalvikvm: Trying to load lib /data/app-lib/com.serenegiant.usbcameratest3-1/libUVCCamera.so 0x41b4cd18
01-01 03:06:47.185 30965-30990/com.serenegiant.usbcameratest3 D/dalvikvm: Added shared lib /data/app-lib/com.serenegiant.usbcameratest3-1/libUVCCamera.so 0x41b4cd18
01-01 03:06:47.190 30965-30990/com.serenegiant.usbcameratest3 I/MainActivity: supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["640x480","160x120","176x144","320x176","320x240","352x288","432x240","544x288","640x360","752x416","800x448","800x600","864x480","960x544","960x720","1024x576","1184x656","1280x720","1280x960"]},{"index":2,"type":6,"default":1,"size":["640x480","160x120","176x144","320x176","320x240","352x288","432x240","544x288","640x360","752x416","800x448","800x600","864x480","960x544","960x720","1024x576","1184x656","1280x720","1280x960"]}]}
01-01 03:06:47.190 30965-30990/com.serenegiant.usbcameratest3 V/CameraThread: handleStartPreview:
01-01 03:06:47.255 30965-31130/com.serenegiant.usbcameratest3 I/libUVCCamera: [31130*UVCPreview.cpp:492:prepare_preview]:frameSize=(640,480)@MJPEG
01-01 03:06:47.265 30965-31132/com.serenegiant.usbcameratest3 I/libUVCCamera: [31132*UVCPreview.cpp:258:callbackPixelFormatChanged]:PIXEL_FORMAT_RAW:
01-01 03:06:49.805 30965-31131/com.serenegiant.usbcameratest3 W/libUVCCamera: [31131*UVCPreview.cpp:119:get_frame]:allocate new frame
01-01 03:06:49.805 30965-31130/com.serenegiant.usbcameratest3 W/libUVCCamera: [31130*UVCPreview.cpp:119:get_frame]:allocate new frame
01-01 03:06:49.890 30965-31131/com.serenegiant.usbcameratest3 W/libUVCCamera: [31131*UVCPreview.cpp:119:get_frame]:allocate new frame

saki4510t, 2016-01-06:
Hi,
Thank you for providing feedback.  As I tested on my devices, the sample projects works well and your log shows no problem. So could you provide more details?

If you modified something, could you try original one?
If you can see preview images until you touch record button, the issue may relate to MediaCodec.
Do you use old SAMSUNG devices like GALAXY S2/S3 etc.? Those devices have some issue on MediaCodec and will need to add some workaround.

If you un-filter logCat, you may be able to see additional log related to the issue.
saki

YBJAY00000, 2016-01-07:
thanks for reply, Exactly i test on 'samsung 4412', it seems to GALAXY  S3. i will test on the other platform (samsung 6818 /4418 or rockchip ..)

YBJAY00000, 2016-01-07:
Hi,
i tested on samsung 6818 device is ok(test3/7),but the rockchip device can't display anything..

saki4510t, 2016-01-07:
Hi,
Almost all devices which SOC/chipset is made by Rockchip, MediaTek and Allwinner have issues on their hardware/system software and app will freeze, crash, ANR, reboot... If you use those device I assume it will be better to give up to use UVC camera with those devices.
I had request them to fix the issues but there are no response.

I also have S3 and it's problem is different from others and related to MediaCodec. Some MediaCodec API is extremely slow(almost hung-up) on the device under some conditions...
Try to temporary increase thread priority to maximum while preparing MediaCodec.
Additionally, if you use Android 4.3 or prior, you need special handling when start MediaCodec because it never return `MediaCodec.INFO_OUTPUT_FORMAT_CHANGED` and use `MediaCodec.BUFFER_FLAG_CODEC_CONFIG` instead. And you need to extract PPS/SPS and contract MediaFormat instance and pass it to muxer by yourself.
saki

YBJAY00000, 2016-01-08:
Hi,saki
at first,thanks for your help so much!
i have a doubt,  whether must use the 'MediaCodec' when i use two cameras? because, i preview the second camera just by TextureView  there was an error 
> [1565*stream.c:1588:uvc_stream_start_bandwidth]:fail
> [1565*diag.c:88:uvc_perror]:failed start_streaming:I/O error (-1)

YBJAY00000, 2016-01-08:
i was wrong, I replaced the camera after is OK

closed 2016-02-17 by saki4510t