saki4510t/UVCCamera #33: onAttach(UsbDevice device)

dknee, 2016-01-06:
Hi Saki,

Once again, thanks for the great library and quick responses.
I was trying to implement the code in your cover page with the description "How to start app when user connect UVC device to Android device" and was not able to get the interface callback onAttach() to work. The onConnect() etc. works correctly. I noticed that I could not get the onAttach() toast window to show in any of your examples either. The code that I am designing is a derivative of your USBCameraTest7 and I am using a Acer B1-730HD tablet based on a dual core Atom processor. I am developing the app under Android Studio. Can you give me any ideas as to how to debug my issue?
Ad_thanks_vance.

saki4510t, 2016-01-07:
Hi,
Thank you for providing feedback.
As I know some old devices with old Android never send `attach` event to app defined broadcast receiver but I think most of ATOM devices will work. Does your app start when you connect UVC camera to your device? 

FYI,
1. When app defines intent filter in it's AndroidManifest.xml to request attach event so that app can get  permanent permission, Android system launches the app and send intent to main activity(instead of broadcast receiver) when user connects UVC device and gives permission.
2. When app register USBMonitor, USBMonitor registers it's broadcast receiver to Android system.
    As I mentioned, some old devices with old Android never send `attach` event to broadcast receiver and I don't add intent filter for 'attach' event to this broadcast receiver and use polling method instead to avoid race condition.  But you can try to add `attach` intent filter so that broadcast receiver listen the event by adding `filter.addAction(UsbManager.ACTION_USB_DEVICE_ATTACHED);` to `register` method.
3. USBMonitor post `mDeviceCheckRunnable` in `register` method to detect UVC device(s) periodically using polling method. This will cause race condition if you add intent filter for `attach` event and `onAttach` will be called multiple times(may be called on arbitrary thread). You need to access UVC device exclusively or remove either intent filter or `mDeviceCheckRunnable`.
4. `mDeviceCheckRunnable ` checks number of connected USB device(s) with applying device filter and calls `onAttach` if number of connect USB device increase. And then post itself for next check.

saki

dknee, 2016-01-10:
Hi Saki,
"Does your app start when you connect UVC camera to your device? " Yes this does happen,

I have not had a chance to try your other suggestions. I upgraded from my Acer tablet to a RCA Viking Pro 10.1 tablet since it has two USB ports. I have not had a chance to investigate the use of these ports since I can not get the Google USB driver to work on my PC with this tablet.

closed 2016-01-10 by dknee