saki4510t/UVCCamera #34: libusb_submit_transfer return -1

YBJAY00000, 2016-01-14:
hi,saki
i am trying 3 cameras and got this error in Stream.c (uvc_stream_start_bandwidth).
each camera i use MJPEG mode,just can display 2 cameras,the last one is error.

> stream.c:1576:uvc_stream_start_bandwidth]:ret(-1) != UVC_SUCCESS(0)==>libusb_submit_transfer failed
> stream.c:1591:uvc_stream_start_bandwidth]:fail

Another case, i use 2 camera as YUV, then have the same error, i don't know why(Please forgive me my c++ is so poor :) )?

saki4510t, 2016-01-14:
Hi,
I think it is just USB bandwidth is smaller than total amount of camera(s) requested.
Actual bandwidth depends on many factors like topology of connection, USB hardware of your device, quality of cable, camera, transfer mode, streaming mode, alternate settings(payload size of each frame/micro frame) ....
Theoretically, slower fps, smaller resolution can reduce bandwidth. And regarding a streaming mode, the transfer payload size is h264 < mjpeg << YUV (if the resolution and fps is same).

with best regards,
saki

YBJAY00000, 2016-01-14:
I understand, thanks for reply.

closed 2016-05-09 by saki4510t