saki4510t/UVCCamera #36: Not disabling autofocus

sreyasvpariyath, 2016-01-28:
I had tried to disable the autofocus by calling the Function setAutoFocus. I have passed false as parameter. But the camera is still getting focused automatically. How do I turn off the autofocus and How do I focus manually.

saki4510t, 2016-01-28:
Hi,
Does your camera support manual focus? Some cameras always use autofocus and some cameras are fixed focus. You can check `CTRL_FOCUS_AUTO` constant with `checkSupportFlag` method or setting `DEBUG` flag on and calling `updateCameraParams` method(the result are on logCat).

`nativeSetAutoFocus` method in UVCCamera#setAutoFocus returns zero? If the return value of  `nativeSetAutoFocus` is not zero, your camera may not support disabling autofocus.

saki

sreyasvpariyath, 2016-01-28:
Hey Thanks a Ton.! Everything is working fine. There was an issue at my end. Thanks a lot

closed 2016-01-28 by sreyasvpariyath

glenne, 2017-12-22:
I found that unless I first called updateCameraParams, autofocus would not get applied.  This was because the camera capabilities were never queried and defaulted to all zeros so when it got down to applying autofocus the code didn't even talk to the camera.