saki4510t/UVCCamera #37: Detects camera but doesn't display cameraview

trbLeeciN, 2016-02-01:
Hello, Thank you for great source. You are a life-saver (: 
I tried to run examples in 2 different devices, both of them detects uvc-camera, however one of them does not return view. Do you have any idea, how to fix this? 
Thank you

CAMOBAP795, 2016-07-29:
@trbLeeciN which device you used? possibly your because your ROM, usb kernel module doesn't support ISOC requests

amit13091992, 2017-03-01:
i have tried with lenovo(lollipop) and moto g4(marshmallow) both but its getting crashed after selecting the device.

codeinging, 2017-03-09:
I encounter the same problem.My Test Devices such as RedMi 1s.
The logcat:
03-09 18:36:09.895 7932-7932/com.serenegiant.usbcameratest0 I/View: Touch down dispatch to android.widget.Button{422385e0 VFED..C. ........ 414,0-620,96 #1020019 android:id/button1}, event = MotionEvent { action=ACTION_DOWN, id[0]=0, x[0]=102.21362, y[0]=79.316956, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, metaState=0, flags=0x0, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=14451903, downTime=14451903, deviceId=3, source=0x1002 }
03-09 18:36:09.914 7932-7932/com.serenegiant.usbcameratest0 D/GraphicBuffer: create handle(0x63d0f850) (w:688, h:484, f:1)
03-09 18:36:09.918 7932-7932/com.serenegiant.usbcameratest0 D/OpenGLRenderer: prepareDirty (0.00, 0.00, 684.00, 484.00) opaque 0 <0x63d0f460>
03-09 18:36:09.919 7932-7932/com.serenegiant.usbcameratest0 D/OpenGLRenderer: clear (0.00, 484 - 484.00 = 0.00, 684.00 - 0.00 = 684.00, 484.00 - 0.00 = 484.00) opaque 0 <0x63d0f460>
03-09 18:36:09.923 7932-7932/com.serenegiant.usbcameratest0 D/OpenGLRenderer: finish <0x63d0f460>
03-09 18:36:09.973 7932-7932/com.serenegiant.usbcameratest0 I/View: Touch up dispatch to android.widget.Button{422385e0 VFED..C. ...P.... 414,0-620,96 #1020019 android:id/button1}, event = MotionEvent { action=ACTION_UP, id[0]=0, x[0]=102.21362, y[0]=79.316956, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, metaState=0, flags=0x0, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=14451983, downTime=14451903, deviceId=3, source=0x1002 }
03-09 18:36:09.974 7932-7932/com.serenegiant.usbcameratest0 V/Provider/Settings: get setting for user 0 by user 0 so skipping cache
03-09 18:36:09.974 7932-7932/com.serenegiant.usbcameratest0 V/Provider/Settings:  from settings cache , name = sound_effects_enabled , value = 0
03-09 18:36:09.978 7932-7932/com.serenegiant.usbcameratest0 D/GraphicBuffer: create handle(0x61debfa8) (w:688, h:484, f:1)
03-09 18:36:09.982 7932-7932/com.serenegiant.usbcameratest0 D/OpenGLRenderer: prepareDirty (0.00, 0.00, 684.00, 484.00) opaque 0 <0x63d0f460>
03-09 18:36:09.982 7932-7932/com.serenegiant.usbcameratest0 D/OpenGLRenderer: clear (0.00, 484 - 484.00 = 0.00, 684.00 - 0.00 = 684.00, 484.00 - 0.00 = 484.00) opaque 0 <0x63d0f460>
03-09 18:36:09.984 7932-7932/com.serenegiant.usbcameratest0 D/OpenGLRenderer: finish <0x63d0f460>
03-09 18:36:10.008 7932-7932/com.serenegiant.usbcameratest0 D/OpenGLRenderer: Flushing caches (mode 0)
03-09 18:36:10.009 7932-7932/com.serenegiant.usbcameratest0 D/GraphicBuffer: close handle(0x63d0f360) (w:688 h:484 f:1)
03-09 18:36:10.010 7932-7932/com.serenegiant.usbcameratest0 D/GraphicBuffer: close handle(0x63d39ee0) (w:688 h:484 f:1)
03-09 18:36:10.011 7932-7932/com.serenegiant.usbcameratest0 D/GraphicBuffer: close handle(0x63d0f850) (w:688 h:484 f:1)
03-09 18:36:10.012 7932-7932/com.serenegiant.usbcameratest0 D/GraphicBuffer: close handle(0x61debfa8) (w:688 h:484 f:1)
03-09 18:36:10.022 7932-7932/com.serenegiant.usbcameratest0 D/ActivityThread: ACT-AM_ON_PAUSE_CALLED ActivityRecord{41fadc30 token=android.os.BinderProxy@41fad290 {com.serenegiant.usbcameratest0/com.serenegiant.usbcameratest0.MainActivity}}
03-09 18:36:10.039 7932-7932/com.serenegiant.usbcameratest0 D/ActivityThread: ACT-PAUSE_ACTIVITY handled : 1 / android.os.BinderProxy@41fad290
03-09 18:36:10.414 7932-7932/com.serenegiant.usbcameratest0 D/ActivityThread: ACT-STOP_ACTIVITY_SHOW handled : 0 / android.os.BinderProxy@41fad290
03-09 18:36:17.639 7932-7932/com.serenegiant.usbcameratest0 V/Provider/Settings: get setting for user 0 by user 0 so skipping cache
03-09 18:36:17.639 7932-7932/com.serenegiant.usbcameratest0 V/Provider/Settings:  from settings cache , name = access_control_lock_enabled , value = null
03-09 18:36:17.640 7932-7932/com.serenegiant.usbcameratest0 D/ActivityThread: ACT-AM_ON_RESUME_CALLED ActivityRecord{41fadc30 token=android.os.BinderProxy@41fad290 {com.serenegiant.usbcameratest0/com.serenegiant.usbcameratest0.MainActivity}}
03-09 18:36:17.642 7932-7932/com.serenegiant.usbcameratest0 V/PhoneWindow: DecorView setVisiblity: visibility = 0 ,Parent =ViewRoot{421b9618 com.serenegiant.usbcameratest0/com.serenegiant.usbcameratest0.MainActivity,ident = 0}, this =com.android.internal.policy.impl.PhoneWindow$DecorView{42042b48 V.E..... R....... 0,0-720,1280}
03-09 18:36:17.660 7932-7932/com.serenegiant.usbcameratest0 D/ActivityThread: ACT-RESUME_ACTIVITY handled : 0 / android.os.BinderProxy@41fad290
03-09 18:36:17.660 7932-7932/com.serenegiant.usbcameratest0 V/InputMethodManager: onWindowFocus: null softInputMode=32 first=false flags=#41810100
03-09 18:36:17.664 7932-7932/com.serenegiant.usbcameratest0 I/SurfaceView: updateWindow -- OnPreDrawListener, mHaveFrame = true
03-09 18:36:17.666 7932-7932/com.serenegiant.usbcameratest0 D/OpenGLRenderer: prepareDirty (0.00, 0.00, 720.00, 1280.00) opaque 1 <0x6195acf8>
03-09 18:36:17.667 7932-7932/com.serenegiant.usbcameratest0 D/OpenGLRenderer: finish <0x6195acf8>
03-09 18:36:17.690 7932-7956/com.serenegiant.usbcameratest0 V/USBMonitor: processConnect:device=UsbDevice[mName=/dev/bus/usb/001/028,mVendorId=7758,mProductId=265,mClass=239,mSubclass=2,mProtocol=1,mInterfaces=[Landroid.os.Parcelable;@422d9508]
03-09 18:36:17.696 7932-7956/com.serenegiant.usbcameratest0 I/dalvikvm: Could not find method android.hardware.usb.UsbInterface.getAlternateSetting, referenced from method com.serenegiant.usb.USBMonitor$UsbControlBlock.getInterface
[... cut ...]
03-09 18:36:17.736 7932-7954/com.serenegiant.usbcameratest0 I/libusb/usbfs: [7954*android_usbfs.c:512:op_init2]:call android_scan_devices
03-09 18:36:17.738 7932-7954/com.serenegiant.usbcameratest0 I/MainActivity: supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["640x480","352x288","320x240","176x144","160x120"]}]}
03-09 18:36:17.757 7932-8119/com.serenegiant.usbcameratest0 I/libUVCCamera: [8119*UVCPreview.cpp:492:prepare_preview]:frameSize=(640,480)@YUYV
03-09 18:36:18.415 7932-8127/com.serenegiant.usbcameratest0 D/dalvikvm: threadid=15: interp stack at 0x63f06000
Thanks

piskorzj, 2017-07-21:
Hello, I think I have similar issue.
Environment: Samsung S5, LineageOS 14.1 (Nougat).

First I had exceptions and crashes but I was able to resolve them by downgrading NDK to suggested 14b version.

Camera is detected correctly (it shows entry on popup, and also prints info about available frames in logcat). But there is no preview, tried all test apps.

Here is output from UsbCameraTest0:
```
07-21 12:44:21.526 2607-2607/? I/art: Late-enabling -Xcheck:jni
07-21 12:44:21.591 2607-2607/com.serenegiant.usbcameratest0 I/art: Starting a blocking GC AddRemoveAppImageSpace
07-21 12:44:21.683 2607-2607/com.serenegiant.usbcameratest0 V/MainActivity: onStart:
07-21 12:44:21.730 2607-2630/com.serenegiant.usbcameratest0 I/Adreno-EGL: <qeglDrvAPI_eglInitialize:379>: EGL 1.4 QUALCOMM build: Nondeterministic_AU_msm8974_LA.BF.1.1.3_RB1__release_AU (Ia6c73e7530)
                                                                          OpenGL ES Shader Compiler Version: E031.29.00.00
                                                                          Build Date: 12/04/15 Fri
                                                                          Local Branch: mybranch17080070
                                                                          Remote Branch: quic/LA.BF.1.1.3_rb1.5
                                                                          Local Patches: NONE
                                                                          Reconstruct Branch: NOTHING
07-21 12:44:21.732 2607-2630/com.serenegiant.usbcameratest0 I/OpenGLRenderer: Initialized EGL, version 1.4
07-21 12:44:21.732 2607-2630/com.serenegiant.usbcameratest0 D/OpenGLRenderer: Swap behavior 1
07-21 12:44:21.754 2607-2607/com.serenegiant.usbcameratest0 V/MainActivity: surfaceCreated:
07-21 12:44:21.754 2607-2607/com.serenegiant.usbcameratest0 V/MainActivity: surfaceChanged:
07-21 12:44:42.730 2607-2629/com.serenegiant.usbcameratest0 V/MainActivity: onAttach:
07-21 12:44:54.439 2607-2629/com.serenegiant.usbcameratest0 D/UsbDeviceConnectionJNI: close
07-21 12:44:54.439 2607-2629/com.serenegiant.usbcameratest0 I/USBMonitor: name=/dev/bus/usb/001/002,desc=61,busnum=1,devnum=2,rawDesc=[B@aa19691
07-21 12:44:54.440 2607-2629/com.serenegiant.usbcameratest0 V/MainActivity: onConnect:
07-21 12:44:54.454 2607-2623/com.serenegiant.usbcameratest0 D/UsbDeviceConnectionJNI: close
07-21 12:44:54.454 2607-2623/com.serenegiant.usbcameratest0 I/libusb/core: [2623*core.c:2085:libusb_init2]:libusb v1.0.19.10903
07-21 12:44:54.454 2607-2623/com.serenegiant.usbcameratest0 I/libusb/usbfs: [2623*android_usbfs.c:504:op_init2]:start up hotplug event handler
07-21 12:44:54.454 2607-2623/com.serenegiant.usbcameratest0 I/libusb/usbfs: [2623*android_usbfs.c:512:op_init2]:call android_scan_devices
07-21 12:44:54.455 2607-2623/com.serenegiant.usbcameratest0 I/MainActivity: supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["1280x720","1920x1080","2304x1296","2304x1536","640x480"]},{"index":2,"type":6,"default":1,"size":["1280x720","1920x1080","2304x1296","2304x1536","640x480"]}]}
07-21 12:44:54.458 2607-2623/com.serenegiant.usbcameratest0 D/UsbDeviceConnectionJNI: close
07-21 12:44:54.458 2607-2623/com.serenegiant.usbcameratest0 V/MainActivity: onDisconnect:
07-21 12:45:30.085 2607-2629/com.serenegiant.usbcameratest0 D/UsbDeviceConnectionJNI: close
07-21 12:45:30.085 2607-2629/com.serenegiant.usbcameratest0 I/USBMonitor: name=/dev/bus/usb/001/002,desc=62,busnum=1,devnum=2,rawDesc=[B@84786c9
07-21 12:45:30.086 2607-2629/com.serenegiant.usbcameratest0 V/MainActivity: onConnect:
07-21 12:45:30.089 2607-2623/com.serenegiant.usbcameratest0 D/UsbDeviceConnectionJNI: close
07-21 12:45:30.090 2607-2623/com.serenegiant.usbcameratest0 I/libusb/core: [2623*core.c:2085:libusb_init2]:libusb v1.0.19.10903
07-21 12:45:30.090 2607-2623/com.serenegiant.usbcameratest0 I/libusb/usbfs: [2623*android_usbfs.c:504:op_init2]:start up hotplug event handler
07-21 12:45:30.090 2607-2623/com.serenegiant.usbcameratest0 I/libusb/usbfs: [2623*android_usbfs.c:512:op_init2]:call android_scan_devices
07-21 12:45:30.092 2607-2623/com.serenegiant.usbcameratest0 I/MainActivity: supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["1280x720","1920x1080","2304x1296","2304x1536","640x480"]},{"index":2,"type":6,"default":1,"size":["1280x720","1920x1080","2304x1296","2304x1536","640x480"]}]}
07-21 12:45:30.094 2607-2623/com.serenegiant.usbcameratest0 D/UsbDeviceConnectionJNI: close
07-21 12:45:30.094 2607-2623/com.serenegiant.usbcameratest0 V/MainActivity: onDisconnect:
07-21 12:45:36.076 2607-2607/com.serenegiant.usbcameratest0 V/MainActivity: surfaceDestroyed:
07-21 12:45:36.327 2607-2607/com.serenegiant.usbcameratest0 V/MainActivity: onStop:
07-21 12:45:36.328 2607-2607/com.serenegiant.usbcameratest0 V/MainActivity: onDestroy:
07-21 12:49:02.769 2607-2607/com.serenegiant.usbcameratest0 V/MainActivity: onStart:
07-21 12:49:02.812 2607-2607/com.serenegiant.usbcameratest0 V/MainActivity: surfaceCreated:
07-21 12:49:02.812 2607-2607/com.serenegiant.usbcameratest0 V/MainActivity: surfaceChanged:
07-21 12:49:03.775 2607-3865/com.serenegiant.usbcameratest0 V/MainActivity: onAttach:
07-21 12:49:14.741 2607-3865/com.serenegiant.usbcameratest0 D/UsbDeviceConnectionJNI: close
07-21 12:49:14.742 2607-3865/com.serenegiant.usbcameratest0 I/USBMonitor: name=/dev/bus/usb/001/003,desc=65,busnum=1,devnum=3,rawDesc=[B@f305bc1
[... cut ...]
07-21 13:01:58.149 2607-2607/com.serenegiant.usbcameratest0 V/MainActivity: onStart:
07-21 13:01:59.156 2607-7687/com.serenegiant.usbcameratest0 V/MainActivity: onAttach:
07-21 13:03:02.285 2607-2607/com.serenegiant.usbcameratest0 V/MainActivity: onStop:
```
As you can see, there is no exception or something which could shed some light on an issue. But it is not showing preview.

I'm sure phone is working, is USB-OTG, camera is working, is OVC and is capable of video streaming because I'm investigating other options also, for example by using v4l2.

piskorzj, 2017-07-21:
Ok, I've got it working. Examples have hard-coded frame sizes and also frame rates. My camera is only supporting 60fps in MJPEG mode, so it was just silently ignored after detecting.

Here is what fixed it:

```
--- a/usbCameraTest0/src/main/java/com/serenegiant/usbcameratest0/MainActivity.java
+++ b/usbCameraTest0/src/main/java/com/serenegiant/usbcameratest0/MainActivity.java
@@ -149,7 +149,7 @@ public class MainActivity extends BaseActivity implements CameraDialog.CameraDia
                                                camera.open(ctrlBlock);
                                                if (DEBUG) Log.i(TAG, "supportedSize:" + camera.getSupportedSize());
                                                try {
-                                                       camera.setPreviewSize(UVCCamera.DEFAULT_PREVIEW_WIDTH, UVCCamera.DEFAULT_PREVIEW_HEIGHT, UVCCamera.FRAME_FORMAT_MJPEG);
+                                                       camera.setPreviewSize(1920, 1080, 1, 61, UVCCamera.FRAME_FORMAT_MJPEG, 1.0f);
                                                } catch (final IllegalArgumentException e) {
                                                        try {
                                                                // fallback to YUV mode
(END)
```

sappho192, 2017-07-22:
Check this code change in the pull request [https://github.com/saki4510t/UVCCamera/pull/175](https://github.com/saki4510t/UVCCamera/pull/175)
The code gets supported resolution.

abhishekkumar1902, 2017-08-29:
I am running usbCameraTest0 project sample and find that for the first time it runs fine, and now when it is running, if you detach the camera, and plug it again, scan for the attached camera and open it, so the app does not display anything. Any idea how to resolve this?