saki4510t/UVCCamera #38: Support for RockChip RK3288

dknee, 2016-02-03:
Hi Saki,

I had read in your other web comments that your UVC camera software was not compatible with the RockChip RK8233. Have you made any progress with fixing this issue?

 I have recently purchased a PIPO 7 tablet which is great for debugging/developing USB host code on an Android tablet since this device has separate micro USB connectors for both HOST and DEVICE plus also has a separate charging plug. However, as you know, I only get black images from my external stereo USB Webcams. 

Actually for the code that I am developing I only need bulk transfer still images from each of the cameras and I process these images, in a tight loop, with OpenCV. Do you know a way I could get the images from my two Webcams (Minoru) using bulk transfer and bypassing the UVC - RK8233 issues?

saki4510t, 2016-02-04:
Hi,
As I know there is no way to use isochronous transfer endpoint as bulk transfer endpoint because they have physically different protocl and Minoru stereo camera only supports isochronous transfer. So...
I had requested Mediatek etc. to fix the issue but there is no response. I assume they don't want to fix or don't want user to use USB devices that need isochronous transfer like camera, DAC etc.

FYI, some Android devices(Android TV) that use MediaTeck, Rockchip etc. can use external camera(connected over USB) as internal camera(like a emulator on PC).
saki

saki4510t, 2016-06-23:
closed due to inactivity.

closed 2016-06-23 by saki4510t