saki4510t/UVCCamera #39: Epson moverio bt-200 could not open camera!

saeed-discovery, 2016-02-04:
Hi,

Thanks for your efforts!
I tried to test "_usbCameraTest_" and "_usbCameraTest0_" and they ran successfully in my phone (Samsung galaxy note 3 / android 5.0) but when i try to run or compile them in my _Epson Moverio BT-200_ glasses it can find the camera but fails with "_**could not open camera:err=-99**_" error!
I tried using different parameters for width, height, fps etc but nothing happened... :(
any help or comment on how to solve this issue would be greatly appreciated!
th

saeed-discovery, 2016-02-10:
As no one answered yet! This problem has fixed!
If you connect the camera via a USB hub, BT-200 can recognize it correctly!!
Now it connects to the camera with no error and the camera LED turns on.
But still no image!
Everything is OK but no image is displaying!! even the "onButton" method in "usbcameratest" sample works when I press the camera button.

Any comments on how to solve this issue?

saki4510t, 2016-02-11:
Hi,
Sorry my answer delayed.
Well I don't have BT-200 and I don't understand what camera do you use and how you connect it to BT-200.

As I can see informations about BT-200,
- Android4.0.4(latest firmware), may be API level15
- OMAP chipset (I couldn't find more information)
- kernel will be 3.0.21
- headset has 30 megapixel camera
- have micro USB connector (As I can see on web, expected use case of this connect is client, connect to PC and copy data).  

I assume you are trying to connect your camera to micro USB connector via SUB hub(self powered or bus powered?). Your test result shows BT-200 seems to be supports USB host function.  I'm not sure BT-200 supports isochronous transfer mode (that web camera needs to transfer video images), but I assume starting isochronous transfer will fail; `libusb_submit_transfer` will return error in  `uvc_stream_start_bandwidth` function in `stream.c` in `libuvc`. If so there are lot of expected reason.

- bandwidth issue, this will be related to what resolution/what camera you try to use, SOC and settings of kernel module
- isochronous transfer mode is disabled in kernel module
- SOC does not support isochronous transfer mode
- app can not detach kernel module. kernel may statically link USB related module instead on using kernel module.
- USB current issue, using self powered USB2.0 hub may solve this.
- kernel module may have something limit to protect hardware related to USB current.

As I can see web site of Epson, you will able to get kernel source code(because it is open source under GPL) and you will confirm how they implements in kernel(module) if the limitation come from kernel/kernel module.

saki

saeed-discovery, 2016-02-13:
Thanks for the reply
I think there is no problem related to the kernel because I tested another app ("USB Camera App" from http://www.infinitegra.co.jp/en/solution/AndroidApp1.htm ) and it worked!
However it freezes every few seconds!!

I am using A4tech PK-710g webcam ( http://www.a4tech.com/hproduct.asp?m=&cid=77&scid=224&id=884 ) and the connection to BT-200 is via a bus-powered hub and an OTG cable to the BT-200 micro USB port.

Using a self powered hub didn't solve the issue but changing the hub causes some effects, e.g. using a shorter cable for the hub reduces the freezing rate and vice versa in "USB Camera App" but still no image in UVCCamera ...
So I'm wondering why there is no image while there are no errors and camera button works?
Where can I trace the video transferring and showing procedure? and how can I test if there is a problem in receiving images or showing them?

Thanks again

saki4510t, 2016-02-14:
Hi,
> So I'm wondering why there is no image while there are no errors and camera button works?
I may misunderstand what you say. Do you mean still capturing and/or movie capturing work?
If so, USB communication itself have no problem  (there may be still exist signal error or power issue if other app is unstable) and the problem will be rendering.
Did you see any error on logCat(without applying any log filter)? If you can, could you provide a log?
saki

saki4510t, 2016-06-23:
closed due to inactivity.

closed 2016-06-23 by saki4510t