saki4510t/UVCCamera #41: USBCameraTest3 - How to make video files streamable?

PinkJazz, 2016-02-11:
I am currently writing a modified version of USBCameraTest3 with playback, and would like to use a VideoView for playback.  However, today I just found out that the created video files are non-streamable, as I get the error "Can't play this video" when loading the video into my VideoView.  Apparently VideoView requires streamable video files.

I was wondering, what changes should I make to make my output video files streamable?

saki4510t, 2016-02-13:
Hi,
I'm not sure what actually you want to do and how you do it.
If you try to playback from local storage after capturing and finish saving video images to local storage,  I assume you will misunderstand how to use VideoView and use wrong way to initialize, load and start playback. 
Because all VideoView, MediaPlayer and MediaCodec decoder can playback regardless the movie file is streamable or not if it is on local storage.
I assume most possible reason is you use wrong path, url and/or you will call `VideoView#start` just after calling `VideoView#setVideoPath` etc. without waiting to finish preparing.

FYI, if you really need streamable format, you need to convert it by yourself after saving movie file(s) 
using FFMPEG etc.

saki

PinkJazz, 2016-02-13:
FYI, it looks like I got the videos to work using ExoMedia, a VideoView-style wrapper for ExoPlayer.  Thanks for your response anyway.

saki4510t, 2016-02-15:
Hi,
As I mentioned before, I'm not sure how you try to do it, but as I tested, VideoView can play generated movie files well on all my devices(from Android4.1.2-Android6). Did you try to play video over network?
saki

saki4510t, 2016-05-09:
closed due to inactivity.

closed 2016-05-09 by saki4510t