saki4510t/UVCCamera #42: Random could not open camera with errors -2 and -50

btut, 2016-02-17:
Hi saki,
I am having problems with a camera of mine. Sometimes it works fine, other times not (error message: could not open camera:err=-2 or -50). I cannot distinguish any dependencies on other circumstances, sometimes the error is thrown, other times the preview is ok.
Information: I am using an edited version of sample 7, but the error -2 shows up in an unchanged version of sample 7 as well.
I can not figure out what the errors mean, do you have an idea?

Also, in my version I do not want any user interface. If a camera is connected, the preview should start on any free surface (max 2 at a time). However I noticed that in the onAttach Method of the OnDeviceConnectListener Interface I implemented, I always get device == null. I have to iterate through all connected devices with:
`for(UsbDevice d : mUSBMonitor.getDeviceList())`
and request permission for all of them. Do you have an idea why the method is not called with the device that was most recently connected but with null instead?

Thanks in advance for your help!

saki4510t, 2016-02-18:
Hi,
`err=-2 or -50` seems you may initialize camera with wrong parameter or app fail to load descriptor or your camera does not meet UVC specification/requirements.

As you can see, `USBMonitor` call `onAttach` callback method at two condition.

1. when Android system send broadcast and `BroadcastReceiver` of `USBMonitor` receives `UsbManager.ACTION_USB_DEVICE_ATTACHED` action.  
    In this case, you can receive correct `USBDevice` instance at `onAttach` callback. Unfortunately some devices never broadcast this action and USBMonitor run polling task at the same time.
2. when polling task detects the number of connected device(s) are changed.  
    In this case you can receive `null` on `onAction` method.

As I can see your comment, you will want to start app and preview when user connects camera and I assume you will need to handle several conditions. In all case, you need to know and keep which UVC device you already handled and your are handling now if you want to use multiple camera at the same time.

1. App starts by Android system when user connect UVC device.  
    As I wrote before,  Android system starts app with Intent which action is `UsbManager.ACTION_USB_DEVICE_ATTACHED` with `UsbDevice` as extra of `UsbManager.EXTRA_DEVICE` key when app has Intent filter on it's `AndroidManifest.xml`.
    In this case Android guarantees that app has permission for the UVC device because app has permanent permission or user permit just before app starting when connecting UVC device.  
    Unfortunately UsbMonitor can not receive this intent. So you need to use ether way;

    1. hold `UsbDevice` came from this intent and use it when `onResume` is called or after creating Surface/SurfaceTexture for preview,
    2. ignore this intent and handle everything when `onAttach` callback is called (with null) from polling task.
2. App start by user regardless connection of UVC device  
    You need to handle `onAttach` callback.
3. User connects UVC device while app running.  
    In this case, Android system will show permission dialog and app will lost focus when you add intent filter of `UsbManager.ACTION_USB_DEVICE_ATTACHED` on `AndroidManifest.xml` and your app does not have permission for the device. `onAttach` will be called with actual `UsbDevice` after getting permission.  
    If your app does not have intent filter of `UsbManager.ACTION_USB_DEVICE_ATTACHED` on `AndroidManifest.xml`, polling task will detect and call `onAttach` callback with null.

saki

btut, 2016-02-18:
Hi,
thanks for your fast answer. I went for the second method you pointed out, ignored the intent and requested the permission when onAttach is called. The app works fine and it starts when I connect an UVC device, showing the previews fine. But sometimes I don't get any preview or only one (I usually connect two cameras) and get err -2 or -50. Since it works sometimes I do not think it is due to wrong parameters, but I would like to check. What parameters do you mean exactly and where can I change them?
Thanks
Ben

saki4510t, 2016-02-20:
Hi,
It depends on where/when the error occurs.

As I can see your comment, I assume your issue seems to relate to timing of initialization and hardware of your device/camera. Did you try to initialize camera manually one by one? If manual initialization works well on both camera, it may be better to start camera sequentially, one by one using semaphore, synchronize etc. Because initialization and starting preview run on `CameraHander`, simultaneously on worker thread.

I'm not sure your actual implementation, did you start camera after creation of `SurfaceTexture`?
`usbcameratest7` is just sample and primarily intended to start camera by manual. And most of error checks are omitted from it's code.
saki

btut, 2016-02-27:
Manual initialisation doesn't help. I tried with the other samples as well, and the errors still occur. CameraTest0 even crashes most of the time with the following error log (unfiltered version below):

`02-27 13:58:08.097 11702-11702/? E/Zygote: MountEmulatedStorage()
02-27 13:58:08.097 11702-11702/? E/Zygote: v2
02-27 13:58:08.097 11702-11702/? I/libpersona: KNOX_SDCARD checking this for 10125
02-27 13:58:08.097 11702-11702/? I/libpersona: KNOX_SDCARD not a persona
02-27 13:58:08.117 11702-11702/? I/SELinux: Function: selinux_compare_spd_ram, SPD-policy is existed. and_ver=SEPF_SM-G900F_5.0 ver=27
02-27 13:58:08.117 11702-11702/? I/SELinux: Function: selinux_compare_spd_ram , priority [2] , priority version is VE=SEPF_SM-G900F_5.0-1_0032
02-27 13:58:08.117 11702-11702/? E/SELinux: [DEBUG] get_category: variable seinfo: default sensitivity: NULL, cateogry: NULL
02-27 13:58:08.117 11702-11702/? I/art: Late-enabling -Xcheck:jni
02-27 13:58:08.147 11702-11702/? D/TimaKeyStoreProvider: TimaSignature is unavailable
02-27 13:58:08.147 11702-11702/? D/ActivityThread: Added TimaKeyStore provider
02-27 13:58:08.187 11702-11702/com.serenegiant.usbcameratest0 D/ResourcesManager: creating new AssetManager and set to /data/app/com.serenegiant.usbcameratest0-1/base.apk
02-27 13:58:08.217 11702-11722/com.serenegiant.usbcameratest0 V/BitmapFactory: DecodeImagePath(decodeResourceStream3) : res/drawable-xxhdpi-v4/sym_def_app_icon.png
02-27 13:58:08.217 11702-11724/com.serenegiant.usbcameratest0 V/BitmapFactory: DecodeImagePath(decodeResourceStream3) : res/drawable-hdpi-v4/ic_launcher.png
02-27 13:58:08.217 11702-11723/com.serenegiant.usbcameratest0 V/BitmapFactory: DecodeImagePath(decodeResourceStream3) : res/drawable-xxhdpi-v4/ic_menu_camera.png
02-27 13:58:08.227 11702-11702/com.serenegiant.usbcameratest0 V/BitmapFactory: DecodeImagePath(decodeResourceStream3) : res/drawable-xxhdpi-v4/ic_ab_back_holo_dark_am.png
02-27 13:58:08.237 11702-11702/com.serenegiant.usbcameratest0 V/BitmapFactory: DecodeImagePath(decodeResourceStream3) : res/drawable-xxhdpi-v4/sym_def_app_icon.png
02-27 13:58:08.237 11702-11702/com.serenegiant.usbcameratest0 D/Activity: performCreate Call secproduct feature valuefalse
02-27 13:58:08.237 11702-11702/com.serenegiant.usbcameratest0 D/Activity: performCreate Call debug elastic valuetrue
02-27 13:58:08.247 11702-11702/com.serenegiant.usbcameratest0 V/MainActivity: onResume:
02-27 13:58:08.247 11702-11725/com.serenegiant.usbcameratest0 D/OpenGLRenderer: Render dirty regions requested: true
02-27 13:58:08.277 11702-11725/com.serenegiant.usbcameratest0 I/Adreno-EGL: <qeglDrvAPI_eglInitialize:410>: EGL 1.4 QUALCOMM build:  ()
                                                                            OpenGL ES Shader Compiler Version: E031.25.01.03
                                                                            Build Date: 03/03/15 Tue
                                                                            Local Branch: LA.BF.1.1_RB1_20150108_025_1077123_1158499
                                                                            Remote Branch: 
                                                                            Local Patches: 
                                                                            Reconstruct Branch: 
02-27 13:58:08.277 11702-11725/com.serenegiant.usbcameratest0 I/OpenGLRenderer: Initialized EGL, version 1.4
02-27 13:58:08.297 11702-11725/com.serenegiant.usbcameratest0 I/OpenGLRenderer: HWUI protection enabled for context ,  &this =0xb3b090d8 ,&mEglDisplay = 1 , &mEglConfig = 8 
02-27 13:58:08.297 11702-11725/com.serenegiant.usbcameratest0 D/OpenGLRenderer: Enabling debug mode 0
02-27 13:58:08.307 11702-11702/com.serenegiant.usbcameratest0 V/MainActivity: surfaceCreated:
02-27 13:58:08.307 11702-11702/com.serenegiant.usbcameratest0 V/MainActivity: surfaceChanged:
02-27 13:58:08.367 11702-11702/com.serenegiant.usbcameratest0 I/Timeline: Timeline: Activity_idle id: android.os.BinderProxy@19e8b7dd time:365664
02-27 13:58:14.537 11702-11702/com.serenegiant.usbcameratest0 D/ViewRootImpl: ViewPostImeInputStage ACTION_DOWN
02-27 13:58:15.247 11702-11702/com.serenegiant.usbcameratest0 D/ViewRootImpl: ViewPostImeInputStage ACTION_DOWN
02-27 13:58:15.297 11702-11702/com.serenegiant.usbcameratest0 E/shsh:  810 12 48 1032
02-27 13:58:15.297 11702-11702/com.serenegiant.usbcameratest0 D/AbsListView: Get MotionRecognitionManager
02-27 13:58:15.947 11702-11702/com.serenegiant.usbcameratest0 D/ViewRootImpl: ViewPostImeInputStage ACTION_DOWN
02-27 13:58:16.107 11702-11702/com.serenegiant.usbcameratest0 E/ViewRootImpl: sendUserActionEvent() mView == null
02-27 13:58:16.617 11702-11702/com.serenegiant.usbcameratest0 D/ViewRootImpl: ViewPostImeInputStage ACTION_DOWN
02-27 13:58:16.747 11702-11702/com.serenegiant.usbcameratest0 V/MainActivity: onConnect:
02-27 13:58:16.747 11702-11702/com.serenegiant.usbcameratest0 E/ViewRootImpl: sendUserActionEvent() mView == null
02-27 13:58:16.767 11702-11860/com.serenegiant.usbcameratest0 I/MainActivity: supportedSize:{"formats":[{"index":1,"type":4,"default":1,"size":["640x480","1920x1080","1280x720","2592x1944"]},{"index":2,"type":6,"default":1,"size":["640x480","1920x1080","1280x720","2592x1944"]}]}
02-27 13:58:16.767 11702-11860/com.serenegiant.usbcameratest0 A/art: art/runtime/runtime.cc:1110] Thread[13,tid=11860,Native,Thread*=0xb4f11c00,peer=0x12d3db20,"pool-1-thread-1"] attempting to detach while still running code
[... cut ...]
02-27 13:58:17.687 1176-4078/? D/LockPatternUtilsCache: getCarrierLockPlusMode()
02-27 13:58:17.687 1176-4078/? D/LockPatternUtilsCache: value : false
02-27 13:58:17.687 1176-4107/? D/LockPatternUtilsCache: getCarrierLockPlusMode()
02-27 13:58:17.687 1176-4107/? D/LockPatternUtilsCache: value : false
02-27 13:58:17.687 1176-1646/? D/LockPatternUtilsCache: getCarrierLockPlusMode()
02-27 13:58:17.687 1176-1646/? D/LockPatternUtilsCache: value : false
02-27 13:58:17.687 1176-5049/? D/LockPatternUtilsCache: getCarrierLockPlus

btut, 2016-03-01:
Well I found out the -50 errors are due to missing interfaces. The USB Device sometimes returns getInterfaceCount == 0. Fixed this Bug by downgrading to 4.4.2 from 5.0 on my Samsung Galaxy S5 (This is a known Bug, Android Issue 159529).
The -2 errors are still happening though, and sometimes there is no error but the preview of one camera is black.
Any comment would be great!
Thank you!

saki4510t, 2016-03-01:
Hi,
As your previous log seems to relate button callback, could you try to comment line 138 and 139 on `UVCCamera.cpp` like as follows?
```
//				mStatusCallback = new UVCStatusCallback(mDeviceHandle);
//				mButtonCallback = new UVCButtonCallback(mDeviceHandle);
```

I also have GALAXY S5 with Android 5.0 but I never meet `Android Issue 159529` yet. So this may not solve your issue, I update `usbCameraTest7` on `minoru001` branch so that app can automatically start preview when app has permission (this is very rough implementation and there may be issue but it works well on my GALAXY S5 with Minoru stereo camera). Could you also try this?

FYI, native implementation of libuvc on `minoru001` branch always uses alternate setting 4, and your camera may not work well. In that case please rewind line 1667, 1692, 1693 and 1706 on `stream.c` or adjust alternate setting by yourself.

saki

btut, 2016-03-01:
Hi,
thanks a lot for your reply! I highly appreciate it since I have no Idea where to continue.
Unfortunately, disabling the callbacks you mentioned did not change anything. usbCameraTest7 is not able

btut, 2016-03-08:
Hi saki,
can you tell me more about the altsettings in this library? My cameras do not have an audio interface, may that be a problem?
I cannot find a lot of information about altsettings online.

saki4510t, 2016-03-12:
Hi, my understanding related to altsettings are as follows.

on USB2.0 specification,
* there are three transfer mode; `low-speed mode`, `full-speed mode` and `high-speed mode` and maximum data transfer speed of high-speed mode is 480Mbps(=60MB/s)
* there are four transfer mode; `control transfer`, `interrupt transfer`, `bulk transfer` and `isochronous transfer` and maximum data transfer speed of `isochronous transfer` of `high-speed mode` is 24576KB/s(24MB/s) (defined on USB spec.)
* most Web cameras on the market use `isochronous transfer` to transfer images.
* additional packets need to control data transfer as defined on USB/UVC specification.
* actual data transfer speed depends on quality/length of cable, model of controller etc. 
* As I know actual maximum data transfer speed of isochronous transfer of high-speed mode is approx. 20MB/s(=bandwidth).

If we want to use `640x480,YUV, 30fps`, image size is `640x480x3/2 = 450KB/image = approx.13MB/s`. This means we don't use two camera with `640x480,YUV, 30fps` because the total required bandwidth exceeds available ones.15fps is maximum for two camera with `640x480,YUV`.

* each transfer data has structure called `frame` for `low/full speed mode` and `micro frame` for `high speed mode` and controller needs to transfer `micro frame` every 125 micro seconds.
* each `frame/micro frame` has multiple `transaction`.
* each `transaction` has multiple `packet`

If we want 30fps, 1sec(=1000 msec) / 30fps = approx.33msec. and maximum number of `micro frame` is approx. 260/sec.
So if we want to use two camera at the same time, we need to reduce number of transfer of `micro frame` and maximum available number of  `micro frame` for each camera is approx. 130/sec. This means we will need to increase amount of transfer bytes for each `micro frame`.

On the other hand, maximum transfer data bytes for each packet of each device(camera) depends on resolution, format of image(mjpeg, yuv, h264 etc.) and camera spec. etc. and USB device needs to show available maximum packet size for each `interface` on it's device descriptor and this is called `altsetting`.

We need to select `altsetting` suitable for resolution, format, fps etc. before starting `isochronous transfer`.

If you want to know more details/correctly, please read USB and UVC specification(available on USB.org as PDF file).

saki

btut, 2016-03-19:
Hi,
thanks a lot for your implementation. In fact, I am using USB3.0 and not USB2.0 and assumed that I the higher bandwidth allows me to use two cameras at a higher resolution. After checking the USB2.0 and 3.0 specifications, I came to the understanding that this assumption does not hold true. It could be that if I want to use higher frame rates or resolutions I would need to use the bulk-stream instead of isochronous transfer, and it seems that this is not implemented in your library. In fact, the datasheet of my cameras states: "Note: Please note that the See3CAMCU50 supports UVC-compliant streaming in Bulk Transport only." I don't really understand this, because the cameras are working sometimes, but maybe the isochronous transfer mode is implemented as well but with less focus and therefore more buggy.
Do you think it is possible to adapt the library to support bulk-streams? They are implemented in the current libusb, but I guess it would need a lot of work to get it working.
I would be very greatful for any input, maybe even an idea where to start implementing the bulk-stream compatibility.
And thanks again for your help so far.
Ben

saki4510t, 2016-03-19:
Hi,
I'm not sure details about your camera, this repository support `bulk` transfer, but does not support `bulk-stream` transfer. As I know the most important difference of them is `bulk-stream` transfer has non-zero stream id and needs to handle(hold and track) these stream id.
I assume you will need to use `libusb_alloc_streams`, `libusb_free_streams`, `libusb_fill_bulk_stream_transfer` instead of using `libusb_fill_bulk_transfer` etc. when start `bulk-stream` transfer.
Unfortunately I don't have camera that support USB3.0...

with best regards,
saki

btut, 2016-03-19:
Hi saki,
thanks a lot for your help. I will look into it and get back with results.

btut, 2016-03-20:
Hi saki,
after crawling through the sources all day I still have no idea how to implement this. I don't know where to get the parameters for the streams-version of the functions and how to populate the struct strmh afterwards. I am afraid Im not making progress on this project for far to long and am way to desperate. Probably I will have to give up.

saki4510t, 2016-03-20:
Hi, could you provide device descriptor of your camera as raw byte array as hex string?  I don't have much time now but I may give some advice for you.
saki

btut, 2016-03-21:
Hi,
I am very greatful for your input! I don't know how to generate the data you are asking for. lsusb shows [these](http://pastebin.com/TvrsPv2G) descriptors, but it is not a raw byte array nor a hex string. Please let me know how to generate the data. Thank you very much once more!

saki4510t, 2016-03-23:
Hi,
Thank you for providing information.
If you are familiar to Java than C/C++, most easy way will be read it as `byte` array using `UsbDeviceConnection#getRawDescriptors` and write it to `logCat` after converting hex string.
saki

btut, 2016-04-04:
Hi saki,
sorry for my late reply. I went out of town for a while and forgot the OTG cable...
The raw descriptors are:
12010002EF020140602551C10000010203010902E10103010080FA080B00020E03000209040000010E0100020D240100015000006CDC0201011224020101020000000000000000030000000C240502010040035F1400001C240603FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF000102030000000009240304010100030007058203400001052503400009040100010E0200000F2401023B018300040200000100001B240401045955593200001000800000AA00389B711001000000001E240501008002E0010000941100009411006009001516050001151605001E24050200800738040080533B0080533B00483F0080841E000180841E001E240503000005D002000057300000BC3400201C0040420F000140420F001E24050400200A980700D0144800D0144800C69900404B4C0001404B4C000A24030001200A98070006240D0000000B240602040101000000001E240701008002E0010000941100009411006009001516050001151605001E24070200800738040080533B0080533B00483F002A2C0A00012A2C0A001E240703000005D002000057300000BC3400201C001516050001151605001E24070400200A980700D0144800D0144800C69900D012130001D012130006240D00000007058302000201090402000203000000092111010001221D000705850340000107050603400001

You have my highest respect for knowing what any of this means!
Thanks for your help.

saki4510t, 2016-04-07:
Hi,
Your descriptor starts with `12010002EF020...`. This means;
```
Device Descriptor
12			bLength						18 bytes
01			bDescriptorType				Device Descriptor
0002		bcdUSB						"0200" = USB2.0
ef			bDeviceClass				Miscellaneous Device Class
02			bDeviceSubClass				Common Class
01			bDeviceProtocol				Interface Association Descriptor
40			bMaxPacketSize0				Control endpoint packet size is 64 bytes
6025		idVendor					Vendor ID=0x2560 (e-consystems)
51c1		idProduct					Product ID=0xc151 (See3CAMCU50)
0000		bcdDevice
....continue
```
And it says your camera recognized that the connection is `USB2.0`.
As I can see previous log by lsusb, the camera can work as USB3.0 and if the camera recognizes and work as USB3.0 device, `bcdUSB` should be `0300` (actual bytes in descriptor are `0003` because all values are little endian). 
Unfortunately as I know, there are very few Android devices that support USB3.0/3.1 so I assume your Android device also does not support USB3.0/3.1 or have some issue on cabling etc.
saki

btut, 2016-04-07:
Hi saki,
thanks for your help.
I am pretty sure the device supports USB3.0, thats why I have chosen the Samsung Galaxy S5. It was the only device I was able to find with an USB3.0 connector. The hub is USB3.0 as well.
Aside from that, the cameras do not work separately and on low resolutions / frame rate either, so I do not think that this is a USB3.0 issue.

saki4510t, 2016-04-08:
Hi,
Well I also have GALAXY S5(Android5.0) but it always works as `USB2.0 host` (instead of `USB3.0 host`) and sometimes half or more slower when using USB3.0 OTG cable than usual USB2.0 OTG cable.
As I can see on internet, S5 can work as `USB3.0 device` under some condition(seems to depend on firmware) but I can't find information that S5 should work as `USB3.0 host` and I assume `USB3.0 host`  feature may be disabled on my S5(and may be same on your S5).

So I think your should not expect your S5 really works with `USB3.0 host` unless you really confirm it works as `USB3.0 host` with super speed mode by USB analyzer etc.
And your camera says the connection is `USB 2.0`. This is the only truth that we could really confirm now and your S5 works only `USB 2.0 host`.

saki

btut, 2016-04-08:
Hi,
I do not question what you have read from the descriptors. I just wondered. I am astonished by your comments that sometimes your USB3.0 OTG cable is slower than the 2.0 one, I prefer to use the 3.0 cable since it provides more power, at least that function of the USB3.0 host seems to be implemented.
But the main issue is still open. Even with a USB 2.0 HUB and on resolutions that should be no problem for USB2.0, the cameras are not working.