saki4510t/UVCCamera #43: Device Has OTG support but camera not getting detected

sreyasvpariyath, 2016-02-23:
Hi 

I Have an android device which has USB OTB support(I am able to use my pen-drive) But when I connect a camera which is working fine on some other Android devices not getting detected. Do you have any idea.? Is there any way to get it detected.?

saki4510t, 2016-02-23:
Hi,

Whether Android devices can detect USB connected camera or not depends of hardware/system software(Android version) of Android device, cabling, camera etc.

As you said the camera works fine with other Android device, the issue will be 1) on the device(hardware/software), 2) on the cabling. 

1. Could you try to use USB2.0 hub?
    As my experience, using USB2.0 hub sometimes solve issues. 

2. Could you try to un-filter of logCat and check messages?
    Most Android devices will output messages tagged with `EventHub` or `UsbDeviceManager` or `UsbHostManager` like as follows when the device can successfully detect camera (this log came from my Nexus5 when I connect C930e and C920r over USB hub and I can also get similar log on Nexus6p, Nexus7, Nexus9, GALAXY S5 etc. );
    ```
02-17 15:53:36.105 800-878/? D/EventHub: No input device configuration file found for device 'HD Pro Webcam C920'.
02-17 15:53:36.123 800-878/? W/EventHub: Unable to disable kernel key repeat for /dev/input/event6: Function not implemented
02-17 15:53:36.123 800-878/? I/EventHub: New device: id=24, fd=167, path='/dev/input/event6', name='HD Pro Webcam C920', classes=0x80000001, configuration='', keyLayout='/system/usr/keylayout/Generic.kl', keyCharacterMap='/system/usr/keychars/Generic.kcm', builtinKeyboard=false, wakeMechanism=EVIOCSSUSPENDBLOCK, usingClockIoctl=true
02-17 15:53:36.463 800-878/? D/EventHub: No input device configuration file found for device 'Logicool Webcam C930e'.
02-17 15:53:36.468 800-878/? W/EventHub: Unable to disable kernel key repeat for /dev/input/event7: Function not implemented
02-17 15:53:36.468 800-878/? I/EventHub: New device: id=25, fd=231, path='/dev/input/event7', name='Logicool Webcam C930e', classes=0x80000001, configuration='', keyLayout='/system/usr/keylayout/Generic.kl', keyCharacterMap='/system/usr/keychars/Generic.kcm', builtinKeyboard=false, wakeMechanism=EVIOCSSUSPENDBLOCK, usingClockIoctl=true
    ```

with best regards,
saki

sreyasvpariyath, 2016-02-24:
Hi Saki

I don't think there is a cabling issue. I have tried using usb hub2.0. I
have tried checking the log messages as well. I have tested it with some
usb diagnostic application there it was showing in the list view as usb hd
cam. Camera is not getting powered up basically.

I don't know whether its a hardware issue. But USB drive is getting
detected . Do you have any idea.? could you please guide me.?

-- 

Thanks & Regards
*Sreyas V Pariyath | *Sr.Software Developer
*Biosense Technologies Private Limited*
Plot No:A-233,Road No 21-Y Wagle Industrial Estate, Thane West, Maharashtra
400604
*+91 22 2589 6771* {Mon-Fri, between 10 AM and 7 PM} |
*http://www.biosense.in* <http://www.biosense.in/>
M: *+91 9167556294* | Work: 1800 123 7890
CONFIDENTIALITY: This email (including any attachments) may contain
confidential, proprietary and privileged information, and unauthorized
disclosure or use is prohibited. If you receive this email in error, please
notify the sender and delete this email from your system.

jp1017, 2016-05-27:
May be supply power matters.