saki4510t/UVCCamera #44: UVCCamera7 not always works on my samsung s4.

sudharsan06, 2016-03-03:
UVCCamera7 works good on webcam C270 for first time. But After Some time UVCCamera7 dialogs "No USB Camera found" message. Some times it working , some time not , I don't know that what is the exact reason for this errror and i not able to find solution for that. 

I got the same problem when i tested using webcam pro 9000.
[image]

sudharsan06, 2016-03-03:
For connectivity i using USB 2 . There is no issues on connectivity!

saki4510t, 2016-11-20:
Hi,
I update many things today, mainly to support Android 6 and Android7. This also my be fixed.
So I close this once. I you still have trouble, please re-open or make new issue.
saki

closed 2016-11-20 by saki4510t