saki4510t/UVCCamera #48: [question] Take a picture

palmerabollo, 2016-03-18:
What is the best way to capture a single snapshot with the USB camera without processing all the video stream? Is there a function in the API to take a picture? Thank you for your help.

saki4510t, 2016-03-19:
Hi,
Unfortunately there are no way to take picture only now. Only several cameras support still capture feature of UVC specification and I don't plan to support still capture feature of UVC specification on this repository now.
So if you want to capture a image without preview, you need to modify native library so that it can start streaming without preview(without `Surface`) and get a image then stop streaming.
Or you can pass `SurfaceTexture` as offscreen preview `Surface`, start streaming, take image and stop streaming.

Note: There is a possibility that first several video frames from camera are broken so it will be better to skip first several frames and then capture a image.

saki

palmerabollo, 2016-03-19:
Thanks for the explanation and the tips, saki. I'll try with the offscreen preview, that seems easier than modifying the native library.

closed 2016-03-19 by palmerabollo

saki4510t, 2016-03-19:
Hi,
Some hints for using SurfaceTexture as offscreen is available on `UVCCameraTextureView.java`, `RenderHandler.java` in `usbCameratest3` etc.
saki

palmerabollo, 2016-03-19:
Ok. I'll take a look. I started with `usbCameratest4` because I also want to run it as a `Service` and take the picture when it receives an external notification (ex. an incoming SMS or when I press the volume buttons). That's why I don't need an `Activity` + a `Surface`. I hope that's possible...

saki4510t, 2016-03-19:
`usbWebCameraTest4` also includes offscreen preview because offscreen is necessary for background service. But you need at least one `Activity` to keep permanent permission and install apt on your device (by limitation of Android OS itself).
saki

palmerabollo, 2016-03-19:
Yes, that is not a problem. What I don't know is if it will continue working when the screen is off.

ducanh841988, 2017-10-03:
Did you achieve taking picture @palmerabollo ?

palmerabollo, 2017-10-03:
@ducanh841988 Yes. I can say that it's doable but it was just a quick & dirty experiment and I don't remember the details of how I managed to do it :(

fengbenpaao, 2017-11-07:
if I use the onFremeCallback   to   get the frame date to  imgae  ?
 but  the data I get,  don't show  ok,    it will be green,  reb  。。。。

akshayah3, 2019-08-31:
Did anyone achieve to do this?. Basically i want to capture the image in a background service without any preview or display

KanwalKhan-20, 2021-05-29:
How to take picture of UVCCameraTextureView alongwith textview ?
I am only able to take screenshot of either UVCCameraTextureView or textview.

Keerthana-N, 2023-03-06:
> @ducanh841988 Yes. I can say that it's doable but it was just a quick & dirty experiment and I don't remember the details of how I managed to do it :(

Hi Can you please share the solution.