saki4510t/UVCCamera #49: Exposure control

DrakkLord, 2016-03-22:
Hi!

I'm working on a project where we use and older fork of your code where exposure control is still available and it does weird things.
First of all may I ask why setting exposure time is not exposed in the code currently on master is there a reason for it? I did check other controls with the cameras we have here and they work fine, gain control / brightness control however changing exposure may do the following things depending on the camera:

- image disappears if you try to set the exposure
- exposure setting is different based on the camera in some cases it's from  say -16 to -1 or a time in milli seconds, I guess this is related to the UVC version used by the camera if so how can I get the UVC version used by the camera attached.
- setting the exposure level breaks the camera meaning that I can set the exposure once which works then exiting the app, stopping the preview will make the camera stuck in a state where it acts as if it were told to use some super high exposure, however reconnecting the camera fixes the isssue
- in some cases the exposure setting doesn't work after setting it once, so any subsequent changes will have no effect until you reconnect the camera.

saki4510t, 2016-03-23:
Hi,
Thank you for providing information and question.
Well as you know this repository is based on `libusb` and `libuvc` and `livubc` implemented video control functions with wrong way(currently works well). If you use older fork, I assume it will be better to check/compare `ctrl.c` and apply it as possible as you can.

The actual values that each camera control function can accept depends on each camera and it does not depend on `UVC version`. You need to get them using `UVC_GET_MIN`, `UVC_GET_MAX`, `UVC_GET_RES` etc.

We can independently use most of other camera control functions. But regarding `exposure`, there are four camera control functions in UVC specification directly related to `exposure`; `Auto-Exposure Mode Control`, `Auto-Exposure Priority Control`, `Exposure Time(Absolute) Control` and `Exposure Time (Relative) Control `. And they strongly depend on each other. 
Unfortunately they sometimes depend on other camera functions like `Iris (Absolute) control`, `Iris (Relative) Control` and acceptable frame rate etc. and your need to use available combination of values only. So if your camera does not work well on some condition, the reason is just simple, because the value was not a acceptable value on the current camera state.
As described in UVC specification, if you set wrong value to camera control function, the control shall result in a protocol STALL and an error code of `bRequestErrorCode="Wrong state"` and you need to reset. You need to read UVC specification very well and test well with the camera that you want to use.

Totally, controlling `exposure` is difficult than other functions and some cameras do not support this control and I have not tested well yet and don't know the best way to get available vales and get it work well for all camera that have `exposure` control.
saki

saki4510t, 2016-06-23:
closed due to inactivity.

closed 2016-06-23 by saki4510t