saki4510t/UVCCamera #51: UnsatisfiedLinkError on Samsung Galaxy Tab 3

PinkJazz, 2016-03-31:
I am currently using your library, and for some reason, I tested it with a Samsung Galaxy Tab 3, and I am getting an UnsatisfiedLinkError when I try to open the camera.

The Tab 3 runs Android 4.4.2.

I wonder if the Intel Atom processor could have something to do with it.  It works fine on a Samsung Galaxy Tab 4, which has an ARM architecture processor.

saki4510t, 2016-04-01:
Hi,
Thank you for providing feedback.
Could you provide log?
This repository can build x86 binary with default settings. And as I know most devices that use ATOM processor on the market support x86 binary translator and ARM binary still works on the devices.

FYI, as I can see on internet, unfortunately `GALAXY TAB 3 7''` does not support OTG function and this means this repository never work on the device. (GALAXY TAB 3 8'' and 10'' support OTG though)
saki

saki4510t, 2016-05-09:
closed due to inactivity.

closed 2016-05-09 by saki4510t