saki4510t/UVCCamera #52: Web cam is not working in Lenovo A3300-hv tab.

MitulAhir, 2016-04-02:
Thank you for this great open source library. I am using this library to start a UVC compliant  web camera in android. It is working fine in some android devices and in other  android device like Lenovo A3300-hv I am facing a an issue in receiving a frame from a UVC compliant web Camera. Tab is rooted and has a  Video4Linux enabled custom kernel with Mediatek chip set.

Some of UVC compliant camera is working by downgrading the resolution and some are not working. 

Application get stucked in "_uvc_user_caller()" function at "pthread_cond_wait(&strmh->cb_cond, &strmh->cb_mutex);" . 

Can you please tell me the reason why I am not receiving any frame from a Camera. 

Please find attached  log file. 
[Applog.txt](https://github.com/saki4510t/UVCCamera/files/200680/Applog.txt)

saki4510t, 2016-04-02:
Hi,
As I know isochronous transfer can not work well from user-land app on most of China made SOC like MediaTek ones (or can not detach kernel driver). And most of web cameras only support isochronous transfer.
As your log also shows that isochronous transfer returns error status and no data,
```
04-02 10:14:07.163: I/libUVCCamera(9445): [stream.c:810:_uvc_process_payload_iso]:bad packet:status=1,actual_length=0
...
```
So if you/app has root permission (and v4l2 kernel driver is enabled), I assume directly using v4l2 will be better on those devices. 
saki

MitulAhir, 2016-04-02:
Dear Saki,
Thank you for your response.
I have a two web camera one is Logitek C110 and Tech-com USB2.0 PC camera,both are working fine with mediatek chipset with your project in Lenovo tab.
we have another web camera module with  AV318s chip set which is not working particularly in Lenovo A3300-hv model. Is there any specific settings for this camera module? 

Can you please share any project for reference on V4l2 using which I can start streaming on Lenovo A3300-hv rooted device?

MitulAhir, 2016-04-04:
Dear Saki,
We have tried a video4linux API also to start device. In that process also device detection, initialization, buffer management, Querying video capture capability , setting start Stream on  functionalities are working fine. But when we try to read a buffer using "IOCTL VIDIOC_DQBUF" API. We receive 0 byte buffer from a device. 

Can you Please suggest why it is return a 0 byte buffer?

Thank you