saki4510t/UVCCamera #53: Sometimes get "bad packet:status 0xcc/0xcd" on ASUS Zenphone2 Z011D

elviskuo07, 2016-04-05:
Hi

Sometimes I get bad packet message and screen freeze, is there any suggestion to avoid that ?

I created some threads in system, if system overloading  will occur bad packet issue?

Thanks.

elviskuo07, 2016-04-05:
Log message
04-05 16:50:56.058 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.058 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.058 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.058 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.059 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.059 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.059 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.060 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.060 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.060 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.061 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.061 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.061 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.062 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.062 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.063 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.064 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.064 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.064 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.065 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.065 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.065 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.066 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.066 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.067 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.067 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.067 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc
04-05 16:50:56.067 1182-3706/? I/libUVCCamera: [stream.c:842:_uvc_process_payload_iso]:bad packet:status=0xcc

saki4510t, 2016-04-05:
Hi,
There are many reason.
If your ZenPhone2 is not too hot, I assume most possible reason will be your cable is too long or quality of cable/OTG cable is low, etc. In this case, using shower cable, other cable, other OTG cable/connector, USB2.0 hub will help you.
If your ZenPhone2 is too hot than usual, try to use lower resolution or if you use YUV, try to use  MJPEG mode etc.

saki

elviskuo07, 2016-04-05:
Got it.
Thanks for your advice.

closed 2016-05-09 by saki4510t