saki4510t/UVCCamera #55: UVCCamera4 can not works for a long time on  samsung s5 s6 nexus7.

lsy17096535, 2016-04-16:
thanks for your library. I am currently using UVCCamera4, when I test on my device , it can not works more than twenty minutes, I don't know if the system kill the service.
                                                                  syan

saki4510t, 2016-04-17:
Hi, thank you for providing informations.
As I tested, Android system kill usbCameratest4 after 15 minutes of background recording(screen was also off) on Nexus7(2013, Android6.0.1) and log shows as follows.
```
I/ActivityManager: Killing 18095:com.serenegiant.usbcameratest4:uvcservice/u0a459 (adj 9): excessive cpu 616390 during 900050
I/ActivityManager: Killing 17732:com.serenegiant.usbcameratest4/u0a459 (adj 13): remove task
```
Nexus7(2013, Android5.1.1) worked for 17minutes.

I assume if you want to work for long period, it will be better to use notification and run uvcservice as foreground service. Please check `Service#startForeground`.
saki

lsy17096535, 2016-04-17:
Thanks for the reply.
I put the app at system/app Folder on Samsung 5, it can work two hours, but on s6 or nexus7, the app can not run.
I will try  Service#startForeground  
thanks again.

lsy17096535, 2016-06-22:
i tried Foreground service, acitvity also been killed. Bindservice has been used in usbCameratest4. when i use startservice instead of bindservice , and also write as foreground sevice . it can work well. but the service can communicate with the activity.
what should do to let it work well and communicate with activity?
thanks!
syan

closed 2017-05-04 by lsy17096535