saki4510t/UVCCamera #56: Camera Button click

ahmgsk, 2016-04-26:
I have been trying to get button on camera get snap of the picture get triggered.
but failing continuously.

Help would be very appreciated.

Thank you,
ahmgsk

saki4510t, 2016-04-27:
Hi, as I know there are several types to notify/get button state and the way to get still image capture depends on what features your camera supports.
Can you receive button/status callback when you press button on camera?
Does your camera support hardware trigger(`bTriggerSupport` field on Class-specific VS Interface Input Header Descriptor is 1)?
What still image capture method does your camera support?
saki

ahmgsk, 2016-04-27:
How can i receive that button click is my question, once i get that click from usb device i can then take things further.
i will try bTriggerSupport and get back

ahmgsk, 2016-04-28:
Basically what i trying to achieve is Micro Camera working with UVC CAMERA APP.
Camera is working perfectly, Very thankful for that.
But the the snap button on the camera. does no trigger nothing. I tried usb keyboard bases to capture its keyvalue. i tried to integrate USBmanager endpoint bases to gather bytes but failed to get it working.
its working with camerafi app.. so button is functional thats for sure.

oury1, 2019-06-16:
Hi i reopen this issue .
I'm trying to handle the webcam hardware button. i'm register to setButtonCallback in the UVCCamera but don't have any callback called when button pressed 
someone can help how i can handle the webcam hardware button

thanks