saki4510t/UVCCamera #57: Dual camera not working with Silicon Motion camera.

noisysoil, 2016-04-27:
I have 2x DBPower endoscope cameras, which have a Silicon Motion chipset - unfortunately I cannot get them both to work with usbcameratest7 - one appears fine, but not both.  I have tried the minoru branch and none of them get displayed.  Here's a dump of the device with lsusb on Linux:

udo lsusb -s 002:004 -v

Bus 002 Device 004: ID 090c:f37d Silicon Motion, Inc. - Taiwan (formerly Feiya Technology Corp.) Endoscope camera
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x090c Silicon Motion, Inc. - Taiwan (formerly Feiya Technology Corp.)
  idProduct          0xf37d Endoscope camera
  bcdDevice            0.04
  iManufacturer           1 SMI
  iProduct                2 USB 2.0 Camera
  iSerial                 3 webcam
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          695
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               1 SMI
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0 
      iInterface              1 SMI
      VideoControl Interface Descriptor:
        bLength                13
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.00
        wTotalLength           77
        dwClockFrequency       30.000000MHz
        bInCollection           1
        baInterfaceNr( 0)       1
      VideoControl Interface Descriptor:
        bLength                18
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0201 Camera Sensor
        bAssocTerminal          0
        iTerminal               0 
        wObjectiveFocalLengthMin      0
        wObjectiveFocalLengthMax      0
        wOcularFocalLength            0
        bControlSize                  3
[... 476 lines cut ...]
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            5
          Transfer Type            Isochronous
          Synch Type               Asynchronous
          Usage Type               Data
        wMaxPacketSize     0x0280  1x 640 bytes
        bInterval               1
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)

What could be the issue here?

Regards.

saki4510t, 2016-05-10:
Hi, sorry my answer delayed.
Unfortunately I don't know certain way to get them work well, could you try followings?
1. Connect one camera
2. open and start preview with 640x480, MJPEG. if it works, set smaller value to bandwidth factor than before (please repeat step 2 until you find minimum value of bandwidth factor).
3. set minimum bandwidth factor for both camera
4. connect two camera and try to start preview.

FYI, shorter cabling and self powered USB hub (if you use bus powered USB hub now) may produce better result.
saki

noisysoil, 2016-05-11:
Hey, thanks for getting in touch.  I have tried changing DEFAULT_BANDWIDTH in master branch with various values (0.2f, 0.5f, 0.67f, 0.8f and 1.0f) but the issue still occurs.

I have done some further testing on master branch and this is the sequence of what happens:

- Run app with no cameras connected.
- Connect camera 1.
- Select left view on UI; dialog with unique USB appears with Camera 1 listed - select camera 1 ID.
- Camera 1 stream appears on left view of UI.
- Connect camera 2.
- Select right view on UI; dialog with unique USB appears with Camera 2 listed - select camera 2 ID.
- **Camera 2 stream appears on left view of UI**.

The second camera takes over the first.  Here's the LogCat of the above sequence:

```
05-11 13:45:16.055 10289-10289/com.serenegiant.usbcameratest7 W/InputEventReceiver: Attempted to finish an input event but the input event receiver has already been disposed.
05-11 13:45:16.725 10289-10320/com.serenegiant.usbcameratest7 V/CameraThread: handleStopRecording:mMuxer=null
05-11 13:45:16.725 10289-10320/com.serenegiant.usbcameratest7 V/CameraThread: handleStopPreview:
05-11 13:45:16.725 10289-10289/com.serenegiant.usbcameratest7 V/UVCCameraTextureView: onPause:
05-11 13:45:16.725 10289-10289/com.serenegiant.usbcameratest7 V/UVCCameraTextureView: release:
05-11 13:45:16.725 10289-10320/com.serenegiant.usbcameratest7 V/CameraThread: handleClose:
05-11 13:45:16.725 10289-10317/com.serenegiant.usbcameratest7 V/CameraThread: handleStopRecording:mMuxer=null
05-11 13:45:16.725 10289-10320/com.serenegiant.usbcameratest7 V/CameraThread: handleStopRecording:mMuxer=null
05-11 13:45:16.725 10289-10317/com.serenegiant.usbcameratest7 V/CameraThread: handleStopPreview:
05-11 13:45:16.725 10289-11244/com.serenegiant.usbcameratest7 D/UVCCameraTextureView: RenderThread finishing
05-11 13:45:16.725 10289-10289/com.serenegiant.usbcameratest7 V/UVCCameraTextureView: onPause:
05-11 13:45:16.725 10289-11244/com.serenegiant.usbcameratest7 V/UVCCameraTextureView: RenderThread#release:
05-11 13:45:16.725 10289-11244/com.serenegiant.usbcameratest7 V/EGLBase: EglSurface:release:
05-11 13:45:16.725 10289-11244/com.serenegiant.usbcameratest7 V/EGLBase: makeDefault:
05-11 13:45:16.725 10289-11244/com.serenegiant.usbcameratest7 V/EGLBase: destroySurface:
                                                                         
                                                                         [ 05-11 13:45:16.725 10289:11244 D/         ]
                                                                         Surface destroy: ANDROID_NATIVE_WINDOW_MAGIC
05-11 13:45:16.725 10289-10317/com.serenegiant.usbcameratest7 V/CameraThread: handleClose:
05-11 13:45:16.725 10289-11244/com.serenegiant.usbcameratest7 V/EGLBase: destroySurface:finished
05-11 13:45:16.725 10289-11244/com.serenegiant.usbcameratest7 V/EGLBase: release:
05-11 13:45:16.725 10289-11244/com.serenegiant.usbcameratest7 V/EGLBase: destroyContext:
05-11 13:45:16.725 10289-10317/com.serenegiant.usbcameratest7 V/CameraThread: handleStopRecording:mMuxer=null
05-11 13:45:16.725 10289-10289/com.serenegiant.usbcameratest7 V/UVCCameraTextureView: release:
05-11 13:45:16.725 10289-11245/com.serenegiant.usbcameratest7 D/UVCCameraTextureView: RenderThread finishing
05-11 13:45:16.725 10289-11245/com.serenegiant.usbcameratest7 V/UVCCameraTextureView: RenderThread#release:
05-11 13:45:16.725 10289-11245/com.serenegiant.usbcameratest7 V/GLDrawer2D: deleteTex:
05-11 13:45:16.725 10289-11245/com.serenegiant.usbcameratest7 V/EGLBase: EglSurface:release:
05-11 13:45:16.725 10289-11245/com.serenegiant.usbcameratest7 V/EGLBase: makeDefault:
05-11 13:45:16.725 10289-11245/com.serenegiant.usbcameratest7 V/EGLBase: destroySurface:
                                                                         
                                                                         [ 05-11 13:45:16.725 10289:11245 D/         ]
                                                                         Surface destroy: ANDROID_NATIVE_WINDOW_MAGIC
05-11 13:45:16.725 10289-11245/com.serenegiant.usbcameratest7 V/EGLBase: destroySurface:finished
05-11 13:45:16.725 10289-11245/com.serenegiant.usbcameratest7 V/EGLBase: release:
05-11 13:45:16.725 10289-11245/com.serenegiant.usbcameratest7 V/EGLBase: destroyContext:
05-11 13:45:19.325 10289-10289/com.serenegiant.usbcameratest7 V/UVCCameraTextureView: onResume:
05-11 13:45:19.325 10289-10289/com.serenegiant.usbcameratest7 V/UVCCameraTextureView: RenderThread#getHandler:
05-11 13:45:19.325 10289-13137/com.serenegiant.usbcameratest7 D/UVCCameraTextureView: RenderThread started
05-11 13:45:19.325 10289-13137/com.serenegiant.usbcameratest7 V/UVCCameraTextureView: RenderThread#init:
05-11 13:45:19.325 10289-13137/com.serenegiant.usbcameratest7 V/EGLBase: EGLBase:
05-11 13:45:19.325 10289-13137/com.serenegiant.usbcameratest7 V/EGLBase: init:
                                                                         
                                                                         [ 05-11 13:45:19.325 10289:13137 D/         ]
                                                                         droid_create_context : config id = 1 conf->NativeVisualID=1
                                                                         
                                                                         
                                                                         [ 05-11 13:45:19.325 10289:13137 D/         ]
[... cut ...]
05-11 13:45:41.015 10289-10317/com.serenegiant.usbcameratest7 V/CameraThread: handleStartPreview:
05-11 13:45:41.045 10289-13469/com.serenegiant.usbcameratest7 I/libUVCCamera: [13469*UVCPreview.cpp:492:prepare_preview]:frameSize=(640,480)@MJPEG
05-11 13:45:42.325 10289-13470/com.serenegiant.usbcameratest7 W/libUVCCamera: [13470*UVCPreview.cpp:119:get_frame]:allocate new frame
05-11 13:45:42.325 10289-13469/com.serenegiant.usbcameratest7 W/libUVCCamera: [13469*UVCPreview.cpp:119:get_frame]:allocate new frame
```

ZejiaZheng, 2016-09-25:
I had the same issue. 
The second camera would take either take over, or it would start a blank screen.