saki4510t/UVCCamera #60: Cannot compile library ndk-build

kdtdeta, 2016-05-11:
Ths for this nice tools !

I can't compile the library using ndk-build : error

 CreateProcess(NULL, D:/android-ndk-r11c-windows-x86/android-ndk-r11c/build//../toolchains/arm-linux-androideabi-4.9/prebuilt/windows-x86_64/bin/arm-linux-androideabi-g++ -MMD -MP -MF D:/eclipse-SDK-4.2.2-win32-x86_64/Workspace/main/obj/local/armeabi/objs/UVCCamera/_onload.o.d -fpic -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -march=armv5te -mtune=xscale -msoft-float -fno-exceptions -fno-rtti -O2 -g -DNDEBUG -fomit-frame-pointer -fstrict-aliasing -funswitch-loops -finline-limit=300 -ID:/eclipse-SDK-4.2.2-win32-x86_64/Workspace/main/jni/UVCCamera/ -ID:/eclipse-SDK-4.2.2-win32-x86_64/Workspace/main/jni/UVCCamera/../ -ID:/eclipse-SDK-4.2.2-win32-x86_64/Workspace/main/jni/UVCCamera/../rapidjson/include -ID:/android-ndk-r11c-windows-x86/android-ndk-r11c/build//../sources/cxx-stl/system/include -ID:/eclipse-SDK-4.2.2-win32-x86_64/Workspace/main/jni/libusb/android/jni/../../ -ID:/eclipse-SDK-4.2.2-win32-x86_64/Workspace/main/jni/libusb/android/jni/../../libusb -ID:/eclipse-SDK-4.2.2-win32-x86_64/Workspace/main/jni/libuvc/android/jni/../../ -ID:/eclipse-SDK-4.2.2-win32-x86_64/Workspace/main/jni/libuvc/android/jni/../../include -ID:/eclipse-SDK-4.2.2-win32-x86_64/Workspace/main/jni/libuvc/android/jni/../../include/libuvc -ID:/eclipse-SDK-4.2.2-win32-x86_64/Workspace/main/jni/libjpeg-turbo-1.4.0/ -ID:/eclipse-SDK-4.2.2-win32-x86_64/Workspace/main/jni/UVCCamera -DANDROID -ID:/eclipse-SDK-4.2.2-win32-x86_64/Workspace/main/jni/UVCCamera/ -ID:/eclipse-SDK-4.2.2-win32-x86_64/Workspace/main/jni/UVCCamera/../ -ID:/eclipse-SDK-4.2.2-win32-x86_64/Workspace/main/jni/UVCCamera/../rapidjson/include -DANDROID_NDK -DLOG_NDEBUG -DACCESS_RAW_DESCRIPTORS -O3 -fstrict-aliasing -fprefetch-loop-arrays -Wa,--noexecstack -Wformat -Werror=format-security -isystem D:/android-ndk-r11c-windows-x86/android-ndk-r11c/build//../platforms/android-14/arch-arm/usr/include -c D:/eclipse-SDK-4.2.2-win32-x86_64/Workspace/main/jni/UVCCamera/_onload.cpp -o D:/eclipse-SDK-4.2.2-win32-x86_64/Workspace/main/obj/local/armeabi/objs/UVCCamera/_onload.o, ...) failed.
make (e=2): Le fichier spécifié est introuvable.

make: *** [D:/eclipse-SDK-4.2.2-win32-x86_64/Workspace/main/obj/local/armeabi/objs/UVCCamera/_onload.o] Error 2

I need your help to recompile and have the .so files to run the samples.

Kdt

saki4510t, 2016-05-30:
Hi,
Sorry my answer delayed.
As I can see you log, it seems to be build on Eclipse? and error `Le fichier spécifié est introuvable.` means `specific file not found`?
`D:/android-ndk-r11c-windows-x86/android-ndk-r11c/build//../toolchains/arm-linux-androideabi-4.9/prebuilt/windows-x86_64/bin/arm-linux-androideabi-g++` etc. is correct path? I think duplicated slash after `build` will be wrong.
Is it possible to build using gradle or ndk-build from command line?
saki

rashmifiroz, 2016-06-25:
Hi Saki,
I am having this specified file not found error when compiling the library. Could you kindly help to resolve it?
I have followed the steps you have given for eclipse.
I am using ndk-build from Windows command prompt. Android-NDK is r12.

D:\AndroidProjects\UVCcode\UVCCamera\libuvccamera\src\main>ndk-build
Android NDK: WARNING: APP_PLATFORM android-21 is larger than android:minSdkVersi
on 14 in ./AndroidManifest.xml
[armeabi] "Compile++ arm  ": "UVCCamera <= _onload.cpp"
process_begin: CreateProcess(NULL, C:/android-ndk-r12/build//../toolchains/arm-l
inux-androideabi-4.9/prebuilt/windows-x86_64/bin/arm-linux-androideabi-g++ -MMD
-MP -MF ./obj/local/armeabi/objs/UVCCamera/_onload.o.d -fpic -ffunction-sections
 -funwind-tables -fstack-protector-strong -no-canonical-prefixes -g -march=armv5
te -mtune=xscale -msoft-float -fno-exceptions -fno-rtti -marm -O2 -DNDEBUG -Ijni
/UVCCamera/ -Ijni/UVCCamera/../ -Ijni/UVCCamera/../rapidjson/include -IC:/androi
d-ndk-r12/build//../sources/cxx-stl/system/include -Ijni/libusb/android/jni/../.
./ -Ijni/libusb/android/jni/../../libusb -Ijni/libuvc/android/jni/../../ -Ijni/l
ibuvc/android/jni/../../include -Ijni/libuvc/android/jni/../../include/libuvc -I
jni/libjpeg-turbo-1.5.0/ -Ijni/libjpeg-turbo-1.5.0/include -Ijni/libjpeg-turbo-1
.5.0/simd -Ijni/UVCCamera -DANDROID -Ijni/UVCCamera/ -Ijni/UVCCamera/../ -Ijni/U
VCCamera/../rapidjson/include -DANDROID_NDK -DLOG_NDEBUG -DACCESS_RAW_DESCRIPTOR
S -O3 -fstrict-aliasing -fprefetch-loop-arrays -Wa,--noexecstack -Wformat -Werro
r=format-security -isystem C:/android-ndk-r12/build//../platforms/android-21/arc
h-arm/usr/include -c jni/UVCCamera/_onload.cpp -o ./obj/local/armeabi/objs/UVCCa
mera/_onload.o, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [obj/local/armeabi/objs/UVCCamera/_onload.o] Error 2

saki4510t, 2016-06-26:
Hi,
As I tested on my Windows 7(64bit) with r12(64bit), my repositories can build and works well.
I also tested on my Windows 7(64bit) with r12(32bit), it generated same error as yours.

So I assume you might install wrong version of NDK, ex. 64bit version on 32bit Windows or 32bit version on 64 bit Windows. Because the download link of latest NDK(r12) on Android Developers is wrong (reversed). 

It looks like this;

| Platform | Package | Size (Bytes) | SHA1 Checksum |
| ----- | ----- | ----- | ----- |
| Windows 32-bit | android-ndk-r12-windows-x86_64.zip | 749444245 | ... |
| Windows 64-bit | android-ndk-r12-windows-x86.zip | 706332762 | ... |
| ... | ... | ... | ... |

If you use 32bit Windows, you should download and install `android-ndk-r12-windows-x86.zip` and if you use 64bit Windows, you should download and install `android-ndk-r12-windows-x86_64.zip`

I don't have 32bit Windows now and I'm not sure r12(32bit) works on 32bit Windows though.

with best regards,
saki

rashmifiroz, 2016-06-26:
Thanks a lot for your response. Mine is 64bit Windows and It worked when i installed android-ndk-r12-windows-x86_64.zip. Actually in the website the names are wrongly mentioned for the 32bit and 64bit.

closed 2016-06-26 by saki4510t