saki4510t/UVCCamera #61: Unity3d support

sdnavin, 2016-05-29:
Hi, You have any idea for making a jar file from the .so library ? so i can use it in unity3d to make use of this?

saki4510t, 2016-05-29:
Hi,
I'm not so familiar Unity3d, but if you need jar , you can compile the .aar file with Android Studio , extract it(.aar file is compressed by zip), and use the 'classes.jar' (and .so) to access the java classes inside the project that you have.

sdnavin, 2016-06-01:
@saki4510t  : I almost intergrated with unity3d, But can you help in the Iframecallback? 
Do i need to StartPreview? to get the Iframecallback work ? 

Current Situation : I got the libraries in unity, Get it detected as camera, Got the supported resolutions. But the Iframecallback is not working.

saki4510t, 2016-06-01:
Hi,
@sdnavin yes, you need to call `open`, `setPreviewSize`, `setPreviewXXX` and then `startPreview` to receive camera images.

I'm not sure how you want to use images from UVC camera on Unity, if you want to use images as texture(as `Texture2D` on Unity), I think it is not a good idea to use `IFrameCallback`. It is better to accept them as OpenGL|ES texture(using `SurfaceTexture`) and pass OpenGL|ES texture handle to Unity using `Texture2D.CreateExternalTexture` and `Texture2D.UpdateExternalTexture` otherwise quite inefficient lot of memory copy over JNI are performed.

sdnavin, 2016-06-07:
Can you help me in here,

Java side : On draw
	                        mEglSurface.makeCurrent();
				// update texture(came from camera)
				mPreviewSurface.updateTexImage();
				// get texture matrix
				mPreviewSurface.getTransformMatrix(mStMatrix);
				if (mEncoder != null) {
					// notify to capturing thread that the camera frame is available.
					if (mEncoder instanceof MediaVideoEncoder)
						((MediaVideoEncoder)mEncoder).frameAvailableSoon(mStMatrix);
					else
						mEncoder.frameAvailableSoon();
				}
				// draw to preview screen
				mDrawer.draw(mTexId, mStMatrix);
				mEglSurface.swap();

Unity Side :
Texture2D.UpdateExternalTexture(mTexId);

But i got no data in unity.

pun777chy, 2016-12-27:
@sdnavin Brother if you have resolved this issue and successfully made unity package for USB Camera, can you share that plugin?

CatsNipYummy, 2017-02-23:
@sdnavin were you able to create the Unity package? Any way you could share the same?

tairtec, 2017-03-07:
@sdnavin I've seen someone like you're going to make unity plugins. [here](https://github.com/TimGroeneboom/libuvc_android_unity)

pun777chy, 2017-03-07:
@tairtec  Crashing at the beginning 
No, it doesn't work at all.
First its manifest doesn't have unity activity. Even if we add that it still crash.

MikeMijnheer, 2017-05-11:
Since this is the only topic I found about .jar files.. can someone make a .jar or .aar from this project and share it? I really have no clue on how to do this. Thanks!

lb5160482, 2017-09-27:
@pun777chy I tried that, it actually worked. Looks promising.

lambadaa, 2019-01-16:
Update, I have managed up update the previously linked repo by Timgroeneboom and managed to get it to connect to the camera and fetch a list of supported resolutions. However, I've hit another roadblock. The device I'm testing on runs Android 7.0, and when I try to fetch preview frames it fails and returns this in the adb log:

`
01-16 16:45:44.471 28160 28191 I libUVCCamera: [28191*UVCPreview.cpp:264:callbackPixelFormatChanged]:PIXEL_FORMAT_YUV:
01-16 16:45:44.473 28160 28364 E libUVCCamera: [28364*UVCPreview.cpp:510:prepare_preview]:could not negotiate with camera:err=-51
01-16 16:45:48.248 28160 28191 W libUVCCamera: [28191*UVCPreview.cpp:365:stopPreview]:UVCPreview::terminate capture thread: pthread_join failed
`

It seems to be the same issue as #100. I've tested it on an in-house custom camera and a logitech c270 and both ends up with the same result (err=-51). I'll attempt to dig into this issue and update if I manage to find anything, but in the meantime if anyone's had experience with this input would be appreciated.

HyroVitalyProtago, 2019-03-22:
@lambadaa Any news? Have you successfully fetch frames?

joaodafonseca, 2019-04-11:
@lambadaa any luck with this?

flyAssassinator, 2019-08-14:
I found a demo about this: https://www.youtube.com/watch?v=DKQUrHi3oOk