saki4510t/UVCCamera #63: stream usb cam data to a streaming server

srirammadduri, 2016-06-16:
hi,

is it possible to stream this to a remote server so that users will see this camera video

is it possible with this library?

saki4510t, 2016-06-26:
Hi,

The libraries on this repository just get video images from UVC camera and does not include libraries to handle video streaming protocol. So you need to combine other libraries to handle streaming protocol like RTSP, RTMP, WebRTC etc.

Arlem, 2016-09-09:
I have the same problem, what's your suggestion (my project is based on USBCameraTest4)? Using IFrameCallback or modifying the MediaMuxerWrapper? Any streaming type would suit, RTMP preferred thought.

KitPan, 2016-09-17:
if you want less latency you better use MJPG,that means just listen at a certain port and redirect the image from the camera to a connected socket, if you want RTSP or RTMP etc, you may need create a virtual display and then using media codec encode it to H.264 then streaming by UDP, you can take a look at https://github.com/KitPan/spydroid-ipcamera about video streaming

Arlem, 2016-10-19:
Can you expand on "if you want less latency you better use MJPG,that means just listen at a certain port and redirect the image from the camera to a connected socket"? How do I listen to the USB Camera image?

If I understood correctly, VirtualDisplay can be used only to capture screen, which is not showing the USB Camera preview.

FanHongchuang, 2017-12-07:
@srirammadduri 
Hello,
Are you resolve it about stream video to a remote server?

jp1017, 2018-01-12:
RTSP with UVC here: @FanHongchuang