saki4510t/UVCCamera #64: Google vulnerabilty for libjpeg-turbo

ericeche, 2016-06-18:
https://support.google.com/faqs/answer/7008337

Google suggests to update to libjpeg-turbo v1.4.2 or greater. Any idea how to do the update ?

ericeche, 2016-06-18:
Here is the git repo with update:
https://github.com/libjpeg-turbo/libjpeg-turbo

saki4510t, 2016-06-18:
Hi,
I replaced libjpeg-turbo 1.4.0 with 1.5.0. please try latest master branch.

ericeche, 2016-06-18:
Thanks for your prompt reply

ericeche, 2016-06-18:
What happen with mips build ? May I ask why it was taken out of the build. Just curiosity

saki4510t, 2016-06-22:
Hi,
There are few devices that SOC is powered by mips.  As I know there are no mips powered smartphone/tablet in Japan and I also don't have.  This means I can't confirm whether these libraries work on the devices or not. So originally I have not support mips but someone added it on pull request.  
I added settings for mips now but it is just for convenience and I'm not sure it works on mips powered devices.

closed 2016-06-22 by saki4510t

ybdesale-vanlentsystems, 2025-07-09:
@saki4510t I want to upgrade libjpeg-turbo 1.5.0 with 2.1.5. Please suggest the changes. I had tried but not able to do that. @ericeche you can also help if you have any idea about this.