saki4510t/UVCCamera #65: usbCameraTest7 is not working......

Linjieqiang, 2016-06-19:
The first,thanks your great work~!But I meeted a question.When I try to use the demo(usbCameraTest7 ) with double camera input,but it's not working..Why?I uploaded some photos to give you have a look.I am looking forward to you can fix it ?May be my mistakes?It only single channel worked,and the another uvc channel doesn't work.

Linjieqiang, 2016-06-19:
[image]
[image]

noisysoil, 2016-07-28:
I'm having a similar related problem: https://github.com/saki4510t/UVCCamera/issues/57

ZejiaZheng, 2016-09-25:
I had the same issue but it was resolved by reverting the code to a previous commit:
https://github.com/saki4510t/UVCCamera/blob/4be6cab3bef3d2007fe2078d31e561df303fe7b1/usbCameraTest7/src/main/java/com/serenegiant/usbcameratest7/MainActivity.java