saki4510t/UVCCamera #69: How to adjust camera RPG parameter?

enight0000, 2016-07-14:
I tried to adjust brightness, sharpness, saturation and gamma, it works!
So I want to adjust camera rpg parameter, because the image is red. I need to write some date on camera's temporary storage.
First I add a function in ctrl.c and call it from UVCCamera.java, but it doesn't work.
`uvc_error_t uvc_set_dsp(uvc_device_handle_t *devh,uint16_t address,uint16_t asicdata){
	uint8_t data[4];
	uvc_error_t ret;
	//SHORT_TO_SW(asicdata, data);
	data[0] = address;
	data[1] = (address >> 8);
	data[2] = asicdata;
	data[3] = 0x00;	

	ret = libusb_control_transfer(devh->usb_devh, 0x21, 0x01,
			0x100,
			devh->info->ctrl_if.input_term_descs->request,
			data, sizeof(data), CTRL_TIMEOUT_MILLIS);

	//int API_EXPORTED libusb_control_transfer(libusb_device_handle *dev_handle,
	//uint8_t bmRequestType, uint8_t bRequest, uint16_t wValue, uint16_t wIndex,
	//unsigned char *data, uint16_t wLength, unsigned int timeout)
	if (LIKELY(ret == sizeof(data)))
		return UVC_SUCCESS;
	else
		return ret;
}`
Then, I write the get function becaust people from the camera factory told me to get cpu data from 0x101f, if result is 0x83, it's correct. And I add a get function just like the above one , but the result is always 0.
I read the source code many times ,but I fix it, so any help here? Or any advice?

enight0000, 2016-07-14:
Is there any way can I directly send byte date to camera?

saki4510t, 2016-08-05:
Hi,
Sorry for my answer delayed.
I assume it will be caused by color space conversion and there are several way to fix it.
I also assume most fast way is changing coefficient values from yuv to rgb(x) in `frame.c` to fit your camera.
saki

enight0000, 2016-11-10:
After several months testing, I finally find the way to control the rpg paramter.
at ctrl.c in line 1842, change the limie condition.

closed 2016-11-10 by enight0000