saki4510t/UVCCamera #75: UVCCamera not working with Android N

rockyzhang, 2016-08-24:
UVCCamera used to work on Nexus 5X, However, after upgrading Nexus 5X to Android N. There is a notification about USB devices is plugged and ask for permission. But Application can not display any information/video about USB camera.

saki4510t, 2016-08-25:
Hi,
I know the issue for several month. The cause of issue is some permissions related to usbfs(most of them are for hotplug) are dropped on Android N and later. 
Actually those permission and accessing related functions/data are not not necessary on Android because all of them should be done on Android System(`UsbManager`) on current Android Permission Model for USB. Unfortunately current libusb (and libuvc) requires and strongly depend on those functions/data and their permissions. 

I have other library to access USB device on Android fully wrote in C++ with object oriented architecture and works on Android N. So if we remove these dependancies from libusb/libuvc, they should work on Android N but it will takes a time.

rockyzhang, 2016-08-27:
Interesting! so the C++ lib is still able to access USB without accessing usbfs? Which file descriptor it is writing/reading? Is it supporting isochronous transfer?

saki4510t, 2016-08-27:
Hi,
Yes, as I tested, accessing USB device(s) using valid raw file descriptor came from `UsbDeviceConnection`(got by `UsbManager#openDevice`) itself is still possible including isochronous transfer(if the USB device and Android device support it).
My apps on Google Play include two type backend libraries to access USB device, one is libusb+libuvc on this repository and the other is my own library. And `Compatibility setting` is ON (on `Expert settings` on settings screen), my apps use libusb+libuvc (does not work on Android N now) and if the setting is OFF, my app use my own backend library and still works on Android N.
You can try [my app](https://play.google.com/store/apps/details?id=com.serenegiant.usbwebcamera)

rockyzhang, 2016-08-27:
I'll take a look, thanks!

--Rocky

CAMOBAP795, 2016-09-02:
> I have other library to access USB device on Android fully wrote in C++ with object oriented architecture and works on Android N.

Are these libraries open source?

saki4510t, 2016-11-15:
I can get images using this repository on Android N now and will update within several days.

saki4510t, 2016-11-20:
Now this repository works on AndroidM(6) and Android N(7) including dynamic permission model.

closed 2016-11-20 by saki4510t

SimonMarquis, 2016-11-20:
Awsome, huge refactoring but it worth it!
Could you tell us what are now the required runtime permissions to use a usb camera (Manifest.permission.CAMERA and Manifest.permission.RECORD_AUDIO ?)

saki4510t, 2016-11-20:
No, as same as before, `libuvccamera` module itself i  this repository still does not need any permission and you don't need `Manifest.permission.CAMERA` and `Manifest.permission.RECORD_AUDIO` to access USB/UVC devices.
But, still it will be better to write;
```
<uses-feature android:name="android.hardware.usb.host" android:required="true"/>
```
This is just a hint, not a permission request though.

Other permissions on sample apps are just for recording. If you need recording and/or still capture, you also need `RECORD_AUDIO`, 'MODIFY_AUDIO_SETTINGS' and/or 'WRITE_EXTERNAL_STORAGE'.

barak353, 2017-04-28:
Hey, i want to compile the libusb folder with the new pathces into .so file.
How can i do this? I tried using cygwin and the ndk-build, but it failled, then i tried to do this with android studio but i got to many error messages in this files.
Can you please include the Makefile that you have used, or explain how you compile the libusb folder into .so file?

barak353, 2017-04-28:
Hey Saki, can i take regular web cam and connect it to a phone that have the libuvccamera application installed on him. And it should work? even if the camera is running android 7.x?