saki4510t/UVCCamera #77: This project cann't work on MTK8665 chipset but work on Qualcomm8976

lookfuyao, 2016-08-27:
Hi Saki，
         I can run usbCameraTest2/4 success with modify preview size as 1280*720 on Alcatel Idol4 which base on Qualcomm8976 chipset. But fail on a tablet base on MTK8665. Did you know any information about this?
         Thank you very much!

lookfuyao, 2016-08-27:
Add more infomation. The tablet's SW is Android L.

saki4510t, 2016-08-28:
Hi,
The reason why? unfortunately it is just MediaTek, Allwiner, Rockchip and similar China SOC vendors do not want user use such USB devices like UVC camera. 
USB keyboard, mouse and USB memory will work if the device supports OTG function. 

Actually as I know there are at least 4 types of issue on those SOC.

1. OTG disabled or does not have OTG function
2. issue on buffer handling in kernel driver. Buffer overflow will occur and crash in kernel driver(and the device will reboot). Recent devices may not have this type issue.
3. Power supply. Some devices that the SOC is made by such China vendors can not supply enough USB current and/or overcurrent protection will malfunction. And the system  become unstable or sometimes reboot or just stop USB current.
4. issue on isochronous transfer. Detaching kernel driver can not work well or system always has all bandwidth and never release.

And your MTK8665 powered device will also have one or more above issue(s).

I had ask those vendors about those issues for more than a year, but there were no response and I think they don't want to fix them and it will be better to give up those type SOC until they fix.

lookfuyao, 2016-08-28:
Dear Saki，
      Thanks for your helpful answer.
      Now I test my MediaTek device, those 4 types of issue you said, I think I have trouble at NO4.
      If I can custom the device kernel, do you have any suggest for me to check what problem with the kernel. I'm a newer at Android kernel.
      Thanks!

saki4510t, 2016-08-31:
Hi,

If libuvc(libusb) failed to start isochronous transfer when start streaming, the issue will be No.4.
If you can get one or several frames, it might be No.2 or No.3.

Well I'm not so familiar with inside of kernel, as I had see sources of MTK kernel and kernel modules before, there were no significant difference than other kernel(and kernel modules) although I have no confidence... And I also assume the issue may exist on other part of Android System on those  devices such as pre-installed (system) app(s).

1. Does your MTK device is board computer like Raspberry Pi or similar device to Android TV?
Those type devices sometimes can access UVC camera as same way as usual internal camera. Does pre-installed camera app access to UVC camera?
2. If you can access your device with root permission and your device has (and enable) v4l2 kernel driver, you may be able to use UVC camera via v4l2 as same as usual Linux.

with best regards,
saki

lookfuyao, 2016-10-11:
Dear Saki,
        Thanks for you help, I'm give up MTK platform now, Thanks again!

closed 2016-10-11 by lookfuyao