saki4510t/UVCCamera #78: Segmentation fault on SoFIA tablet

MartynasPetkevicius, 2016-09-08:
When try opening usbCameraTest camera preview on my Rockchip Sofia 3GR tablet, I get the following error in Logcat:

```
09-08 09:02:05.653 5966-6012/com.serenegiant.usbcameratest I/libUVCCamera: [6012*UVCPreview.cpp:492:prepare_preview]:frameSize=(640,480)@YUYV
09-08 09:02:05.663 5966-6014/com.serenegiant.usbcameratest I/libUVCCamera: [6014*UVCPreview.cpp:258:callbackPixelFormatChanged]:PIXEL_FORMAT_RAW:
09-08 09:02:06.067 512-687/system_process E/WifiConfigStore: updateConfiguration freq=2472 BSSID=6c:72:20:d8:dc:6e RSSI=-56 "Techn"WPA_PSK
                                                             
                                                             --------- beginning of crash
09-08 09:02:07.321 5966-6011/com.serenegiant.usbcameratest A/libc: Fatal signal 11 (SIGSEGV), code 1, fault addr 0xe1300000 in tid 6011 (pool-1-thread-1)
09-08 09:02:07.427 204-204/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
09-08 09:02:07.427 204-204/? A/DEBUG: Build fingerprint: 'Android/sofia3gr_tablet_64/sofia3gr:6.0.1/MMB29M/szbnd07131002:userdebug/release-keys'
09-08 09:02:07.428 204-204/? A/DEBUG: Revision: '0'
09-08 09:02:07.428 204-204/? A/DEBUG: ABI: 'x86'
09-08 09:02:07.430 204-204/? A/DEBUG: pid: 5966, tid: 6011, name: pool-1-thread-1  >>> com.serenegiant.usbcameratest <<<
09-08 09:02:07.430 204-204/? A/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0xe1300000
09-08 09:02:07.473 204-204/? A/DEBUG:     eax 000007f8  ebx e177eec0  ecx 000001ca  edx df1d62bc
09-08 09:02:07.473 204-204/? A/DEBUG:     esi df1d638c  edi e1300000
09-08 09:02:07.473 204-204/? A/DEBUG:     xcs 00000023  xds 0000002b  xes 0000002b  xfs 0000006f  xss 0000002b
09-08 09:02:07.474 204-204/? A/DEBUG:     eip e1726144  ebp ec822ac0  esp e1506640  flags 00210202
09-08 09:02:07.484 204-204/? A/DEBUG: backtrace:
09-08 09:02:07.484 204-204/? A/DEBUG:     #00 pc 0001b144  /data/app/com.serenegiant.usbcameratest-1/lib/x86/libuvc.so
09-08 09:02:07.485 204-204/? A/DEBUG:     #01 pc 000096ec  /data/app/com.serenegiant.usbcameratest-1/lib/x86/libusb100.so (usbi_handle_transfer_completion+204)
09-08 09:02:07.485 204-204/? A/DEBUG:     #02 pc 00010f6a  /data/app/com.serenegiant.usbcameratest-1/lib/x86/libusb100.so
09-08 09:02:07.486 204-204/? A/DEBUG:     #03 pc 00009412  /data/app/com.serenegiant.usbcameratest-1/lib/x86/libusb100.so
09-08 09:02:07.486 204-204/? A/DEBUG:     #04 pc 0000a072  /data/app/com.serenegiant.usbcameratest-1/lib/x86/libusb100.so (libusb_handle_events_timeout_completed+306)
09-08 09:02:07.486 204-204/? A/DEBUG:     #05 pc 0000a257  /data/app/com.serenegiant.usbcameratest-1/lib/x86/libusb100.so (libusb_handle_events+71)
09-08 09:02:07.486 204-204/? A/DEBUG:     #06 pc 0001aa01  /data/app/com.serenegiant.usbcameratest-1/lib/x86/libuvc.so (_uvc_handle_events+97)
09-08 09:02:07.486 204-204/? A/DEBUG:     #07 pc 00095d0b  /system/lib/libc.so (_ZL15__pthread_startPv+59)
09-08 09:02:07.486 204-204/? A/DEBUG:     #08 pc 00030e8b  /system/lib/libc.so (__start_thread+27)
09-08 09:02:07.487 204-204/? A/DEBUG:     #09 pc 00017636  /system/lib/libc.so (__bionic_clone+70)
09-08 09:02:07.943 204-204/? A/DEBUG: Tombstone written to: /data/tombstones/tombstone_02
09-08 09:02:07.943 204-204/? E/DEBUG: AM write failed: Broken pipe
09-08 09:02:07.945 512-6019/system_process W/ActivityManager:   Force finishing activity com.serenegiant.usbcameratest/.MainActivity
```

Same build works fine on Samsung Galaxy S4. I saw it mentioned in other issues, that UVC support of some Chinese SoC manufacturer's is poor. Can this particular problem be fixed or worked around somehow?

saki4510t, 2016-09-08:
Hi,
As I can see your logCat, it may be different from issues on other China made SOC.
Because it seems the isochronous transfers themselves are successfully executed and app crash when handling received data, will be in libuvc.so.
Could you try `ndk-stack`  with `-sym` option to specify debug symbol?

Ex. 
```
ndk-stack -sym {path to your project}/libuvccamera/src/main/obj/local/x86 -dump logcat.txt > crash.txt
```
This will show you where the error occur with position on source.

with best regards,
saki

saki4510t, 2016-09-08:
Hi,
As I can see spec of your device, it seems to use 64bit processor.
Well actually I don't have tablet/smartphone that CPU is Intel's 64 bit processor.

If you build native libraries as x86-64 binary, could you try to drop `x86-64` off from `Application.mk` and use x86 binaries?
And if your device support Intel's binary translator, could you also try to use only `armeabi-v7a` as `APP_ABI`?

saki

MartynasPetkevicius, 2016-09-09:
I was not building x86-64 binaries and decided to try it. When I used `APP_ABI` option `all`, build failed with the following output:

```
[mips64] Compile        : jpeg-turbo1500_static <= jcmainct.c

jni/libjpeg-turbo-1.5.0/jchuff.c:387:2: error: #error Cannot determine word size
 #error Cannot determine word size
  ^
make: *** [obj/local/mips64/objs/jpeg-turbo1500_static/jchuff.o] Error 1
make: *** Waiting for unfinished jobs....

[mips64] Compile        : jpeg-turbo1500_static <= jcmarker.c
make: Leaving directory `C:/Projektai/escroll/UVCCamera-master/libuvccamera/src/main'
:libuvccamera:ndkBuild FAILED
```

I then tried using `x86_64` option only. Build succeeded, but I couldn't install the APK because of the error `INSTALL_FAILED_NO_MATCHING_ABIS`. Can this happen on 64 bit device?

I then tried your other suggestion and left only `armeabi-v7a` option. This time build succeeded and was even able to start camera preview without app crashing. However, I could only see black surface where the preview was supposed to be and only after updating UI somehow (by opening recent apps, for instance, and then navigating back.

I also ran the log I posted earlier through `ndk-stack`. Here's the output:

```
********** Crash dump: **********
Build fingerprint: 'Android/sofia3gr_tablet_64/sofia3gr:6.0.1/MMB29M/szbnd07131002:userdebug/release-keys'
pid: 5966, tid: 6011, name: pool-1-thread-1  >>> com.serenegiant.usbcameratest <<<
signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0xe1300000
Stack frame 09-08 09:02:07.484 204-204/? A/DEBUG:     #00 pc 0001b144  /data/app/com.serenegiant.usbcameratest-1/lib/x86/libuvc.so: Routine _uvc_process_payload_iso at C:\Projektai\escroll\UVCCamera-master\libuvccamera\src\main/jni/libuvc/android/jni/../../src/stream.c:918
Stack frame 09-08 09:02:07.485 204-204/? A/DEBUG:     #01 pc 000096ec  /data/app/com.serenegiant.usbcameratest-1/lib/x86/libusb100.so (usbi_handle_transfer_completion+204): Routine usbi_handle_transfer_completion at C:\Projektai\escroll\UVCCamera-master\libuvccamera\src\main/jni/libusb/android/jni/../../libusb/io.c:1603
Stack frame 09-08 09:02:07.485 204-204/? A/DEBUG:     #02 pc 00010f6a  /data/app/com.serenegiant.usbcameratest-1/lib/x86/libusb100.so: Routine handle_iso_completion at C:\Projektai\escroll\UVCCamera-master\libuvccamera\src\main/jni/libusb/android/jni/../../libusb/os/android_usbfs.c:2640
Stack frame 09-08 09:02:07.486 204-204/? A/DEBUG:     #03 pc 00009412  /data/app/com.serenegiant.usbcameratest-1/lib/x86/libusb100.so: Routine handle_events at C:\Projektai\escroll\UVCCamera-master\libuvccamera\src\main/jni/libusb/android/jni/../../libusb/io.c:2109
Stack frame 09-08 09:02:07.486 204-204/? A/DEBUG:     #04 pc 0000a072  /data/app/com.serenegiant.usbcameratest-1/lib/x86/libusb100.so (libusb_handle_events_timeout_completed+306): Routine libusb_handle_events_timeout_completed at C:\Projektai\escroll\UVCCamera-master\libuvccamera\src\main/jni/libusb/android/jni/../../libusb/io.c:2194
Stack frame 09-08 09:02:07.486 204-204/? A/DEBUG:     #05 pc 0000a257  /data/app/com.serenegiant.usbcameratest-1/lib/x86/libusb100.so (libusb_handle_events+71): Routine libusb_handle_events at C:\Projektai\escroll\UVCCamera-master\libuvccamera\src\main/jni/libusb/android/jni/../../libusb/io.c:2270
Stack frame 09-08 09:02:07.486 204-204/? A/DEBUG:     #06 pc 0001aa01  /data/app/com.serenegiant.usbcameratest-1/lib/x86/libuvc.so (_uvc_handle_events+97): Routine _uvc_handle_events at C:\Projektai\escroll\UVCCamera-master\libuvccamera\src\main/jni/libuvc/android/jni/../../src/init.c:107
Stack frame 09-08 09:02:07.486 204-204/? A/DEBUG:     #07 pc 00095d0b  /system/lib/libc.so (_ZL15__pthread_startPv+59)
Stack frame 09-08 09:02:07.486 204-204/? A/DEBUG:     #08 pc 00030e8b  /system/lib/libc.so (__start_thread+27)
Stack frame 09-08 09:02:07.487 204-204/? A/DEBUG:     #09 pc 00017636  /system/lib/libc.so (__bionic_clone+70)
```

Thank you for the help.

saki4510t, 2016-09-10:
Hi
As I ca see your crash dump, Segmentation fault occurred at `stream.c:918` in libuvc.
That is
https://github.com/saki4510t/UVCCamera/blob/master/libuvccamera/src/main/jni/libuvc/src/stream.c#L918

I assume buffer overflow will occur because of invalid received data.  
Could you try debug build instead of release build so that assertion before line 918 are enabled or confirm received values like `pkt->actual_length`, 'header_len'.
You can change to debug build with 'APP_OPTIM := debug' in `Application.mk`.

Regarding `INSTALL_FAILED_NO_MATCHING_ABIS`, I assume your device only support x86(and armeabi-v7a) abi although your device use 64 bits processor on spec. on Web.

saki

MartynasPetkevicius, 2016-09-10:
I can tell your assumption was right, here's what I've got in the log:

```
09-10 11:22:51.506 204-204/? A/DEBUG: signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
09-10 11:22:51.543 204-204/? A/DEBUG: Abort message: 'jni/libuvc/android/jni/../../src/stream.c:915: _uvc_process_payload_iso: assertion "strmh->got_bytes + odd_bytes < strmh->size_buf" failed'
```

What does this mean? Is this driver's fault?

saki4510t, 2016-09-11:
Hi,
it means libuvc received more data that it is expected and there are several possibilities.
Could you try to check values like `pkt->actual_length` when the assert condition hits, `if (strmh->got_bytes + odd_bytes >= strmh->size_buf)'?
If the values are acceptable value..., I assume force drop several first frames may help you.
Or the value is too huge, libusb or driver of your device has some mistakes on conversion of endian.

saki

MartynasPetkevicius, 2016-09-12:
I added logging at `stream.c:918` like this:

```
__android_log_print(ANDROID_LOG_VERBOSE, "UVC", "%d + %d %s %d", strmh->got_bytes, odd_bytes, strmh->got_bytes + odd_bytes < strmh->size_buf ? "<" : ">=", strmh->size_buf);
memcpy(strmh->outbuf + strmh->got_bytes, pktbuf + header_len, odd_bytes);
```

[Here's](https://github.com/saki4510t/UVCCamera/files/466765/buffer_sizes.txt) the output I got. Can you help me figure it out? I'm not too familiar with low level details of UVC.

saki4510t, 2016-09-12:
Hi,
As I can see the file, I feel received bytes are no problem but libuvc can not detect end of frame.
Could you check frame ID bit(you can use `UVC_STREAM_FID` constant) and EOF bit (you can use `STREAM_HEADER_BFH_EOF` constant) on `header_info(=pktbuf[1])`?
If you can't see those flags, your camera might never set the flag(s) or driver of your device drops.

And other possibilities are `is_isight flag` hits or receive the flag(s) with `UVC_STREAM_ERR` flag or... So adding log output to check those before `continue` in `_uvc_process_payload_iso` function may also help you.

FYI, you can use `LOGV`, `LOGD`, `LOGI`, `LOGW`, `LOGE` and `MARK` macros instead of directly use `__android_log_print(ANDROID_LOG_VERBOSE, ...` etc. because `__android_log_print ` is too long to use.
You can control these macros with combination of `USE_LOGALL`, `USE_LOGV`, `USE_LOGD`, `USE_LOGI`, `USE_LOGW`, `USE_LOGE`, `USE_LOGF` macros in `localdefines.h` and `LOCAL_CFLAGS += -DLOG_NDEBUG` in `Android.mk` etc. (`USE_LOGALL `, `LOGV` and `LOGD` are disabled on default settings and those code never include in resulted binaries).
Those macros and available conditions and how they work are in `utilbase.h`.

saki

MartynasPetkevicius, 2016-09-13:
I'm not sure if I've done it correctly, here's the code I used at line 849:

```
header_info = pktbuf[1];
LOGI("UVC_STREAM_FID: %d, STREAM_HEADER_BFH_EOF: ?, is_isight: %d, UVC_STREAM_ERR: %d\n",
        header_info & UVC_STREAM_FID,
        /*header_info & STREAM_HEADER_BFH_EOF,*/
        strmh->devh->is_isight,
        header_info & UVC_STREAM_ERR);
```

It didn't compile if I used `STREAM_HEADER_BFH_EOF`, said the constant is used for the first time or something like that. [Here](https://github.com/saki4510t/UVCCamera/files/469692/flags.txt) is the output I've got.

saki4510t, 2016-09-13:
Hi
Sorry I mixed libuvc with my own USB backend library. `STREAM_HEADER_BFH_EOF` is for my own backend. It is same value as `UVC_STREAM_EOF`.

As I can see your result and if you can not see `UVC_STREAM_EOF` flag in `header_info `(this will be most possible situation because libuvc on my repository already check this flag on line 922, `USE_EOF` macro is always defined.), only you can do are;

1. call `_uvc_swap_buffers` when `strmh->got_bytes` reached to a specific value.  
    Unfortunately this way is only available when you use YUY2 format. You can't use this way for MJPEG format.
2. As I can see previous log, there are two types of `odd_bytes` value. One is 12 - several hundreds, the other is more than 2000.  
    So I assume when the `odd_bytes` changed from the former to the latter or the latter to the former will be a breakpoint between frame data. But this is not fit to UVC specification.

Well same camera works with Samsung Galaxy S4?
If so, could you try to remove any `LOG_NDEBUG` macro definition from `Android.mk` etc.?
And if you still can't see other error message on logCat, driver software on your device will drop header info from each payload.

saki

MartynasPetkevicius, 2016-09-14:
Yes, `UVC_STREAM_EOF` is always `0` too. I tried to follow your suggestion to `_uvc_swap_buffers` and made the following change at lines 900-something:

```
static size_t previous_odd_bytes = 12;

...

				memcpy(strmh->outbuf + strmh->got_bytes, pktbuf + header_len, odd_bytes);
				strmh->got_bytes += odd_bytes;
			}
#ifdef USE_EOF
            const size_t odd_bytes = pkt->actual_length - header_len;
            LOGI("Odd bytes difference: %d\n", abs(previous_odd_bytes - odd_bytes));
			if ((pktbuf[1] & UVC_STREAM_EOF) && strmh->got_bytes != 0 || abs(previous_odd_bytes - odd_bytes) > 2000) {
				/* The EOF bit is set, so publish the complete frame */
				_uvc_swap_buffers(strmh);
			}
			previous_odd_bytes = odd_bytes;
#endif
```

It no longer crashes, but when I start the preview, it's of continuous [image] and doesn't change at all when I move camera. Does that tell anything?

I also tried commenting out the line `LOCAL_CFLAGS += -DLOG_NDEBUG` from `libuvccamera\src\main\jni\UVCCamera\Android.mk`, but I didn't find anything new in the app logs.

saki4510t, 2016-09-19:
Hi,
I remember the green screen means all bytes of received data are zero in YUV2 pixel format.
I assume it may be better to dump all received bytes into file and check by binary editor etc.

By the way, I remember you use `YUV2` instead of `MJPEG`. Do you have special reason?
If you can use `MJPEG` as pixel format, usual `MJPEG` has `JFIF` structure and has several markers. And app will be able to know when the image starts(`SOI`) and ends(`EOI`) even if `UVC_STREAM_EOF` is always zero and `UVC_STREAM_FID` never flip. (App needs to scan all bytes to find markers and app become slow, CPU load increase on this approach so I don't recommend this though)

If all received bytes are zero or can not parse as `YUV2`/`MJPEG` pixel format, kernel driver of your device will not be able to handle isochronous transfer correctly.

saki

MartynasPetkevicius, 2016-09-19:
I tried logging the sum of all `pktbuf` bytes and it's always 0. I also tried plugging in a camera that supports MJPEG, but results are the same.

Is UVC driver to blame here? If so, should I try reinstalling it? Can it be caused by faulty USB driver? Can there be a bug in `libusb`? Can this be permissions or power issue?

saki4510t, 2016-09-23:
Hi,

I get SoFIA 3-GR(`Intel C3230 RK`) device and...

1.  As I can see data between camera and the device by USB analyzer, communication between camera and the device itself seems to work well including isochronous transfer.  
    But unfortunately app can not receive video images(all packet were broken or zero length).
2. I have different USB backend library other than `libusb`/`libuvc` and it has same issue.
3. I also tried different app that does not use `libusb`/`libuvc` and could not work well.
4. As I can see informations on WEB, `Intel C3230 RK` is made in other than Intel's fab (same fab as other Rockchip's SoC, will be TSMC's).
5. I had test with different tablet that SoC is Intel's Atom x3-C3200(I think this is made in Intel's fab) and works well so this is not a x3-C3000 series issue but issue(s) on peripheral part and/or kernel driver.

So I can say this is a new kind of issue on China made SoC related to isochronous transfer.

Unfortunately it is hard to replace system software (including kernel / kernel driver(s)) on usual Android unless you can get root permission and unlock boot loader and get all sources including kernel driver or someone make custom ROM.
As I can see [XDA FORUMS](http://www.xda-developers.com), there seems to be custom kernel for some devices that SoC is C3230 RK. So it will be possible although I'm not a kernel developer and can not help that.

saki

MartynasPetkevicius, 2016-09-23:
Thanks so much for your help.

closed 2016-10-20 by saki4510t