saki4510t/UVCCamera #79: how to record video every 10 minutes

lsy17096535, 2016-09-19:
hi, saki4510t
    Thanks for your project. when use Timer in project usbcameratest4 to get video every 10 minutes, I found that the muxer will not be released. When i start a new video, the former one have not be closed.
when i push the stopservice button, the code will close the video completely. could you tell me how to record video every 10 minutes in usbcameratest4? thanks again.
                                                                              syan

jp1017, 2017-02-12:
@lsy17096535 The latested version makes well. :smile:

jp1017, 2017-02-12:
@lsy17096535 like this: 

https://github.com/saki4510t/UVCCamera/issues/73

lsy17096535, 2017-02-13:
@jp1017 Thank you !

closed 2017-02-13 by lsy17096535