saki4510t/UVCCamera #83: not display preview

daydreamer219, 2016-09-24:
Hi, saki.
I have used usbcameratest0 app on google glass(Android os 5.1.1) that is connected to webcam.
But I got "connected", "attached" message. And blank screen.
After 20s, I got following logs.

`InputDispatcher     channel 'af7b444 com.serenegiant.usbcameratest0 / com.serenegiant.usbcameratest0.MainActivity (server)' - Channel is unrecoverably broken and will be disposed.`

and the app died.
How can I fix it?
Thanks.

saki4510t, 2016-09-24:
Hi,
As your log shows that app deadlocked in somewhere.
Unfortunately I don't have google glass so this is just assuming, I feel it will be in `SurfaceHolder.Callback`, may be in line 225-226.
If so, try to run line 223-229 on EXECUTER's thread as same way as in `onConnect` method.

FYI,
There are several way that you can find where app deadlocks. Most simple one is **printf debug**, just add logout (`Log.v` or `Log.d` etc.) on anywhere.

Second way is using `Android Device Monitor` of SDK. If you have used `Eclipse`, this is almost same interface as `DDMS` screen in `Eclipse`.
After launching `Android Device Monitor`, please try to select app process and click `Update Threads`. Now you will see attached threads to the process in `Threads` tab and click a thread to show stack trace.

Most detail one is using `adb bugreport` command.
In this way, you need to connect `adb` over network first. (Ref. [How to debug over network](https://github.com/saki4510t/UVCCamera/wiki/howto_debug_over_network) ).
Then type following on your shell(command prompt) while app deadlocks (of course you can use any file name for redirect).
```
adb bugreport > bugreport.txt
```
This will generate huge file, ...more than some MBs and will take a time.
You can see all stack trace and try to find threads that tries to lock same mutex.

saki

daydreamer219, 2016-09-27:
if I set the DEFAULT_PREVIEW_WIDTH and DEFAULT_PREVIEW_HEIGHT from 640*480 to 320*240, the error occurs after 40s. if the size is 640*480, the error occurs after 20s.  And preview is black screen.
What is the problem? Thanks.

saki4510t, 2016-10-10:
Hi,

> InputDispatcher channel 'af7b444 com.serenegiant.usbcameratest0 / com.serenegiant.usbcameratest0.MainActivity (server)' - Channel is unrecoverably broken and will be disposed.

Above one (on your first message) is not a actual error on the app itself, it is just a result caused by ANR. Actual ANR(deadlock) occurred before the message. I assume one of the most possible reason of ANR will be deadlock in OpenGL|ES system library on your glass.