saki4510t/UVCCamera #85: no display preview and get the log "supportedSize:null"

codeinging, 2016-10-08:
First, thanks you open library ,I clone you project and open it in android studio,I find one issues:in some devices all module works great but in some devices,such as Nexus 5X, vivo and hongmi note 1,it can detact the usbcamera but no display preview and i get the same log "supportSize:null",how can i solve the problem?Thanks!

saki4510t, 2016-10-09:
Hi,

If your `Nexus5X` is Android N or later, `libuvc`(that this repository depends on) does not work well now. see #75 .
And the SoC of `Hongmi Note` is made by MediaTek, it will be `MT6592M` or `MT6592` and some features like isochronous transfer that this repository needs do not work well. see #38, #52, #77, #78 etc.

Initializing camera will take a long time, it will be 1-3 seconds, when using some combination of Android devices and cameras you may need special handling after open camera (before try to get any data from camera) on those combination.
saki

codeinging, 2016-10-10:
Hi Saki,
 Thanks for your helpful answer!

CAMOBAP795, 2016-11-07:
> And the SoC of Hongmi Note is made by MediaTek, it will be MT6592M or MT6592 and some features like isochronous transfer that this repository needs do not work well.

@saki4510t would you kindly clarify 
1. Is this hardware (SoC) or software issue (kernel)?
2. Is there any way to detect such devices (whiteout own black list)?

saki4510t, 2016-11-07:
HI,
As my experience the issue will be on kernel driver in these days.

Some Android TV or Android stick use SoC by MediaTek and/or similar China vendors. And pre-installed camera app many times can access external camera as `internal camera` on those devices (also many camera apps that can handle usual `internal camera` on Google Play often become unstable on those devices though)
But this repository and my apps(and similar apps came from other competitors) can not work.
So I assume main issues are on their kernel driver, I assume one/some/all of followings;  
The kernel driver

1. can not handle isochronous transfer correctly  
    As I know some devices corrupt buffer(over run/under run) while handing isochronous data in kernel driver(and crash/restart).
2. can not handle detach/attach request
3. always hold all bandwidth of USB. 
4. issues on `ioctl()`
5. others

(those reasons sometimes depend on each other)

Unfortunately no. Only ways are;

1. omit by SoC vender name by programmatically  
    (or omit by device name on Google Play...this is bad work though)
2. try and error

djamb, 2016-11-11:
what are u using in your app in the store?.  is it based in other c library for usb?

saki4510t, 2016-11-20:
Hi,
My apps include two backend libraries to access USB devices, one is `libusb+libuvc` and the other is my own library written by C++.
Today I updated this repository to support Android N(7.x). So you will able to access UVC devices even if your device runs Android 7.x.

closed 2016-11-20 by saki4510t