saki4510t/UVCCamera #86: Set Custom Preview Size

stelian008, 2016-10-09:
Hello, I'm very happy to use your UVC Camera.
However, unfortunately, I can't set preview size to my desiring size such as 1024 * 1280.
The app works when I set it 640 * 480, but it is crashed when I set size to 1024 * 1280.
How can I fix it?

Sincerely,
Stelian

saki4510t, 2016-10-09:
Hi,

1. Does your camera supports the resolution?
2. Which sample app do you base on?
3. Where the app / library crash?, logCat will help you.
4. Where and how do you modify?

saki

saki4510t, 2016-11-20:
Hi,
If your camera does not support 1024 * 1280, you can't set the resolution.
You can get available resolutions and frame rates that active camera supports using `UVCCamera#getSupportedSize` as JSON string, using `UVCCamera#getSupportedSizeList` as `List<Size>`. And you can select one one them using `UVCCamera#setPreviewSize`.
saki