saki4510t/UVCCamera #88: How to compatible on Mediatek turnkey bsp

wormboss, 2016-10-19:
Hi Master,
I am studying porting UVC feature on my own device, Mediatek 6735 (mt6735) turnkey solution.
The android version is 5.1 and kernel version is 3.10.61.

I have enabled the uvc-related configs in Kconfig, also /dev/video0 is visible when I plug in the webcam.
However, I only get the "whole black" frame just like nothing captured from framebuffer when use your exampl app, usbcameratest.

Would you please give some advice or debug guide?

Thanks.

adb logcat msg is as follows in brief
-----

10-19 17:14:03.089 W/linker  ( 2904): /data/app/com.serenegiant.usbcameratest-1/lib/arm64/libusb100.so: unused DT entry: type 0x6fffffff arg 0x1

10-19 17:14:03.091 W/linker  ( 2904): /data/app/com.serenegiant.usbcameratest-1/lib/arm64/libuvc.so: unused DT entry: type 0x6ffffffe arg 0x4a78

10-19 17:14:03.093 W/linker  ( 2904): /data/app/com.serenegiant.usbcameratest-1/lib/arm64/libUVCCamera.so: unused DT entry: type 0x6ffffffe arg 0x18a0

10-19 17:14:03.202 I/libUVCCamera( 2904): [descriptor.c:867:libusb_get_device_descriptor]:desc=0x7f92aae610,dev=0x7fa136a3e0,device_descriptor=0x7fa136a448

0-19 17:14:05.550 I/EventHub(  883): New device: id=12, fd=246, path='/dev/input/event9', name='USB 2.0 PC Camera', classes=0x80000001, configuration='', keyLayout='/system/usr/keylayout/Generic.kl', keyCharacterMap='/system/usr/keychars/Generic.kcm', builtinKeyboard=false, wakeMechanism=EPOLLWAKEUP, usingClockIoctl=true

10-19 17:14:05.550 I/InputReader(  883): Device added: id=12, name='USB 2.0 PC Camera', sources=0x00000101

10-19 17:14:05.551 D/Surface ( 2904): Surface::setBuffersDimensions(this=0x7fa71c7e00,w=540,h=864)

10-19 17:14:05.553 V/ViewRootImpl( 2904): ViewRoot performTraversals and draw- : mResizeBuffer = null, mResizeAlpha = 0, this = ViewRoot{1fbd714b com.serenegiant.usbcameratest/com.serenegiant.usbcameratest.MainActivity,ident = 0} frame#7 (cancelDraw = false)

10-19 17:14:05.554 D/InputReader(  883): InputReader:: getSwitchState lock 

10-19 17:14:05.554 D/InputReader(  883): InputReader::getStateLocked:: return 

10-19 17:14:05.555 V/ViewRootImpl( 2904): scheduleTraversals: mTraversalBarrier = 17,this = ViewRoot{1fbd714b com.serenegiant.usbcameratest/com.serenegiant.usbcameratest.MainActivity,ident = 0}

10-19 17:14:05.555 V/DisplayEventReceiver( 2904): call nativeScheduleVsync mReceiverPtr = 547960936832

10-19 17:14:05.556 D/SurfaceFlinger(  258): requestNextVsync(2904)

10-19 17:14:05.556 D/GraphicBuffer(  258): free, handle(0x7f9ac127a0) (w:640 h:480 s:640 f:0x1 u:0x000133)

10-19 17:14:05.557 D/SurfaceFlinger(  258): requestNextVsync(258)

10-19 17:14:05.559 D/FeatureProxyBase(  883): FeatureProxyBase class constructor

10-19 17:14:05.560 D/FeatureProxyBase(  883): getService(), serviceName = multiwindow_service_v1

10-19 17:14:05.563 V/DisplayEventReceiver(  883): call nativeScheduleVsync mReceiverPtr = 547960716608

10-19 17:14:05.563 D/SurfaceFlinger(  258): requestNextVsync(883)

10-19 17:14:05.578 D/SurfaceFlinger(  258): postEvent(258, v/c=2658)

10-19 17:14:05.578 D/SurfaceFlinger(  258): postEvent(883, v/c=1651)

10-19 17:14:05.578 D/SurfaceFlinger(  258): postEvent(2904, v/c=1651)

10-19 17:14:05.578 D/DisplayEventReceiver( 2904): rev(0x7f95045980) w/ vsync, t=225766914803, id=0, c=1651

10-19 17:14:05.578 D/DisplayEventReceiver(  883): rev(0x7f9500fd40) w/ vsync, t=225766914803, id=0, c=1651

10-19 17:14:05.579 V/ViewRootImpl( 2904): ViewRoot performTraversals+ : mFirst = false, this = ViewRoot{1fbd714b com.serenegiant.usbcameratest/com.serenegiant.usbcameratest.MainActivity,ident = 0}

10-19 17:14:05.581 D/OpenGLRenderer( 2904): Updating texture layer 0x7fa10dfa80, 540x404, fbo 0, texture 2, renderTarget 0x8d65, alpha 255

10-19 17:14:05.581 D/OpenGLRenderer( 2904): Layer 0x7fa10dfa80 bindTexture: glBindTexture(0x8d65, 2)

10-19 17:14:05.582 D/OpenGLRenderer( 2904): DrawFrameTask 0x7f95073298, canUnblockUiThread 1, canDrawThisFrame 1

10-19 17:14:05.582 D/Surface ( 2904): Surface::setBuffersDimensions(this=0x7fa71c7e00,w=540,h=864)

10-19 17:14:05.584 V/ViewRootImpl( 2904): ViewRoot performTraversals and draw- : mResizeBuffer = null, mResizeAlpha = 0, this = ViewRoot{1fbd714b com.serenegiant.usbcameratest/com.serenegiant.usbcameratest.MainActivity,ident = 0} frame#8 (cancelDraw = false)

saki4510t, 2016-10-20:
Hi,
This repository is designed to access UVC device without root permission on Android and does not use `v4l2` and use `libusb/libuvc` instead.
If you have root access and can prepare kernel driver of v4l2, I assume it will be better to use v4l2 as same way as usual Linux.
Because as I know and as you can see some issues on this repository, most SoC made by MediaTek and similar China vendors have issues on their hardware/system software and we can not access USB devices via `libusb/libuvc`.

saki4510t, 2016-11-20:
Close this because as I know this can not solve until Mediatek want to fix their issue,

closed 2016-11-20 by saki4510t