saki4510t/UVCCamera #90: android ndk failed to build libuvccamera

Domantas97, 2016-11-05:
I have a problem with building (gradle build) https://github.com/saki4510t/UVCCamera sample applications.

When i am building application **with console** (gradle build command), shows this error:

```
    What went wrong:
    Execution failed for task ':libuvccamera:ndkBuild'.
    > Process 'command 'C:\Program Files\Android_NDK\android-ndk-r13b/ndk-build.cmd'' finished with non-zero exit value 1
```

When i am building application **with Android Studio**, shows this error:

    `Error:(38, 0) Cannot invoke method getAbsolutePath() on null object`

**38 Line:** 

    `def ndkBuildingDir = plugins.getPlugin('com.android.library').sdkHandler.getNdkFolder().getAbsolutePath()`

full build.gradle file:
```

    apply plugin: 'com.android.library'
    
    import org.apache.tools.ant.taskdefs.condition.Os
    
    android {
        compileSdkVersion 22
        buildToolsVersion "22.0.1"
    
        defaultConfig {
            minSdkVersion 14
            targetSdkVersion 22
        }
    
        buildTypes {
            release {
                minifyEnabled false
                proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.txt'
            }
        }
    	sourceSets {
    		main {
    			jniLibs.srcDir 'src/main/libs'
    			jni.srcDirs = []
    		}
    	}
    }
    
    dependencies {
        compile fileTree(dir: new File(buildDir, 'libs'), include: '*.jar')
    }
    
    tasks.withType(JavaCompile) {
    	compileTask -> compileTask.dependsOn ndkBuild
    }
    
    task ndkBuild(type: Exec, description: 'Compile JNI source via NDK') {
    	println('executing ndkBuild')
    	def ndkBuildingDir = plugins.getPlugin('com.android.library').sdkHandler.getNdkFolder().getAbsolutePath()
    	def ndkBuildPath = ndkBuildingDir
    	if (Os.isFamily(Os.FAMILY_WINDOWS)) {
    		ndkBuildPath = ndkBuildingDir + '/ndk-build.cmd'
    	} else {
    		ndkBuildPath = ndkBuildingDir + '/ndk-build'
    	}
    	commandLine ndkBuildPath, '-j8', '-C', file('src/main').absolutePath
    }

```

saki4510t, 2016-11-05:
Hi,
I assume you didn't set correct path of ndk directory to `ndk.dir` property in `local.properties` file.

Domantas97, 2016-11-05:
Hi saki4510t,

my local.properties file:
```
sdk.dir=C:/Users/D/AppData/Local/Android/sdk
ndk.dir=C:/Program Files/Android_NDK/android-ndk-r13b
```

I download android ndk and extract in program files, then copied link and past it here.
It's how i should do that?

Domantas97, 2016-11-06:
Thank you! i change ndk directory and build successfully applications.

saki4510t, 2016-11-07:
Hi
Your NDK path includes space(between `Program` and `Files`) and I assume that is the reason. I feel there is a workaround on script side but the best way is just to avoid those directories.

closed 2016-11-07 by saki4510t

CL-LCQ, 2020-06-03:
Hi,

I've got the same here:
`* What went wrong:
Execution failed for task ':libuvccamera:ndkBuild'.
> A problem occurred starting process 'command '/Users/XX/Library/Android/android-ndk-r21c /ndk-build''
`
local.properties
`sdk.dir=/Users/XX/Library/Android/sdk
ndk.dir=/Users/XX/Library/Android/android-ndk-r21c 
`

jiaoery, 2020-07-17:
hello,i've get the same error and i correct the ndk dir path .my AS is 3.6.1.the error msg is:
```
D:/SDK/ndk-bundle/build//../build/core/setup-app.mk:76: *** Android NDK: Aborting    .  Stop.
make: Entering directory 'D:/work/newhuoyan-zhengzhou/libuvccamera/src/main'
Android NDK: android-14 is unsupported. Using minimum supported version android-16.    
Android NDK: WARNING: APP_PLATFORM android-16 is higher than android:minSdkVersion 1 in ./AndroidManifest.xml. NDK binaries will *not* be compatible with devices older than android-16. See https://android.googlesource.com/platform/ndk/+/master/docs/user/common_problems.md for more information.    
Android NDK: The armeabi ABI is no longer supported. Use armeabi-v7a.    
Android NDK: NDK Application 'local' targets unknown ABI(s): armeabi mips    
Android NDK: Please fix the APP_ABI definition in ./jni/Application.mk    
make: Leaving directory 'D:/work/newhuoyan-zhengzhou/libuvccamera/src/main'

> Task :libuvccamera:parseReleaseLibraryResources
> Task :libuvccamera:processReleaseManifest

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':libuvccamera:ndkBuild'.
> Process 'command 'D:\SDK\ndk-bundle/ndk-build.cmd'' finished with non-zero exit value 2
```

DivyaBotcha, 2020-08-14:
> Thank you! i change ndk directory and build successfully applications.

hi what was the path you given for ndk, please share

silvioalvim, 2020-12-08:
> Hi,
>
> I've got the same here:
> `What went wrong:
> Execution failed for task ':libuvccamera:ndkBuild'.
>
>
> local.properties
> `sdk.dir=/Users/XX/Library/Android/sdk [...]

* What went wrong:
Execution failed for task ':libuvccamera:ndkBuild'.
> Process 'command '/Users/xx/Library/Android/sdk/ndk/21.3.6528147/ndk-build'' finished with non-zero exit value 2

Almost same here. Do you fixed your problem? Any hint?

alienyang, 2022-03-31:
Hi,

Update [APP_ABI](https://developer.android.google.cn/ndk/guides/application_mk#app_abi) in libuvccamera\src\main\jni\Application.mk can fix this problem.
APP_ABI := armeabi-v7a arm64-v8a x86 x86_64

chenzf11, 2023-10-27:
> Hi,
> 
> Update [APP_ABI](https://developer.android.google.cn/ndk/guides/application_mk#app_abi) in libuvccamera\src\main\jni\Application.mk can fix this problem. APP_ABI := armeabi-v7a arm64-v8a x86 x86_64

thank you! I successfully build the project!