saki4510t/UVCCamera #91: Unable to start preview on some devices

SimonMarquis, 2016-11-14:
Hi,
I'm unable to start the video preview on some devices such as Nexus 6 and Nexus 9 (works fine on other devices though).

The only trace I have currently is coming right after the call of this method:

`mUVCCamera.open(ctrlBlock);`

> I/USBMonitor: UsbControlBlock:constructor
> I/USBMonitor: UsbControlBlock:name=/dev/bus/usb/003/002, desc=52, rawDesc=[B@c44b0db
> W/package: type=1400 audit(0.0:445): avc: denied { create } for scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:r:untrusted_app:s0:c512,c768 tclass=netlink_kobject_uevent_socket permissive=0
> W/package: type=1400 audit(0.0:446): avc: denied { read } for name="devices" dev="sysfs" ino=6636 scontext=u:r:untrusted_app:s0:c512,c768 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=0

Any idea?

saki4510t, 2016-11-15:
Hi,
I assume your Nexus6 and Nexus7 run Android N. So this is same as #75 .
Trying to run libusb on Android N or latter is still under development and does not work well yet.

SimonMarquis, 2016-11-15:
Thanks, I'll keep watching the #75 instead.

closed 2016-11-15 by SimonMarquis