saki4510t/UVCCamera #95: Adjust Brightness, Contrast & White Balance

bayudwirp, 2016-12-20:
Hello, @saki4510t 
Firstly, thankyou for your shared library, it means a lot for me, to help me build an usb camera apps.
I have a question. How can i adjust contrast, brightness and white balance setting? Could you share your example project, to adjust contrast, brightness and white balance?
Thanks before, for your answers.
Regards, Bayu.

williamdwarke, 2016-12-27:
@bayudwirp 
I recently wrote these features into an app so I'd like to chime in and hopefully save you the time and energy of figuring it out yourself. If you take a look in the library file **libuvccamera/src/main/java/com/serenegiant/usb/UVCCamera.java**, you will find a plethora of get, set, and reset functions for various features, including contrast, brightness, and white balance. You can access these functions from your UVCCamera object within your application. However, I couldn't get anything to work without calling updateCameraParams() after a set/reset call or before a get call.

For example, using the included usbCameraTest3 app from which my app is derived, I made the following calls from within my CameraThread:
```
mUVCCamera.resetBrightness();
mUVCCamera.updateCameraParams();
brightness = mUVCCamera.getBrightness();
```

Just remember that by default, white balance is set to auto, so you will have to disable auto mode before trying to set the value. Example:
			
```
if (mUVCCamera.getAutoWhiteBalance()) {
	mUVCCamera.setAutoWhiteBalance(false);
}
mUVCCamera.setWhiteBalance(whiteBalance);
```
Hope this helps!

bayudwirp, 2016-12-28:
Hi @williamdwarke !
Thankyou very much for help me to solve my problem.
I've tried your suggestion but, i got some problem :
1. If i try to disable the auto white balance mode, and I try to check return value from the previous disable command (in debug mode) the return value is true (though, it should be false, after I set before)
2. The same event also, when I try to change the value brigthness, and when i entered a value (using seekbar) was no problem, but there is no effect change in picture on camera

To describe my problem, I attach the following script i edited from an example project "USBCameraTest" 

	```
private final Object mSync = new Object();
    // for accessing USB and USB camera
    private USBMonitor mUSBMonitor;
	private UVCCamera mUVCCamera;
	private SimpleUVCCameraTextureView mUVCCameraView;
	// for open&start / stop&close camera preview
	private ImageButton mCameraButton;
	private SeekBar set_brightness;             //For Brightness Seekbar
	private Surface mPreviewSurface;
	private boolean check_result;
	final UVCCamera UVC_Camera = new UVCCamera();

	@Override
	protected void onCreate(final Bundle savedInstanceState) {
		super.onCreate(savedInstanceState);
		setContentView(R.layout.activity_main);
		mCameraButton = (ImageButton)findViewById(R.id.camera_button);
		mCameraButton.setOnClickListener(mOnClickListener);

		if (UVC_Camera.getAutoWhiteBlance()) {
			UVC_Camera.setAutoWhiteBlance(false);
		}
		check_result = UVC_Camera.getAutoWhiteBlance(); //Value of Check result, always true although i set the value false before

		set_brightness = (SeekBar) findViewById(R.id.seekBar2);
		set_brightness.setOnSeekBarChangeListener(mSeekBarChangeListener);

		mUVCCameraView = (SimpleUVCCameraTextureView)findViewById(R.id.UVCCameraTextureView1);
		mUVCCameraView.setAspectRatio(UVCCamera.DEFAULT_PREVIEW_WIDTH / (float)UVCCamera.DEFAULT_PREVIEW_HEIGHT);

		mUSBMonitor = new USBMonitor(this, mOnDeviceConnectListener);
	}
```
	
And this script below for Changed value from seekbar
	```
private final SeekBar.OnSeekBarChangeListener mSeekBarChangeListener = new SeekBar.OnSeekBarChangeListener(){
		@Override
		public void onProgressChanged(SeekBar seekBar, int progress, boolean fromUser) {
			UVC_Camera.setWhiteBlance(progress);
			UVC_Camera.updateCameraParams();
		}
		@Override
		public void onStartTrackingTouch(SeekBar seekBar) { }
		@Override
		public void onStopTrackingTouch(SeekBar seekBar) { }
	};
```
	
So, did you see any mistakes that I made?
Thanks before for your solution @williamdwarke

williamdwarke, 2017-01-03:
@bayudwirp,

I apologize for the delay; the holiday season has kept me very busy.

For starters, why do you have 2 UVCCamera objects? Second, try including an updateCameraParams() before you try and read the value of getAutoWhiteBlance(). In my experience, none of the changes to camera settings actually get written until you call that function. That may actually solve your problem. 

One more thing - try moving your camera white balance update code from onProgressChanged to onStopTrackingTouch, and just update a variable in onProgressChanged. The reason for this is that it's not a good idea to write camera parameters multiple times per second (onProgressChanged will constantly fire as the seekbar changes). Instead, I implemented something like this:

```
@Override
public void onProgressChanged(final SeekBar seekBar, final int progress, final boolean fromUser) {
	whiteBalance = progress;
}

@Override
public void onStopTrackingTouch(final SeekBar seekBar) {
	//Update the camera configuration to assign the new setting
	configSetting = ConfigSetting.WHITE_BALANCE;
	Toast.makeText(MainActivity.this, "Setting white balance to " + whiteBalance + "%", Toast.LENGTH_SHORT).show();
	mHandler.updateConfig();
}
```

However, there is a distinct difference which is that I am using a CameraThread (see usbCameraTest3), so my onStopTrackingTouch code triggers a message to the thread that sends the new white balance value to the camera. If you still can't get it to work you may consider switching over to a CameraThread as well, especially if you are you a multi-core device.

bayudwirp, 2017-01-05:
Hi @williamdwarke !

No problem for me. I'm sorry if I always disturb you lately.

Firstly, that's totaly my mistake to create double UVCCamera Object, and i try to using just one UVCCamera object. And still have same problem.
For example, if try to disable auto mode WhiteBalance with this script below:

```
mUVCCamera.updateCameraParams();
if (mUVCCamera.getAutoWhiteBlance()) {
	mUVCCamera.setAutoWhiteBlance(false);
	mUVCCamera.updateCameraParams();
}
mUVCCamera.updateCameraParams();
parameter_status = mUVCCamera.getAutoWhiteBlance();
textView1.setText("Auto Balance Status : "+parameter_status);
```
From this code, i always to write updateCameraParams before i read status and disable a auto mode. And, if i try to check again, whether the status has changed with command mUVCCamera.getAutoWhiteBlance() and it will always give a true status, which means a command to change value before didn't work because it does not change the status from true to false.

And, when i try to set (For Example) Brightness value and I try to look at the processes that occur using debug, i see this part from UVCCamera.java file:
```
    /**
     * @param brightness [%]
     */
	public synchronized void setBrightness(final int brightness) {
    	if (mNativePtr != 0) {
 		   final float range = Math.abs(mBrightnessMax - mBrightnessMin);
 		   if (range > 0)
 			   nativeSetBrightness(mNativePtr, (int)(brightness / 100.f * range) + mBrightnessMin);
    	}
    }
```

I see that the variable mBrightnessMax and mBrightnessMin always given 0, which causes regardless of the value that my input, will always lead to the results of the calculation is always 0. I think, because of it, cause when I try to set the value to set the Brightness, always 0 and can not be changed.
So, what do you think to solve my problem? should i recompile the library to change mBrightnessMax and mBrightnessMin value?

I'm sorry if i had many of question to you.
Thankyou for your attention @williamdwarke

rdbhanz, 2017-02-23:
@bayudwirp Any progress on this issue ? I am facing a very similar problem

bayudwirp, 2017-02-24:
Hi @rdbhanz .
I don't have any progress for this issue, and i'm still stuck if i tried to change parameter value for (example) brightness, contrast, etc.
How about your problem? you can't change value of brightness, contrast, etc?

rdbhanz, 2017-02-24:
@bayudwirp yeah same problem, I am trying to figure out how it is working in **usbCameraTest8** app. Getting a null pointer exception on the camera object that I am trying to mimic from  **usbCameraTest8**, some issue on the thread handling probably. Will post here if there is any progress

iSIlya, 2017-03-13:
@rdbhanz I updated the files in my project from the new version of the library (libuvccamera, usbKameraCommon) and modified the code from the example 8. I gave a sample code. Here is an example of the code that I got. 

```
private SeekBar.OnSeekBarChangeListener seekBarChangeListener =
            new SeekBar.OnSeekBarChangeListener() {
                @Override
                public void onProgressChanged(SeekBar seekBar, int progress,
                                              boolean fromUser) {
                    // TODO Auto-generated method stub
                    br = progress;

                }

                @Override
                public void onStartTrackingTouch(SeekBar seekBar) {
                    // TODO Auto-generated method stub
                }

                @Override
                public void onStopTrackingTouch(SeekBar seekBar) {
                    if (isActive()) {

                                setValue(seekBar.getProgress());

                        }
                    }
                    // TODO Auto-generated method stub
                };

private int setValue( final int value) {
        return mCameraHandler != null ? mCameraHandler.setValue(value) : 0;
    }
```
And edit AbstractUVCCameraHandler.java

```

public int setValue( final int value) {
		checkReleased();
		final CameraThread thread = mWeakThread.get();
		final UVCCamera camera = thread != null ? thread.mUVCCamera : null;
		if (camera != null) {

				camera.setBrightness(value);
				return camera.getBrightness();
		}
		throw new IllegalStateException();
	}

rdbhanz, 2017-03-14:
@iSIlya thanks for the reply, Brightness was not an issue for me, I was having trouble with white balance, which I figured out and is working. Still not able to get results for Exposure and Gain

iSIlya, 2017-03-14:
@rdbhanz I also do not know how to work with the exposition. If you find a solution, please inform.